Ysurac/openmptcprouter #3598: Any affordable 4g dongle?

muko82, 2024-10-17:
guys I need an help.

I have a 4g dongle but with very bad performances (7up/12dowb instead the thetering on mobile with the same SIM which is 35/70).
Do you have any suggestion about hardware (usb dongle) with decent up/down ?

Thanks

a834003831, 2024-10-17:
Hello, you can use Huawei or ZTE's

muko82, 2024-10-17:
I''ve deliberately not mentioned in the first post that I'm already using a 4g dongle made by huawei (e3372 serie) which is giving me such bad performances on my laptop...
I'll try to buy a ZTE mf79u.

Any suggesttion about the model?
Thanks.

a834003831, 2024-10-17:
> I''ve deliberately not mentioned in the first post that I'm already using a 4g dongle made by huawei (e3372 serie) which is giving me such bad performances on my laptop... I'll try to buy a ZTE mf79u.
> 
> Any suggesttion about the model? Thanks.

The two devices you mentioned are already considered excellent products in China. If you want a more stable or fast internet speed, we recommend purchasing 5Gcpe

closed 2025-01-21 by github-actions[bot]