Ysurac/openmptcprouter #3599: Pi-hole with OMR bypass on v0.61rc3-6.6

Nicras, 2024-10-17:
## Expected Behavior
I installed Pi-hole on the VPS as described in the Wiki and I would expect the DNS queries to go through the Pihole.

## Current Behavior
Pi-Hole isn't receiving DNS queries unless I manually set the VPN Address 10.255.255.1 as the only DNS forwarder. 

## Context (Environment)
When I set the Pi-hole as the only forwarder, OMR-Bypass won't work anymore.

I saved the OMR settings wizard and rebooted the router after installing Pi-hole several times now as described in the Wiki

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61rc3-6.6 r0+27346-c7ba5574f5
  - OpenMPTCProuter VPS version: 6.6.36-x64v3-xanmod1 0.1031-test
  - OpenMPTCProuter VPS provider: Strato 
  - OpenMPTCProuter platform: R5S
  - Country: Germany

Ysurac, 2024-10-17:
I'm not sure to understand your DNS configuration, can you put screenshot ?

Nicras, 2024-10-17:
[image]
 Should be absolute default ;-)

Ysurac, 2024-10-17:
It's seems that you enabled DNS over HTTPS, it's not the default.

Nicras, 2024-10-17:
not that I know...but where and how can I check it?

Ysurac, 2024-10-17:
In services->DNS over HTTPS

Nicras, 2024-10-17:
I didn't change anything there...
can I simply stop this service and remove both entries from DNS forwarder settings?

Ysurac, 2024-10-17:
If it's not enabled, then the #5054 is coming from something else...
You can replace both entry by VPN IP, this should not make any issues with OMR-ByPass as DNSMasq is still used.

Nicras, 2024-10-17:
how does the default DNS config look like?

Nicras, 2024-10-17:
After disabling the DoH and setting the VPN address for DNS the Pi-hole works but OMR-Bypass does not, even after a long time waiting.

AndreaLeidi, 2024-10-20:
Voila my config, perfectly working. 
I got a Pi-hole in my LAN at 192.168.100.106, feeding OMR 192.168.100.1, the DNS suggested to clients by DHCP
(seems unnecessarily redundant but leave it this way or you are gonna loose OMR-BYPASS feature). 
Moreover, no need to run DoH or Unbound on OMR.
icloud domains are referring to some annoying things related with iOS and MacOS, just google for them if you're interested.

[image]

closed 2025-01-23 by github-actions[bot]