Ysurac/openmptcprouter #3602: Install VPS KERNEL="6.6"

Silvanskii, 2024-10-18:
Сan't install kernel 6.6.
I am using debian 12.
After booting I enter the commands: 
apt-get update && apt-get upgrade
apt-get install -y ca-certificates && sed -i '/^mozilla\/DST_Root_CA_X3/s/^/!/' /etc/ca-certificates.conf && update-ca-certificates -f
wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL=“6.6” sh
and I get the following error
Maybe I'm doing something wrong?

Silvanskii, 2024-10-18:
root@vds2642154:~# wget -O - https://www.openmptcprouter.com/server/debian-x86_6                                                                                                   4.sh | KERNEL="6.6" sh
--2024-10-18 13:40:55--  https://www.openmptcprouter.com/server/debian-x86_64.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 185.131.64.122
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|185.131.64.122|:                                                                                                   443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 90532 (88K) [application/octet-stream]
Saving to: ‘STDOUT’

-                     0%[                    ]       0  --.-KB/s               C                                                                                                   heck user...
Check Linux version...
Check architecture...
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Remove lock and update packages list...
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://security.debian.org/debian-security bookworm-security InRelease
Reading package lists... Done
Install apt-transport-https, gnupg and openssh-server...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
openssh-server is already the newest version (1:9.2p1-2+deb12u3).
The following additional packages will be installed:
  dirmngr gnupg-l10n gnupg-utils gpg gpg-agent gpg-wks-client gpg-wks-server
  gpgconf gpgsm libassuan0 libksba8 libnpth0 pinentry-curses
Suggested packages:
  pinentry-gnome3 tor parcimonie xloadimage scdaemon pinentry-doc
The following NEW packages will be installed:
  apt-transport-https dirmngr gnupg gnupg-l10n gnupg-utils gpg gpg-agent
  gpg-wks-client gpg-wks-server gpgconf gpgsm libassuan0 libksba8 libnpth0
  pinentry-curses
0 upgraded, 15 newly installed, 0 to remove and 1 not upgraded.
Need to get 7906 kB of archives.
After this operation, 16.0 MB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main amd64 apt-transport-https all 2                                                                                                   .6.1 [25.2 kB]
Get:2 http://deb.debian.org/debian bookworm/main amd64 libassuan0 amd64 2.5.5-5                                                                                                    [48.5 kB]
Get:3 http://deb.debian.org/debian bookworm/main amd64 gpgconf amd64 2.2.40-1.1                                                                                                    [564 kB]
Get:4 http://deb.debian.org/debian bookworm/main amd64 libksba8 amd64 1.6.3-2 [1                                                                                                   28 kB]
Get:5 http://deb.debian.org/debian bookworm/main amd64 libnpth0 amd64 1.6-3 [19.                                                                                                   0 kB]
Get:6 http://deb.debian.org/debian bookworm/main amd64 dirmngr amd64 2.2.40-1.1                                                                                                    [792 kB]
Get:7 http://deb.debian.org/debian bookworm/main amd64 gnupg-l10n all 2.2.40-1.1                                                                                                    [1093 kB]
Get:8 http://deb.debian.org/debian bookworm/main amd64 gnupg-utils amd64 2.2.40-                                                                                                   1.1 [927 kB]
Get:9 http://deb.debian.org/debian bookworm/main amd64 gpg amd64 2.2.40-1.1 [949                                                                                                    kB]
Get:10 http://deb.debian.org/debian bookworm/main amd64 pinentry-curses amd64 1.                                                                                                   2.1-1 [77.4 kB]
Get:11 http://deb.debian.org/debian bookworm/main amd64 gpg-agent amd64 2.2.40-1                                                                                                   .1 [695 kB]
Get:12 http://deb.debian.org/debian bookworm/main amd64 gpg-wks-client amd64 2.2                                                                                                   .40-1.1 [541 kB]
Get:13 http://deb.debian.org/debian bookworm/main amd64 gpg-wks-server amd64 2.2                                                                                                   .40-1.1 [531 kB]
Get:14 http://deb.debian.org/debian bookworm/main amd64 gpgsm amd64 2.2.40-1.1 [                                                                                                   671 kB]
Get:15 http://deb.debian.org/debian bookworm/main amd64 gnupg all 2.2.40-1.1 [84                                                                                                   6 kB]
Fetched 7906 kB in 1s (6032 kB/s)
Selecting previously unselected package apt-transport-https.
(Reading database ... 35470 files and directories currently installed.)
Preparing to unpack .../00-apt-transport-https_2.6.1_all.deb ...
Unpacking apt-transport-https (2.6.1) ...
Selecting previously unselected package libassuan0:amd64.
Preparing to unpack .../01-libassuan0_2.5.5-5_amd64.deb ...
Unpacking libassuan0:amd64 (2.5.5-5) ...
Selecting previously unselected package gpgconf.
Preparing to unpack .../02-gpgconf_2.2.40-1.1_amd64.deb ...
[... cut ...]
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: https://repo.openmptcprouter.com/dists/buster/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/xanmod.gpg are ignored as                                              the file has an unsupported filetype.
-                                  99%[==========================================================> ]  88.00K  4.31KB/s    in 20s

Cannot write to ‘-’ (Broken pipe).

Ysurac, 2024-10-18:
This is stable kernel release, it doesn't support kernel 6.6 yet.
To use kernel 6.6, you need to use Release Candidate or Snapshot script: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

closed 2024-10-18 by Silvanskii