Ysurac/openmptcprouter #3604: 0.1031test 6.6 VPS has high CPU use. even not in use

fareign, 2024-10-19:
[image]
[image]

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 0.1031test
  - OpenMPTCProuter VPS provider:aliyun
  - OpenMPTCProuter platform: 
  - Country: China

a834003831, 2024-10-19:
很高兴在这里见到中国的使用者，希望能与你取得联系，例如微信，因为我希望跟你进行交流。

fareign, 2024-10-19:
> 很高兴在这里见到中国的使用者，希望能与你取得联系，例如微信，因为我希望跟你进行交流。

OK,OK

a834003831, 2024-10-19:
> > 很高兴在这里见到中国的使用者，希望能与你取得联系，例如微信，因为我希望跟你进行交流。
> 
> OK,OK

My, situ3831, how much is yours

cnuhd, 2024-10-24:
我的也是，显示是由于内存使用过高造成的

a834003831, 2024-10-24:
> 我的也是，显示是由于内存使用过高造成的

怎么联系你，交流一下经验，上面有我的

closed 2025-01-27 by github-actions[bot]