Ysurac/openmptcprouter #3605: Z8102-Ax M2-T

Adikul1, 2024-10-19:
[image]

[image]
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

**Describe why you are asking for this support?**

**Additional context**

I need support, this device went into bootloop by firmware provided by zbt and now nothing working

Adikul1, 2024-10-19:
[image]
These lights remain the same

Ysurac, 2024-10-19:
What image did you use ?

Ysurac, 2024-10-20:
The link is not working, and in the screenshot this doesn't seems to be an OpenMPTCProuter image.

Adikul1, 2024-10-20:
Please try this
https://we.tl/t-mCuhAXZ9h3

Ysurac, 2024-10-20:
My firmware is booting on z8102ax. There is 2 version, an old one with an error from ZBT that have only 64Mo flash, and the new one where it's fixed (via a boot firmware update) to 128Mo.
The old one doesn't have the web interface when you press reset button when powering the device, the new one have this web interface to update firmware (and so you should use the 128Mo version).
Image is available here: https://www.openmptcprouter.com/download
or here for latest snapshot image: https://snapshots.openmptcprouter.com/

Adikul1, 2024-10-21:
I flashed firmware provided by you
https://we.tl/t-mT5zFVi3DZ
This is video of steps and its not working, same green and red light are on, nothing is blinking.
I flashed this
[image]

closed 2025-01-24 by github-actions[bot]

rtk-teh, 2025-10-23:
Sorry for necroposting, but i have absolutely the same problem.

[image]

Loading both files, 64 and 128, and have fail in uboot loader.

[image]

Openwrt load normally.

[image]

What should i check?