Ysurac/openmptcprouter #3610: Regarding the inability to use the 6.6 kernel

a834003831, 2024-10-22:
- OpenMPTCProuter platform:r4s

I have tried burning this version multiple times, but it still cannot start properly. Sqashfs and ext4 are both unable to start

SD card replacement attempted - unable to resolve

Replaced burning tools such as BalenaEtcher and Rufus - unable to solve

And I found that the squashfs version, after burning, opened the SD with DiskGenius, and one of the folders showed: unformatted

Snapshot address: https://snapshots.openmptcprouter.com/5.4/r4s/targets/rockchip/armv8/?C=N&O=D

Snapshot name：openmptcprouter-v0.61rc3-6.6-r0+27346-c7ba5574f5-rockchip-armv8-friendlyarm_nanopi-r4s-ext4-sysupgrade.img
openmptcprouter-v0.61rc3-6.6-r0+27346-c7ba5574f5-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img

muko82, 2024-10-22:
On which HW are you trying to use that kernel?

a834003831, 2024-10-22:
> 你打算在哪个硬件上使用这个内核？

R4S,4G

muko82, 2024-10-22:
On RPI4 I had problems with 6.6 on USB power. The unit went in read-only after a reset.
It could be a similar issue

a834003831, 2024-10-22:
> On RPI4 I had problems with 6.6 on USB power. The unit went in read-only after a reset.
> It could be a similar issue

I have encountered problems in many versions: after inserting a USB card and restarting, the system displays as read-only and needs to flash again. The problems we encountered are likely to be the same

muko82, 2024-10-22:
I solved using kernel 5.4
Please consider that kernel 6.6 is coming with a RC and therfore "unstable" release of omr

closed 2025-01-26 by github-actions[bot]