Ysurac/openmptcprouter #3612: Query Regarding Master Link Update in OMR Switching Engine

tejaswini-04, 2024-10-23:
While examining the switching engine in OpenMPTCPRouter (OMR), we've made some observations regarding how an interface is chosen as the "master" link. Specifically, we noticed that the init.d file located in the GitHub repository (openmptcprouter-feeds/mptcp/files/etc/init.d/mptcp) gets triggered whenever there is a change in the interface. This seems to prompt an update to the "master" link.

We wanted to inquire if there is any QoS (Quality of Service) parameter that governs when the master link should be updated, or if this is solely based on interface status changes.

Ysurac, 2024-10-23:
The master link is the default gateway, by default (when "Master interface selection" is set to "balancing") when it's down it then use all available interfaces as gateway in load balancing.

Ysurac, 2024-10-23:
As others links, only if you set to check "link quality" in OMR-tracker Manager where you can, for example, set ping timeout, packet loss,...
Master link is used for initial connection so in most case quality shouldn't be a big issue.

closed 2025-01-28 by github-actions[bot]