Ysurac/openmptcprouter #3614: VPN & Sadowsock-issue

prubin1, 2024-10-27:
Ive installes vps on debian 12 and the latest openmptcp flash file for raspberry pi 4b and i have some issues with vpn connection and shadowsocks rust, but internet connection i have. I tried to reinstall several times everything and the issue is still there, what could be wrong?

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61-6.6 r0+27346-c7ba5574f5
  - OpenMPTCProuter VPS version:  OpenMPTCProuter VPS 0.103
  - OpenMPTCProuter VPS provider: edisglobal.com
  - OpenMPTCProuter platform: Raspberry pi 4b
  - VPS kernel version: 6.6.50

prubin1, 2024-10-27:
log:
Oct 27 08:10:58 OpenMPTCProuter daemon.err openvpn(omr)[6184]: Options error: You must define CA file (--ca) or CA path (--capath) and/or peer fingerprint verification (--peer-fingerprint)
Oct 27 08:10:58 OpenMPTCProuter daemon.warn openvpn(omr)[6184]: Use --help for more information.
Oct 27 08:11:01 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Oct 27 08:11:01 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Oct 27 08:11:03 OpenMPTCProuter daemon.warn openvpn(omr)[7454]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations. 
Oct 27 08:11:03 OpenMPTCProuter daemon.err openvpn(omr)[7454]: Options error: You must define CA file (--ca) or CA path (--capath) and/or peer fingerprint verification (--peer-fingerprint)
Oct 27 08:11:03 OpenMPTCProuter daemon.warn openvpn(omr)[7454]: Use --help for more information.
Oct 27 08:11:04 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Oct 27 08:11:04 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Oct 27 08:11:08 OpenMPTCProuter daemon.warn openvpn(omr)[8045]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations. 
Oct 27 08:11:08 OpenMPTCProuter daemon.err openvpn(omr)[8045]: Options error: You must define CA file (--ca) or CA path (--capath) and/or peer fingerprint verification (--peer-fingerprint)
Oct 27 08:11:08 OpenMPTCProuter daemon.warn openvpn(omr)[8045]: Use --help for more information.
Oct 27 08:11:11 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on 91.132.94.173, no server API answer on 91.132.94.173)
Oct 27 08:11:13 OpenMPTCProuter daemon.warn openvpn(omr)[8375]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations. 
Oct 27 08:11:13 OpenMPTCProuter daemon.err openvpn(omr)[8375]: Options error: You must define CA file (--ca) or CA path (--capath) and/or peer fingerprint verification (--peer-fingerprint)
Oct 27 08:11:13 OpenMPTCProuter daemon.warn openvpn(omr)[8375]: Use --help for more information.
Oct 27 08:11:16 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on 91.132.94.173, no server API answer on 91.132.94.173)
Oct 27 08:11:18 OpenMPTCProuter daemon.warn openvpn(omr)[8896]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations. 
Oct 27 08:11:18 OpenMPTCProuter daemon.err openvpn(omr)[8896]: Options error: You must define CA file (--ca) or CA path (--capath) and/or peer fingerprint verification (--peer-fingerprint)
Oct 27 08:11:18 OpenMPTCProuter daemon.warn openvpn(omr)[8896]: Use --help for more information.
Oct 27 08:11:23 OpenMPTCProuter daemon.warn openvpn(omr)[9462]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations. 
Oct 27 08:11:23 OpenMPTCProuter daemon.err openvpn(omr)[9462]: Options error: You must define CA file (--ca) or CA path (--capath) and/or peer fingerprint verification (--peer-fingerprint)
Oct 27 08:11:23 OpenMPTCProuter daemon.warn openvpn(omr)[9462]: Use --help for more information.
Oct 27 08:11:28 OpenMPTCProuter daemon.warn openvpn(omr)[9866]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations. 
Oct 27 08:11:28 OpenMPTCProuter daemon.err openvpn(omr)[9866]: Options error: You must define CA file (--ca) or CA path (--capath) and/or peer fingerprint verification (--peer-fingerprint)
Oct 27 08:11:28 OpenMPTCProuter daemon.warn openvpn(omr)[9866]: Use --help for more information.
Oct 27 08:11:33 OpenMPTCProuter daemon.warn openvpn(omr)[10347]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations. 
Oct 27 08:11:33 OpenMPTCProuter daemon.err openvpn(omr)[10347]: Options error: You must define CA file (--ca) or CA path (--capath) and/or peer fingerprint verification (--peer-fingerprint)
Oct 27 08:11:33 OpenMPTCProuter daemon.warn openvpn(omr)[10347]: Use --help for more information.
Oct 27 08:11:38 OpenMPTCProuter daemon.warn openvpn(omr)[10630]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations. 
Oct 27 08:11:38 OpenMPTCProuter daemon.err openvpn(omr)[10630]: Options error: You must define CA file (--ca) or CA path (--capath) and/or peer fingerprint verification (--peer-fingerprint)
Oct 27 08:11:38 OpenMPTCProuter daemon.warn openvpn(omr)[10630]: Use --help for more information.
Oct 27 08:11:44 OpenMPTCProuter daemon.warn openvpn(omr)[10891]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher nego [... cut ...]
Oct 27 08:20:35 OpenMPTCProuter daemon.warn openvpn(omr)[19187]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations. 
Oct 27 08:20:35 OpenMPTCProuter daemon.err openvpn(omr)[19187]: Options error: You must define CA file (--ca) or CA path (--capath) and/or peer fingerprint verification (--peer-fingerprint)
Oct 27 08:20:35 OpenMPTCProuter daemon.warn openvpn(omr)[19187]: Use --help for more information.

lkyddyy, 2024-10-27:
This is highly likely a problem with the server's firewall settings

Ysurac, 2024-10-27:
It's not able to contact VPS. Check that VPS key is correctly set and ports 65000-65535 are open on VPS provider side.
You can check that API is reachable, via SSH on the router with ```curl -k https://<vpsip>:65500/``` (you should get a welcome message).

xNagatsu, 2024-10-28:
Same issue here since updating to version 0.1031
On the router, I still see my previous version 0.1028 instead of 0.1031, even though the VPS is correctly updated to 0.1031 and has been restarted.

Restarting GloryTunTCP fixes the VPN issue, but I still get the error: "Unable to contact the administration script."

The command `curl -k https://<vpsip>:65500/`  returns the error: "Connection refused."

However, on my AWS VPS, all ports are open.

Ysurac, 2024-10-28:
@xNagatsu what is the reuslt of ```systemctl restart omr-admin && journalctl -u omr-admin``` ?

xNagatsu, 2024-10-28:
@Ysurac Job for omr-admin.service failed because the control process exited with error code.
See `systemctl status omr-admin.service` and `journalctl -xe` for details.

`systemctl status omr-admin.service`  return 

```
 omr-admin.service - OMR-Admin
   Loaded: loaded (/lib/systemd/system/omr-admin.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2024-10-28 14:00:35 UTC; 9s ago
  Process: 5586 ExecStart=/usr/bin/omr-admin.py (code=exited, status=1/FAILURE)
 Main PID: 5586 (code=exited, status=1/FAILURE)

Oct 28 14:00:35 ip-172-31-11-2 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
Oct 28 14:00:35 ip-172-31-11-2 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Oct 28 14:00:35 ip-172-31-11-2 systemd[1]: omr-admin.service: Service RestartSec=100ms expired, scheduling restart.
Oct 28 14:00:35 ip-172-31-11-2 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 5.
Oct 28 14:00:35 ip-172-31-11-2 systemd[1]: Stopped OMR-Admin.
Oct 28 14:00:35 ip-172-31-11-2 systemd[1]: omr-admin.service: Start request repeated too quickly.
Oct 28 14:00:35 ip-172-31-11-2 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Oct 28 14:00:35 ip-172-31-11-2 systemd[1]: Failed to start OMR-Admin.

```
After wait a moment, `systemctl status omr-admin.service`  return 

```
● omr-admin.service - OMR-Admin
   Loaded: loaded (/lib/systemd/system/omr-admin.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2024-10-28 14:01:24 UTC; 6ms ago
 Main PID: 7223 (omr-admin.py)
    Tasks: 0 (limit: 1151)
   Memory: 0B
   CGroup: /system.slice/omr-admin.service
           └─7223 /usr/bin/python3 /usr/bin/omr-admin.py
```

journalctl -xe return

```
-- 
-- A start job for unit omr-admin.service has finished successfully.
-- 
-- The job identifier is 210015.
Oct 28 14:05:19 ip-172-31-11-2 omr-admin.py[15021]: ERROR:    [Errno -9] Address family for hostname not supported
Oct 28 14:05:19 ip-172-31-11-2 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- An ExecStart= process belonging to unit omr-admin.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 1.
Oct 28 14:05:19 ip-172-31-11-2 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- The unit omr-admin.service has entered the 'failed' state with result 'exit-code'.
Oct 28 14:05:19 ip-172-31-11-2 systemd[1]: omr-admin.service: Service RestartSec=100ms expired, scheduling restart.
Oct 28 14:05:19 ip-172-31-11-2 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 5.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Automatic restarting of the unit omr-admin.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Oct 28 14:05:19 ip-172-31-11-2 systemd[1]: Stopped OMR-Admin.
-- Subject: A stop job for unit omr-admin.service has finished
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A stop job for unit omr-admin.service has finished.
-- 
-- The job identifier is 210067 and the job result is done.
Oct 28 14:05:19 ip-172-31-11-2 systemd[1]: omr-admin.service: Start request repeated too quickly.
Oct 28 14:05:19 ip-172-31-11-2 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- The unit omr-admin.service has entered the 'failed' state with result 'exit-code'.
Oct 28 14:05:19 ip-172-31-11-2 systemd[1]: Failed to start OMR-Admin.
-- Subject: A start job for unit omr-admin.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit omr-admin.service has finished with a failure.
-- 
-- The job identifier is 210067 and the job result is failed.
```

Ysurac, 2024-10-28:
@xNagatsu API is not able to bind correctly to IP. What is the result of ```ip a``` ?

xNagatsu, 2024-10-28:
```
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 9001 qdisc fq state UP group default qlen 1000
    link/ether 06:82:31:04:1a:ee brd ff:ff:ff:ff:ff:ff
    inet 172.31.11.2/20 brd 172.31.15.255 scope global dynamic eth0
       valid_lft 3065sec preferred_lft 3065sec
    inet6 fe80::482:31ff:fe04:1aee/64 scope link 
       valid_lft forever preferred_lft forever
3: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1400 qdisc fq state UNKNOWN group default qlen 2000
    link/none 
    inet 10.255.252.1 peer 10.255.252.2/32 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::efca:ed58:f427:47af/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
4: dsvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 9000 qdisc fq state UNKNOWN group default qlen 500
    link/none 
    inet 10.255.251.1 peer 10.255.251.2/32 scope global dsvpn0
       valid_lft forever preferred_lft forever
    inet6 64:ff9b::aff:fb01 peer 64:ff9b::aff:fb02/96 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::8bea:8e24:e308:24e/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
6: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc cake state UNKNOWN group default qlen 100
    link/none 
    inet 10.255.254.1/30 brd 10.255.254.3 scope global gt-udp-tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::8c4b:7785:4185:c841/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
7: mlvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1444 qdisc fq state UNKNOWN group default qlen 500
    link/none 
    inet 10.255.253.1/24 brd 10.255.253.255 scope global mlvpn0
       valid_lft forever preferred_lft forever
    inet6 fe80::efd7:c37a:26a1:8bf0/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
8: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
9: omr-6in4-user0@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1480 qdisc noqueue state UNKNOWN group default qlen 1000
    link/sit 10.255.255.1 peer 10.255.255.2
    inet6 fe80::aff:ff01/64 scope link 
       valid_lft forever preferred_lft forever
    inet6 fe80::a00:1/126 scope link 
       valid_lft forever preferred_lft forever
10: ovpnbonding4: <BROADCAST,MULTICAST,SLAVE,UP,LOWER_UP,80000> mtu 1440 qdisc fq master omr-bonding state UNKNOWN group default qlen 100
    link/ether 42:e2:bb:0e:c9:3a brd ff:ff:ff:ff:ff:ff
11: ovpnbonding3: <BROADCAST,MULTICAST,SLAVE,UP,LOWER_UP,80000> mtu 1440 qdisc fq master omr-bonding state UNKNOWN group default qlen 100
    link/ether 42:e2:bb:0e:c9:3a brd ff:ff:ff:ff:ff:ff
12: client-wg0: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 8921 qdisc noqueue state UNKNOWN group default qlen 1000
    link/none 
    inet 10.255.246.1/24 scope global client-wg0
       valid_lft forever preferred_lft forever
13: ovpnbonding5: <BROADCAST,MULTICAST,SLAVE,UP,LOWER_UP,80000> mtu 1440 qdisc fq master omr-bonding state UNKNOWN group default qlen 100
    link/ether 42:e2:bb:0e:c9:3a brd ff:ff:ff:ff:ff:ff
14: wg0: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 8921 qdisc noqueue state UNKNOWN group default qlen 1000
    link/none 
    inet 10.255.247.1/24 scope global wg0
       valid_lft forever preferred_lft forever
15: ovpnbonding8: <BROADCAST,MULTICAST,SLAVE,UP,LOWER_UP,80000> mtu 1440 qdisc fq master omr-bonding state UNKNOWN group default qlen 100
    link/ether 42:e2:bb:0e:c9:3a brd ff:ff:ff:ff:ff:ff
16: ovpnbonding7: <BROADCAST,MULTICAST,SLAVE,UP,LOWER_UP,80000> mtu 1440 qdisc fq master omr-bonding state UNKNOWN group default qlen 100
    link/ether 42:e2:bb:0e:c9:3a brd ff:ff:ff:ff:ff:ff
17: ovpnbonding1: <BROADCAST,MULTICAST,SLAVE,UP,LOWER_UP,80000> mtu 1440 qdisc fq master omr-bonding state UNKNOWN group default qlen 100
    link/ether 42:e2:bb:0e:c9:3a brd ff:ff:ff:ff:ff:ff
18: ovpnbonding2: <BROADCAST,MULTICAST,SLAVE,UP,LOWER_UP,80000> mtu 1440 qdisc fq master omr-bonding state UNKNOWN group default qlen 100
    link/ether 42:e2:bb:0e:c9:3a brd ff:ff:ff:ff:ff:ff
19: ovpnbonding6: <BROADCAST,MULTICAST,SLAVE,UP,LOWER_UP,80000> mtu 1440 qdisc fq master omr-bonding state UNKNOWN group default qlen 100
    link/ether 42:e2:bb:0e:c9:3a brd ff:ff:ff:ff:ff:ff
20: tun1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq state UNKNOWN group default qlen 100
    link/none 
    inet 10.255.250.1 peer 10.255.250.2/32 scope global tun1
       valid_lft forever preferred_lft forever
    inet6 fe80::3761:1b96:fbc4:2357/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
21: omr-bonding: <BROADCAST,MULTICAST,MASTER,UP,LOWER_UP,80000> mtu 1440 qdisc noqueue state UP group default qlen 1000
    link/ether 42:e2:bb:0e:c9:3a brd ff:ff:ff:ff:ff:ff
    inet 10.255.248.1/32 scope global omr-bonding
       valid_lft forever preferred_lft forever
    inet6 fe80::40e2:bbff:fe0e:c93a/64 scope link 
       valid_lft forever preferred_lft forever
22: gt-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc mq state UNKNOWN group default qlen 500
    link/none 
    inet 10.255.255.1/30 brd 10.255.255.3 scope global gt-tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::aa86:cb8b:af71:3c86/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
```

Ysurac, 2024-10-28:
Strange, should work.
It's Debian 12 ?

xNagatsu, 2024-10-28:
Nope, Debian 10

Ysurac, 2024-10-28:
So you are using it with kernel 5.4 ?
I think it's due to an old package. I will try to add a workaround.

xNagatsu, 2024-10-28:
@Ysurac Yes, kernel 5.4

prubin1, 2024-10-28:
For the above issue....i solved it with opening ip range tcp/udp with iptables and it worked. Thanks.

closed 2024-10-28 by prubin1

Ysurac, 2024-10-28:
@xNagatsu I've modified VPS test script to update Debian to Debian 12 in all cases, you can test it here: ```wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | KERNEL="5.4" sh```
If this still don't work, please open a new issue.