Ysurac/openmptcprouter #3616: kmod-gpio-button-hotplug v0.61 kernel6.6 x86 compile erro

fareign, 2024-10-27:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.50~410338f4c01886075e6ae825727de685-r1) for kmod-gpio-button-hotplug
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-gpio-button-hotplug found, but incompatible with the architectures configured

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.61
  - OpenMPTCProuter VPS version: 0.1031
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64
  - Country: cn

Ysurac, 2024-10-27:
I'm not sure to understand when you get this error. On a custom build or via package manager ?

fareign, 2024-10-27:
> I'm not sure to understand when you get this error. On a custom build or via package manager ?

clean all .rebluild is ok now.

closed 2024-10-27 by fareign