Ysurac/openmptcprouter #3623: Problem, unable to contact administration script

xNagatsu, 2024-10-29:
## Expected Behavior
I've updated on Debian 10 VPS from 0.1028 to 0.1031. After that i've test your script from issue #3614
wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | KERNEL="5.4" sh

## Current Behavior

Before the test version, I have the message : Unable to contact the administrator script.

I've run your script, ` wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | KERNEL="5.4" sh`  i have an error because my sources.list was : 
`deb http://archive.debian.org/debian buster main [...]` 
and you script change the sources.list to :
`deb http://archive.debian.org/debian bullseye main [...]`
and after taht i've change manually to :
`deb http://deb.debian.org/debian bullseye main [...]`

Now i don't have acces to my server in SSH, but in AWS i've download the journalclt : [here](https://pastebin.com/0c8jrX7y)
And i have the capture of instance below : 
[image]

## Specifications

  - OpenMPTCProuter version: 0.59-5.4
  - OpenMPTCProuter VPS version: 0.1031
  - OpenMPTCProuter VPS provider: AWS
  - OpenMPTCProuter platform: RPI3

Ysurac, 2024-10-29:
And now you can join ```curl -k https://<vpsip>:65500/``` ?

xNagatsu, 2024-10-29:
Nope, but i can't ping my server too :/ `curl: (28) Failed to connect to 1.2.3.4 port 65500 after 15431 ms: Operation timed out`

Ysurac, 2024-10-29:
Check on VPS in ```/etc/shorewall/params.net``` that the correct interface with internet access is set.

xNagatsu, 2024-10-29:
I can't connect to my server with SSH, I can only watch the "instance screenshot". 
I have a new error in instance screenshot, maybe this can help you.
[image]

I think for my server I will having to redeploy a volume, but before i give you a maximum information.

Ysurac, 2024-10-29:
It's AWS, you should have a console access where you can enter login and pass.
Without ping answer it's often a firewall issue.

xNagatsu, 2024-10-29:
I can't use EC2. So I have redeploy a new root volume, Debian12 with 0.1031 and everything is OK.
I've search a changelog between 0.1030 and 0.1031 but i don't find the changelog.
Is possible to have it please ?

closed 2025-02-02 by github-actions[bot]