Ysurac/openmptcprouter #3624: OpenMPTCProuter Server 0.1031 VPS installation script fails on Debian 12 with IPv6 disabled

BigFanOpenmptcprouter, 2024-10-29:
## Expected Behavior
The script should install successfully without errors ie. should end with:
`====================================================================================
OpenMPTCProuter Server 0.1031 is now installed !
 SSH port: 65222 (instead of port 22)
====================================================================================`

## Current Behavior
When IPv6 is disabled using /etc/sysctl.conf
`net.ipv6.conf.all.disable_ipv6=1
net.ipv6.conf.default.disable_ipv6=1
net.ipv6.conf.lo.disable_ipv6=1
net.ipv6.conf.ens3.disable_ipv6=1`
`sysctl -p`

The installer fails with the following error:
`--2024-10-29 08:04:20--  https://www.openmptcprouter.com/server-test/openvpn-bonding8.conf
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 79.127.237.132, 2400:52e0:1e03::886:1
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|79.127.237.132|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 355 [application/octet-stream]
Saving to: '/etc/openvpn/bonding8.conf'

/etc/openvpn/bonding8.conf      100%[====================================================>]     355  --.-KB/s    in 0s      

2024-10-29 08:04:20 (6.97 MB/s) - '/etc/openvpn/bonding8.conf' saved [355/355]

sed: can't read /etc/openvpn.tun0.conf: No such file or directory`

## Possible Solution
Workarounds:
A) Accept the IPv6 setup.
or
B) Stick with version 0.1030 (at least for now)
or
C) Try to edit/remove this part in the code:
`	if [ "$(ip -6 a 2>/dev/null)" = "" ]; then
		sed -i 's/proto tcp6-server//' /etc/openvpn.tun0.conf
		sed -i 's/proto udp6//' /etc/openvpn.tun1.conf
		if [ "$OPENVPN_BONDING" = "yes" ]; then
			sed -i 's/proto udp6//' /etc/openvpn.bonding1.conf
			sed -i 's/proto udp6//' /etc/openvpn.bonding2.conf
			sed -i 's/proto udp6//' /etc/openvpn.bonding3.conf
			sed -i 's/proto udp6//' /etc/openvpn.bonding4.conf
			sed -i 's/proto udp6//' /etc/openvpn.bonding5.conf
			sed -i 's/proto udp6//' /etc/openvpn.bonding6.conf
			sed -i 's/proto udp6//' /etc/openvpn.bonding7.conf
			sed -i 's/proto udp6//' /etc/openvpn.bonding8.conf
		fi
	fi`
It's lines 1532-1545.

## Steps to Reproduce the Problem

  1. Clean install Debian 12 x64, disable IPv6 using /etc/sysctl.con & sysctl -p
  2. Try Installing the script: wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL="6.6" sh
  3. It fails with error.

## Context (Environment)
Debian 12 x64 with IPv6 disabled. That's the issue here.

## Specifications

  - OpenMPTCProuter version: N/A
  - OpenMPTCProuter VPS version: OpenMPTCProuter Server 0.1031
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: x86_64
  - Country: EU/no restrictions.

Ysurac, 2024-10-29:
Thanks, it's now fixed in snapshot script ( https://github.com/Ysurac/openmptcprouter/wiki/Snapshots )

BigFanOpenmptcprouter, 2024-10-29:
Thank you for quick fix. Looking forward for next release. I suppose for now I will stick with the in-build configuration settings:

> network.wan1.ipv6='0'

For others looking for the GUI settings: Wizard - Server settings - "Advanced settings" - IPv6 settings - Enable IPv6" set it to "Disabled".

BigFanOpenmptcprouter, 2024-11-11:
Update: I can confirm that the snapshot from today "OpenMPTCProuter Server 0.1032-test" **solved the issue**.
On router side I used "openmptcprouter-v0.61-6.6-r0"

The IPv6 can be disabled on the VPS side with no issues and the installation passes successfully.

Ysurac, 2024-11-11:
Thanks for the feedback.