Ysurac/openmptcprouter #3625: can V0.61 use 6.1kernel?

fareign, 2024-10-29:
I test 6.6 kernel  find a serious BUG.
if I use USB CDC Device.and internal RM500Q model.
All of the usb device which under VIA805 will be lost.
but kernel 6.1 0.60 is OK
 so there must the drive problem

Ysurac, 2024-10-29:
Please use template.
When you plug a device you loose all other USB devices ?
Did you check Status->kernel log (or dmesg command) ?

fareign, 2024-10-30:
`110.078474] cdc ether 1-1.4:1.0 eth3: NETDEV WATCHD0G: CPU: 0: transmit queue 0 timed out 13340 ms
110.1584831 xhci hcd 8080:85:80.0: xHcI host not responding to stop endpoint command
110.158303] xhci hcd 0008:05:00.0: xHcI host controller not responding, assume dead110.166180] xhci hcd 0000:05:80.0: Hc died; cleaning up110.1726531 usb 1-1:UsB disconnect, device number 2110.1776251 usb 1-1.3:UsB disconnect, device number 5110.182848] cdc ether 1-1.3:1.0 eth2: unregister 'cdc ether’usb-0808:85:08.0-1.3, ZTE CDc Ethernet Device110.192534]usb 2-1:UsB disconnect, device number 2110.198164] option1 ttyUsBe: GsM modem(1-port)converternow disconnected from ttyUSB8110.206331]option 2-1:1.0:device disconnected110.212644] option1 ttyUsB1:GSM modem(1-port)converter now disconnected from ttyUsB1110.220780]option 2-1:1.1: device disconnected118.2260967option1 ttyUsB2: GSM modem(1-port)now disconnected from ttyUsB2converter110.2342757option 2-1:1.2:device disconnectedconverter now disconnected from ttyUsB3110.240398]option1 ttyUsB3:GSM modem(1-port)110.2485467option 2-1:1.3: device disconnected110.253270] cdc ether 2-1:1.4 usbg: unregister 'cdc ether'usb-0008:85:00.8-1, coc Ethernet Device110.3318557usb 2-2:UsB disconnect,device number 3110.337186]option1 ttyUsB4: GSM modem(1-port)converter now disconnected from ttyUsB4110.3453527option 2-2:1.0:device disconnected110.3502217option1 ttyUsB5:GSM modem(1-port)now disconnected from ttyUSB5converter110.3583397option 2-2:1.1: device disconnected110.363319]now disconnected from ttyUSB6option1 ttyUsB6: GsM modem(1-port)converter
110.371498]option 2-2:1.2:device disconnected110.3764041 option1 ttyUsB7:GSM modem(1-port)converter now disconnected from ttyUSB7110.384568] option 2-2:1.3: device disconnected110.389234] cdc ether 2-2:1.4 usb1: unregister 'cdc ether'usb-0008:85:00.0-2, coc Ethernet Device110.5708741 usb 1-1.4:UsB disconnect, device number 6110.5761191 cdc ether 1-1.4:1.0 eth3: unregister 'cdc ether’usb-8000:05:80.0-1.4, 2TE cc Ethernet Deviceroot@OpenMPTCProuter:~#

`

rootdOpenMPTCProuter:~#lsusbBus001 Device001: ID ld6b:0002 inux 6.6.50 xhci-hcd xHcI Host ControllerBus001 Device002:ID 2109:3431USB2.0 HubBus001 Device004:ID 19d2:1405 DEMo,Incorporated DEMo Mobile Boardband002001:ID ld6b:0003 Linux 6.6.50 xhci-hcd xHcI Host ControllerBuSDeviceID 2c7c:0800Bus002Device002:Quectel RM5020-GL003:ID 2c7c:0800Quectel RM5020-GLBuS002DeviceBus003 Device001:ID 1d6b:0002Linux 6.6.50 xhci-hcdXHCI HostControllerroot@OpenMPTCProuter:~#lsusbBus001 Device001:ID 1d6b:0002LinuX6.6.50xhci-hcdXHCIHostController002 Device 001:ID 1d6b:00036.6.50xHCIHostBuSLinu8xhci-hcdController003 Device 00l: ID 1d6b:0002BusLinuX6.6.50xhci-hcdXHCIHostController
root@openMPTcProuter:~#

fareign, 2024-10-30:
SO I'm not sure where is the wrong?,maybe vl805 is not suitable with 6.6 kernel.
So I want to compile 0.61 with 6.1 kernel. and have a try?

fareign, 2024-10-30:
I do a deep test .
hardwave:
cm4+vl805+2*RM500q+1*usb2.0+1*usb3.0
1.only two USB  work fine.
2.rm500q*2+usb2.0 also ok
3.if insert a CDC device on usb3.0 , about 1-3miniute,all of the USB will lost.

Ysurac, 2024-10-30:
This looks like a power supply issue.
Did you try with a powered USB hub ?

fareign, 2024-10-30:
> This looks like a power supply issue. Did you try with a powered USB hub ?

it is kernel and something with vl805 driver
I use 6.1kernel works very well,
NOW,I compile 0.61 with 6.1 kernel. seems no problem.

muko82, 2024-10-31:
I'm having exactly the same issue. It's not a usb power issue because with exactly the same HW I don't have problems on kernel 6.1

Ysurac, 2024-10-31:
I just updated bcm27xx-eeprom in snapshot, a firmware update may help to solve the issue.

muko82, 2024-10-31:
Sorry for the question: how to update the eprom?
I suppose that rpi-eprom -a isn't useful for that... :-)

Ysurac, 2024-10-31:
Should still be via rpi-eeprom-update

muko82, 2024-10-31:
Sorry but I'm a little bit lost: what's the exact folder from which I should download the file?

Ysurac, 2024-10-31:
It's still compiling, then you will need to update with image from https://snapshots.openmptcprouter.com/ and run the ```rpi-eeprom-update -a``` command.

muko82, 2024-10-31:
So i will have to flash 0.62 snapshot and run rpi-update. OK... Thanks

muko82, 2024-11-01:
USB are now working but....
After a normal reboot the FS goes in R/O. We still have bugs with kernel 6.6

fareign, 2024-11-01:
I compile 0.61 with kernel 6.1.everything is OK

closed 2025-02-05 by github-actions[bot]