Ysurac/openmptcprouter #3626: DDNS for MPTCPRouter for VPS

lkimloong, 2024-10-30:
Is MPTCPRouter are able to run on non-fixed ip for the VPS by using ddns only?

Ysurac, 2024-10-30:
Yes but it's not recommended.

lkimloong, 2024-10-30:
Is there any alternative if we do not have fixed public ip?

Ysurac, 2024-10-30:
In all case VPS in datacenter have a fixed IP, else it's not a real VPS.
IPv6 can also me used, it can sometimes be fixed even if IPv4 is not, but all connections must have IPv6 support.

vempire-ghost, 2024-10-30:
I have been using OMR for a year with DDNS instead of a fixed IP, and I haven't had any issues so far.

lkimloong, 2024-10-31:
> In all case VPS in datacenter have a fixed IP, else it's not a real VPS. IPv6 can also me used, it can sometimes be fixed even if IPv4 is not, but all connections must have IPv6 support.

Meaning if im lucky enough the IPv6 sometime can be fixed because of the 340 undecillion addresses produce by IPv6? Currently I am using a self hosted VPS which I do not have the public IPv4.

lkimloong, 2024-10-31:
> I have been using OMR for a year with DDNS instead of a fixed IP, and I haven't had any issues so far.

Is there any special configuration need to be done on the VPS? If yes, do you mind to share the configuration steps?

vempire-ghost, 2024-10-31:
> > I have been using OMR for a year with DDNS instead of a fixed IP, and I haven't had any issues so far.
> 
> Is there any special configuration need to be done on the VPS? If yes, do you mind to share the configuration steps?

You just need to configure the client on the VPS that will update the DDNS. In my case, I use the No-IP client, but there's also ddclient, which is very good and works with various providers.

lkimloong, 2024-11-01:
> You just need to configure the client on the VPS that will update the DDNS. In my case, I use the No-IP client, but there's also ddclient, which is very good and works with various providers.

Thanks for the info. Let me try to configure it first.

closed 2025-02-05 by github-actions[bot]