Ysurac/openmptcprouter #3627: Connected 6 lines, but only 4 have traffic.

iazxq, 2024-10-31:
## Expected Behavior

All 6 aggregated lines should have traffic when connected.

## Current Behavior

I am on the x86 platform with firmware version v0.61-6.6, and I have connected 6 aggregated lines. In the system menu under Aggregated Router - Status, I can see that the status of all 6 lines is normal. However, when I perform a speed test, I find that only 4 lines have traffic. There are a total of 6 lines, wan1 to wan6, but wan2 and wan6 have no traffic. If I turn off two lines, such as wan3 and wan4, the remaining four lines all have traffic. Regardless of how many lines are connected, only 4 lines have traffic. I have tested each line individually, and they are all functioning normally.

## Specifications

  - OpenMPTCProuter version: 0.61_6.6
  - OpenMPTCProuter VPS version: 0.1031 6.1
  - OpenMPTCProuter VPS provider: debian 12
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-10-31:
OpenMPTCProuter allow 8 IPs for now. So if connections have IPv4 & IPv6 support this can make less lines used.
What is the result of ```ip mptcp endpoint``` or Network->MPTCP, "MPTCP fullmesh" ?

iazxq, 2024-10-31:
In Network->mptcp->MPTCP Fullmesh, I can see 6 entries as follows:
192.168.10.1 id 1 subflow fullmesh dev eth1
192.168.20.1 id 2 subflow fullmesh dev eth2
192.168.30.1 id 3 subflow fullmesh dev eth3
192.168.40.1 id 4 subflow fullmesh dev eth4
192.168.50.1 id 5 subflow fullmesh dev eth5
192.168.60.1 id 6 subflow fullmesh dev eth6

Ysurac, 2024-10-31:
All is ok on this side.
I would need a screenshot of System->OpenMPTCProuter, "Status" page.

iazxq, 2024-10-31:
[image]

Ysurac, 2024-10-31:
What are the connections with traffic ? Always the same ?

a834003831, 2024-10-31:
"Hello, what hardware are you using? I also want to buy an x86, but I’m not sure which configuration to choose."

darkman1983, 2024-11-05:
Under Network -> MPTCP, which value do you have for ADD_Address? Standard here is 1, if you add more WAN-Interfaces, you have to increase this value eventually.

Description:
The "Max add address" option in OpenMPTCProuter defines the maximum number of IP addresses that the MPTCP connection can advertise or add to an existing connection. This is useful for managing how many network paths are available for the MPTCP connection to utilize, especially when multiple network interfaces are in use.

In an MPTCP setup with multiple WAN connections, this setting can limit the number of addresses (i.e., network paths) that OpenMPTCProuter will use simultaneously. For instance, if you have several interfaces (like wired, Wi-Fi, or cellular), setting a limit can prevent the router from trying to add too many addresses, which could lead to resource overuse or unnecessary complexity in the connection.

Higher values allow more paths and could improve redundancy and load balancing across multiple connections, while lower values limit paths, which may save resources and reduce latency if only a few connections are required.

So, the thing Ysurac is reffering too, is that other setting:
Max subflows

The "Max subflows" option in OpenMPTCProuter controls the maximum number of subflows (individual TCP connections) that MPTCP can establish within a single logical MPTCP connection.

Here's what this means in practice:

Enhanced Performance: Each subflow corresponds to a separate path across different network interfaces or IP addresses (e.g., one via Wi-Fi, another via cellular). More subflows generally allow better utilization of multiple interfaces, enabling higher throughput or increased resilience if one path has issues.

Network Load and Redundancy: More subflows can enhance redundancy by distributing the load across various interfaces. If one subflow experiences issues (like increased latency or packet loss), the others can maintain the connection’s stability and performance.

Resource Usage: However, more subflows also increase resource usage, both on the router and in terms of bandwidth overhead. Limiting subflows may be desirable to conserve system resources or reduce the complexity of the connection if only a limited number of paths are beneficial.

So, adjusting "Max subflows" allows you to control how many parallel paths the router tries to use for each MPTCP connection, balancing performance with resource and stability considerations in your setup.

closed 2025-02-08 by github-actions[bot]

lhc1993, 2025-09-26:
> across

@darkman1983 Do you have an introduction to the detailed description of each configuration option you saw above? I didn't find any introduction to the relevant options from GitHub.