Ysurac/openmptcprouter #3628: Failed to install MTR on OMR snapshot 0.62 5.4.

vempire-ghost, 2024-10-31:
## Expected Behavior
The installation of the MTR package works normally, as in previous versions of OMR.

## Current Behavior
When I run the command
opkg update
opkg install mtr
The console returns these errors.

root@OpenMPTCProuter:~# opkg update
Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz
*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/packages/Packages.gz
*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/packages/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/base/Packages.gz
*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/base/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/routing/Packages.gz
*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/routing/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/telephony/Packages.gz
*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/telephony/Packages.gz

Downloading https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/targets/x86/64/packages/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/targets/x86/64/packages/Packages.gz

Downloading https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/base/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/base/Packages.gz

Downloading https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/luci/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/luci/Packages.gz

Downloading https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/openmptcprouter/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/openmptcprouter/Packages.gz

Downloading https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/packages/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/packages/Packages.gz

Collected errors:
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/base/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/routing/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/telephony/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/targets/x86/64/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/base/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/luci/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/openmptcprouter/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/packages/Packages.gz, wget returned 8.
root@OpenMPTCProuter:~# opkg install mtr
Unknown package 'mtr'.
Collected errors:
 * opkg_install_cmd: Cannot install package mtr.

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: The console returns these errors.
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1031-test
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: x86_64
  - Country: US

Ysurac, 2024-10-31:
It's now fixed, but I'm not sure I will keep 5.4 for v0.62...

vempire-ghost, 2024-10-31:
Ah, I understand. Unfortunately, I tested the redundant scheduler on 6.6 and didn’t get good results—not close to the efficiency of the redundant scheduler on 5.4. But even so, the current version is excellent; congratulations on the work! If new versions for 5.4 are discontinued, will support for the last official version still be maintained?

Ysurac, 2024-10-31:
No, there is also no real support for 5.4 now.
I think this will be discontinued really when OpenMPTCProuter will switch to 6.12 kernel. 
It's begin to be difficult to have same code base (openmptcprouter feed) for both kernel and this will be too time consuming. MPTCP out-of-tree used for kernel 5.4 is also deprecated.

vempire-ghost, 2024-10-31:
> It's now fixed, but I'm not sure I will keep 5.4 for v0.62...

Tried to install now and still have the same error.
Executando opkg update...
Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/packages/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/packages/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/packages/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/base/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/packages/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/packages/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/base/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/base/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/routing/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/packages/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/packages/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/base/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/base/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/routing/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/routing/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/telephony/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/packages/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/packages/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/base/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/base/Packages.gz

Downloading https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/routing/Packages.gz

*** Failed to download the package list from https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/routing/Packages.gz

[... 167 lines cut ...]

*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/openmptcprouter/Packages.gz

Downloading https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/packages/Packages.gz

*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/packages/x86_64/packages/Packages.gz

Collected errors:

 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/luci/Packages.gz, wget returned 8.

 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/packages/Packages.gz, wget returned 8.

 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/base/Packages.gz, wget returned 8.

 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/routing/Packages.gz, wget returned 8.

 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.62-snapshot-5.4/x86_64/telephony/Packages.gz, wget returned 8.

 * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-snapshot-5.4/x86_64/targets/x86/64/packages/Packages.gz, wget returneInstalando o MTR...
Unknown package 'mtr'.

Collected errors:

 * opkg_install_cmd: Cannot install package mtr.

Ysurac, 2024-11-01:
Realy fixed this time.

closed 2025-02-05 by github-actions[bot]