Ysurac/openmptcprouter #3629: RPI4 v0.61 after reboot readonly

mike-vv, 2024-10-31:
After rebooting the device, the memory goes into read-only mode.
The problem exists only in version 0.61. 
Regardless of whether it is a clean installation or an update.
The file system is ext4.

The problem is not observed in versions 0.59 and 0.60.

I had to go back to version 0.60.

## Specifications

  - OpenMPTCProuter version: 0.61
  - OpenMPTCProuter VPS version: 0.1031 6.6.36-x64v4-xanmod1
  - OpenMPTCProuter VPS provider: space web
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.4 (64-bits)
  - Country: Russia

a834003831, 2024-11-01:
"I’m encountering this issue as well. When I insert the USB network card and reboot, this happens. Without the network card inserted, everything works normally. I'm on version 0.60 with the 6.1 kernel, and this issue persists."

muko82, 2024-11-01:
It's not 0.60 nor 0.61
The problem on rpi4 is kernel 6.6 which had  issue on usb power and now still continue having this problem on the filesystem

fareign, 2024-11-01:
> It's not 0.60 nor 0.61 The problem on rpi4 is kernel 6.6 which had issue on usb power and now still continue having this problem on the filesystem

yes I have the same problem.but it only appear on uSB 3.0 device.

muko82, 2024-11-01:
Yesterday was released the snapshot image for 0.62 which solve this issue. It's still open the one which put the filesystem in read-only after a reboot

fareign, 2024-11-02:
> Yesterday was released the snapshot image for 0.62 which solve this issue. It's still open the one which put the filesystem in read-only after a reboot

I have go back 6.1 kernel.

a834003831, 2024-11-02:
> Yesterday was released the snapshot image for 0.62 which solve this issue. It's still open the one which put the filesystem in read-only after a reboot

The 6.1 kernel doesn't have redundancy options, so I've rolled back to 5.4.

muko82, 2024-11-02:
> > Yesterday was released the snapshot image for 0.62 which solve this issue. It's still open the one which put the filesystem in read-only after a reboot
> 
> The 6.1 kernel doesn't have redundancy options, so I've rolled back to 5.4.

Me too. Just waiting for the bug fix on 6.6
In the meatime yesterday I used for the first time ever OMR on RPI4 in mobility for a real use case and it worked perfectly with kernel 5.4

On usb dongle key and one old mobile phone on the other port. :-D
More than enough for the moment

a834003831, 2024-11-02:
> Me too. Just waiting for the bug fix on 6.6 In the meatime yesterday I used for the first time ever OMR on RPI4 in mobility for a real use case and it worked perfectly with kernel 5.4
>
> On usb dongle [...]

It seems there are some issues with the current compilation; I can only use SH scripts.

Ysurac, 2024-11-03:
For Read-Only issue, did you try image with SquashFS ?

muko82, 2024-11-03:
I'm using ext4.

a834003831, 2024-11-03:
> For Read-Only issue, did you try image with SquashFS ?

I have tried it, R4S, but after flashing it in, it cannot start. There may be a firmware problem, both 6.1 and 6.6 are not working

Ysurac, 2024-11-03:
@a834003831 The subject here is RPI4.

muko82, 2024-11-07:
@Ysurac here I'm. I already answered here to your questions. No more power issues with usb but it remains the problem of read-only file system during a normal reboot 

I'm using ext4

Ysurac, 2024-11-07:
@muko82 So, test SquashFS image please.
There is also a full RPI patch update in latest snapshot.

muko82, 2024-11-08:
Quick update: I'm testing the snaphost you uploaded on November 6.

0.62-r0 factory kernel 6.6 ext4: exactly the same issue. It works until reboot. Than I get "filesystem is in read-only"
0.62-r0 factory kernel 6.6 squashFS: it seems that it's working fine even after reboot. It keeps settings and starts as expected. File system is fully writeable even after reboot.

Now... What are the differences between ext4 and squashFS if compared to OMR? What could be the issue with the ext4 version?

Ysurac, 2024-11-08:
How do you reboot ? If you reboot with removing power, this can (will) corrupt ext4 filesystem.
SquashFS is a Read Only filesystem used for root initial partition (this allow to do restore to factory default) and UBIFS is used for all writed data. It's more robust to power cuts.

muko82, 2024-11-08:
I'm rebooting from cli or web UI. Not certainly removing power.
After doing this the filesystem goes in read-only: if I try with fsck.ext4 I get no errors in the filesystem and the message on the web ui remains.
With kernel 5.4 if I remove power I get the message but fsck fixes some errors. After that the message on the web ui disappears.

muko82, 2024-11-08:
It also seems that with kernel 6.6 (squashFS) I no longer have issues with FTP file transfers. Sounds good... :)

Now: what are the effective benefits in using squashFS instead ext4 (excluding the initial root partition in r/o)?

Are the information useful to fix the issue on the ext4 release?

muko82, 2024-11-08:
UPDATE: while the the download speed is quite good, the upload si around 2Mbit instead of 40/50...

muko82, 2024-11-09:
@Ysurac I see in the snaphosts repo a new image uploaded yesterday. What are the changes? Thanks.

bobradamus, 2024-11-10:
> How do you reboot ? If you reboot with removing power, this can (will) corrupt ext4 filesystem. SquashFS is a Read Only filesystem used for root initial partition (this allow to do restore to factory default) and UBIFS is used for all writed data. It's more robust to power cuts.

honestly - the time I've wasted with this. Had I known, I would've used SquashFS all along...

jpansy, 2024-12-19:
I have the same problem since every build with 0.61 on EXT4, even with the latest 0.62 snapshot has it. The problem does not exist on first boot after sysupgrade, but after every next reboot (through "reboot" command in webui or terminal). `sudo mount -o remount,rw /` fixes it until the next reboot.

I tried to change the scaling governor, removed all usb network devices, but nothing helped. Moved to squashFS build and now the problem seems gone.

fbarrera-fbaudio, 2025-03-25:
I was also having the same problem. I switched to an RPI 5 and managed to find out which partition was in read-only mode (mount command) and deactivated the problematic partition through the OpenMPTCP interface.
Does anyone know what repercussions could come from not having this partition active? Now I reboot and shut down as often as I want, and the read-only error is gone.
Español: 
También tenía el mismo problema. Cambié a una RPI 5 y Logré descubrir qué partición era la que estaba en modo lectura (comando mount) y a través de la interfaz de openmptcp desactivé la partición con problemas.
Alguien sabe que repercuciones podría traer el no tener esta partición activa?. Ahora reinicio y apago cuantas veces quiera y el error de solo lectura desapareció.