Ysurac/openmptcprouter #3630: MPtcpRouter doesn't aggragate wan

orgnemo, 2024-11-02:
**Expected behavior:**
wan1 fibra 12Mbs/1.5Mbs master
wan2 LTE 25Mps/20Mps

Speed test 30Mps/20Mps

**What happens instead of the expected behavior:**
wan1 fibra 12Mbs/1.5Mbs master
wan2 LTE 25Mps/20Mps

Speed test 12Mps/1.5Mps
Only master wan is in function

I found this in log:

`Change in routes, set ip route replace default scope global nexthop via 192.168.0.1 dev wan1 weight 100 nexthop via 192.168.2.1 dev wan2 weight 1 (omrvpn_intf: tun0)
`

On proxmox whitch is image consiglible?
## Specifications

  - OpenMPTCProuter version: 0.61
  - OpenMPTCProuter VPS version: 0.1031
  - OpenMPTCProuter VPS provider: IONOS
  - OpenMPTCProuter platform: Proxmox

Ysurac, 2024-11-03:
Can you take a screenshot of OpenMPTCProuter Status page ?
I would also need the result, via SSH on the router, of ```omr-test-speed wan1```, ```omr-test-speed wan2``` and ```omr-test-speed```.

orgnemo, 2024-11-05:
Wan1 test download:
`host: speedtest.milkywan.fr - ping: 41
Best server is http://par.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 60.0M    0     0  1528k      0  1:46:30  0:00:40  1:45:50 1556k`

Wan2 test download:
`host: speedtest.milkywan.fr - ping: 94
Best server is http://speedtest.frankfurt.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 56.5M    0 56.5M    0     0  1187k      0 --:--:--  0:00:48 --:--:--  508k`

Wan1 and Wan2 test download:
`host: speedtest.milkywan.fr - ping: 115
Best server is http://par.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 68.3M    0     0   968k      0  2:48:02  0:01:12  2:46:50  193k`

[image]

closed 2025-02-08 by github-actions[bot]