Ysurac/openmptcprouter #3632: Prerequisite check failed while building

KaWiSo, 2024-11-02:
I am using openmptcprouter on a currently unsupported platform, the glinet mt6000.
Since it is not currently supported I have to manually compile an image every time I want to upgrade. 
The Router is currently on a version of 0.61rc2 compiled on the 28.09. and i wanted to update to the most recent rc and switch to Kernel 6.6. 

Even though I am using the same VM, which I only use for compiling this image, the compile fails with checking prerequisites.
I am unable to compile for either kernel versions and the following prerequisites for compiling are now failing:

Checking 'working-gcc'... failed.
Checking 'working-g++'... failed.
Checking 'ncurses.h'... failed.
Checking 'realpath'... failed.
Checking 'argp.h'... failed.
Checking 'fts.h'... failed.
Checking 'obstack.h'... failed.
Checking 'libintl.h'... failed.
Prerequisite check failed. Use FORCE=1 to override.

What do I have to install for these to work?

a834003831, 2024-11-03:
[image]
Hello, I would like to ask you how to modify the device tree file when compiling for ARM devices. I’m stuck at this step and haven’t been able to find the folder, not even the CD step.

KaWiSo, 2024-11-04:
> ![image](https://private-user-images.githubusercontent.com/96239138/382560172-d9a13218-d9e4-4f2f-bbad-d20f0ce3ce3d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

If the folder does not exist, you did not start building.
You have to start building it first and if there is no config file for your build platform, you then have to go do these steps.

a834003831, 2024-11-04:
> If the folder does not exist, you did not start building. You have to start building it first and if there is no config file for your build platform, you then have to go do these steps.

You need to compile it once using a sh script first, but I need to adapt it for the new machine.

closed 2024-11-23 by KaWiSo