Ysurac/openmptcprouter #3634: Advice needed. Safest way to upgrade router to newer version?

peterxxl, 2024-11-03:
## Expected Behavior
Update router safely to latest version. What's the best way?

## Current Behavior
I have been running v0.60beta1-6.1 r0+23789-ce6ad123e7 on the router for 1 year and half without major issue
VPS has been running OpenMPTCProuter Server 0.1028

I see there have been many new releases so I guess it's time to update the software. 
I have already upgraded the VPS to 0.1031 via the command and it seems to be working fine. 

Now how do I update remotely the router?
I don't want to break anything as the router is not easily accesible and would rather not have to physically visit.

It's x86_64 machine but I no longer remember if the original image was ext4 or squashfs and UEFI or not? Where could this be found? 

Is sysugrade menu the best way to do this? The proposed version doesn't match the latest version in downloads section.
[image]

or rather this one? as mentioned, for this one I don't remember which image I have used originally
[image]

## Specifications

  - OpenMPTCProuter version: v0.60beta1-6.1 r0+23789-ce6ad123e7
  - OpenMPTCProuter VPS version: 0.1031
  - OpenMPTCProuter VPS provider: Grupo ZGH
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-11-03:
It seems I forgot to update the version for sysupgrade menu as we have v0.61 now. I will purge CDN cache.
You can update via flash image menu, it's using same program to update in fact.
To check filesystem, via SSH on the router, ```mount | awk 'NR==1{print $5}}'``` and for efi ```ls /sys/firmware/efi``` return some data if it's EFI.

peterxxl, 2024-11-04:
Hi and thanks for reply... based on your commands, I have determined it's ext4 with EFI.

Unfortunately first & second try didn't work. 
When using `Flash new firmware image` and selecting `openmptcprouter-v0.61-6.6-r0+27346-c7ba5574f5-x86-64-generic-ext4-combined-efi.img.gz` it got stuck on  30.51%
[image]

on second try I used `Sysupgrade` option and got this
[image]

peterxxl, 2024-11-04:
3rd time got stuck on 41.38%
[image]

peterxxl, 2024-11-04:
one more question... the image should be uploaded as downloaded `img.gz` file or decompressed as `img` only?
the reason for asking is that the upload is painfully slow, especially when uploading decompressed (around 500MB)
I am doing this from machine on the same LAN via remote desktop, so I guess it should be fast

[image]

Ysurac, 2024-11-04:
The image should be uploaded to the router as .img.gz. No need to decompress it.
If upload is slow then you have a LAN network issue I think. it' a VM ?

peterxxl, 2024-11-04:
nope, physical machine bought from Amazon...and  it's connected to LAN via 2.5GbE, so it's really weird why it would take so long... will investigate further.. thanks

peterxxl, 2024-11-04:
is there a command to preform the same action `Flash new firmware image` via SSH if I download the `.img.gz` file directly to router via `wget`?

peterxxl, 2024-11-04:
so I did `sysupgrade -v openmptcprouter-v0.61-6.6-r0+27346-c7ba5574f5-x86-64-generic-ext4-combined-efi.img` and seems to have worked... the router booted up after a minute or so with same configuration 

now, I will check if everything works.... nice new UI btw. awesome. thanks

closed 2025-02-07 by github-actions[bot]

jihadismail8, 2025-07-23:
[image]
this does not make sense !! , the version i have is higher then the one available , its an upgrade or downgrade ?