Ysurac/openmptcprouter #3635: OMR completely useless if Master WAN is down!

ccmks, 2024-11-03:
## Expected Behavior
If master WAN is down, the OMR should continue working and aggregating the remaining active WAN

## Current Behavior
Not doing what expected, in fact, I had to change the master WAN to different WAN in order to work!

## Specifications

  - OpenMPTCProuter version:  0.61-6.6
  - OpenMPTCProuter VPS version: Version 0.1031 6.6.36-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: 
  - OpenMPTCProuter platform: x86_64 Hyper-V EFI

Ysurac, 2024-11-03:
It should not. Master interface is only for default gateway but all connection are set in load balancing mode as gateway. OMR-Tracker remove the connections that are not working.
Check in Status->System log what omr-tracker is doing.

Ysurac, 2024-11-03:
There is options in Services->OMR-Tracker Manager to check link quality.

darkman1983, 2024-11-05:
I can confirm, that this behaviour is affecting me too.
If WAN1 is down and/or not responding, internet is down...

fareign, 2024-11-07:
I think there is some problem on the server

[image]
this picture show ,those master is down ping is ok.but up stream gones up and down.
[image]

and this picture show, the interface stream seems normal.

in this case , we should stop stream and restart srtream.

Ysurac, 2024-11-07:
Can you update VPS to kernel 6.10 ? https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

fareign, 2024-11-08:
> Can you update VPS to kernel 6.10 ? https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

now use is 6.1,I test 6.6,seem the same .I will test 6.10 later

vempire-ghost, 2024-11-26:
> Today, my primary (master) ISP got suspended due to overdue billing, which still make the OMR completely stop working until I have to switch to other ISP as the master. I believe there is bug on this version

The OMR has serious issues identifying whether a WAN is online or offline. Since version 0.59 and even in the current snapshot, most of the time it simply does not remove an offline WAN, which renders it completely inoperable if this WAN is the MASTER.  
If you need to resolve this issue, the simplest solution is to place the MASTER WAN behind a load-balancing hardware device that performs this function, which the OMR is unable to handle.

Ysurac, 2024-11-26:
@vempire-ghost not really constructive...
I would need logs from Status->System log.
Detection is made by OMR-Tracker, you can change settings in Services->OMR-Tracker Manager
(Current snapshot should work better to remove offline WAN)

closed 2025-03-03 by github-actions[bot]