Ysurac/openmptcprouter #3636: Need help banana pi RPI-R3

esquilo444, 2024-11-03:
Good morning, 

I know that the topic has already been discussed but my limited knowledge is blocking me. I just purchased a banana pi-r3. While browsing the forum I saw that several people were looking to compile open mptcp router to adapt it to this hardware. Some have succeeded, a priori. How I don't know how to do it and especially correct the errors that will most likely occur, could someone from the community who successfully completed this compilation share it with me. Usually I like to do things on my own, but this time it's outside of my skills.

 Many thanks in advance.. 

Jerome

Ysurac, 2024-11-09:
You can try images in https://snapshots.openmptcprouter.com/6.6/

esquilo444, 2024-11-09:
Bonjour Yannick,

Merci beaucoup pour le temps que tu consacres .
J'ai reçu mon BPI-R3 hier et j'ai testé dans la foulé la version 6.1 de BlueWaterCrystal, qui dans mon usage spécifique fonctionne plutôt bien ... Test de flux SRT sur 15 minutes pour le moment, le temps de comprendre le fonctionnement spécifique du BPI-R3, j'aurais plus de temps la semaine prochaine...
Je teste ta version 6.6 dès lundi .. et je te fais un retour dans la foulé aussi.

A très bientôt.

Jérôme de Nantes

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

esquilo444, 2024-11-22:
bonjour,

J'ai enfin eu un peu de temps .
Malheureusement, le lancement de l'image bloque au premier démarrage, lors de la création de partitions ...
Ci-joint une photo .

Dans l'attente de votre retour.

jerome

esquilo444, 2024-11-23:
[image]

esquilo444, 2024-11-24:
Autre interrogation, Est-ce normal que dans ta version je vois apparaitre Mediatek MT7530 et non pas MT7986 qui est celui du BPI-R3 .
Voilou.
Je n'ai plus de questions pour aujourd'hui .
A très vite .

Ysurac, 2024-12-02:
You can try latest snapshot, I fixed some drivers, but not sure this is working or not yet...

gysutantoman, 2024-12-03:
stuck after [   11.352517] random: crng init done

openmptcprouter-v0.62-snapshot-6.6-r0+28016-48028cd102-mediatek-filogic-bananapi_bpi-r3-sdcard.img

[openmptcprouter-v0.62-snapshot-6.6-r0+28016-48028cd102-dmesg-bpi-r3.txt](https://github.com/user-attachments/files/17987112/openmptcprouter-v0.62-snapshot-6.6-r0%2B28016-48028cd102-dmesg-bpi-r3.txt)

Ysurac, 2024-12-03:
Seems normal for me. If you press enter you should see login prompt.

esquilo444, 2024-12-03:
in my case no, when I press enter nothing happens... it just goes to the line, but it stays black...

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

gysutantoman, 2024-12-04:
> Seems normal for me. If you press enter you should see login prompt.

press enter stays black.

esquilo444, 2024-12-04:
no better, always black

esquilo444, 2024-12-18:
hello Yannick, I'm still stuck at the same step... even with the latest successive versions. In a previous message you say that at the end of the launch, you can log in, which means that openmtcprouter is launched... unfortunately for me this is not the case... How do you test the BPI-R3 image? Maybe there is something I'm doing wrong... For version 6.1 compiled by blues Cristal, it works... not 100 percent but, it launches... Waiting for your feedback. thanks in advance. Jérôme

esquilo444, 2025-01-27:
Hello,
Thanks for last version .
It work for me ...

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

Can you test latest snapshot ?

closed 2025-05-02 by github-actions[bot]