Ysurac/openmptcprouter #3637: NanoPiR2s Edge X 0.61 not working

Picicorp, 2024-11-03:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Picicorp, 2024-11-03:
Nano Pi R2s 0.61 Firmware not working, no aggregation no vos server connextion token key missing. Tried with Edge X same problem it wont even show server status. So which hardware is rhe best for testing?

Ysurac, 2024-11-04:
So R2S is booting correctly ?
You configured it correctly with the IP and key of your VPS server ?

Picicorp, 2024-11-04:
NanoPi R2s with 0.61 6.6 wont show lan and does not connect to lan 192.168.100.1. Edge X 0.61 6.6 boots coorectly, assigns lan 192.168.100.1 I can connect to gui but when setting are applied, keys and ips of wan, the status page wont work.

Alvaro Panesso
Correo | ***@***.***
Web | Pici.com.co
Insta | Picicorp
Whtasapp | +573108490885

Picicorp, 2024-11-04:
Just Bricked Edge X, Openmptcprouter 0.61 66 not booted but didnot do anything So I donwloaded mar 28 firmware 0.60 61 flashed that firmware and now there is no ip (192.168.100.1), do the reset all leds blink, no ip and tried the console and via putty it says Hi Hi Hi Hi Hi. How can I unbrick the edge X to try to make it work with MPR
[image]
[image]

Ysurac, 2024-11-05:
I don't understand, it's about NanoPi R2S or about Ubiquiti EdgeRouter X  or both ?
For debricking information on Ubiquiti EdgeRouter X: https://openwrt.org/toh/ubiquiti/edgerouter_x_er-x_ka

Picicorp, 2024-11-05:
I´m workinmg with both. The Edge is bricked will continue with NanoPi R2s.

Ysurac, 2024-11-05:
For R2S, I think it's a network issue.
Can you give me the result, via SSH, of ```uci show network``` and ```ip a``` on latest working version.
And same command from https://downloads.openwrt.org/snapshots/targets/rockchip/armv8/ (if it's working)