Ysurac/openmptcprouter #3638: Lost external access going from 5.4 to 6.6

patimages, 2024-11-03:
## Expected Behavior
Ability to access my network remotely using port forwarding. It was working perfectly on 5.4. 

## Current Behavior
Does not work

## Possible Solution
Not sure it if help but I see a warning : "Legacy rules detected"... No idea what that means.

## Steps to Reproduce the Problem

go from 5.4 to 6.6. 

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61-6.6 r0+27346-c7ba5574f
  - OpenMPTCProuter VPS version: 6.6.36-x64v3-xanmod1 0.1031
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64
  - Country: FR

patimages, 2024-11-03:
Digging deeper : The external access works when being on a different network. 

- the IP of my server is 123.123.123.123 
- if I try to connect internally, all works well
- if I try to connect my server at 123.123.123.123:port -> Failure
- if I go on a VPN and try to access my server 123.123.123.123:port, all works well

Any idea why I cannot access remotely from my normal IP ? It is useful to me at times and used to work on 5.4.  

Sill, concerning the warning : "Legacy rules detected"... No idea what that means and whether I should do anything.

Thanks again !

darkman1983, 2024-11-05:
I recommend you to do a fresh install. Upgrading major kernel-versions or openwrt, could lead to compatibility problems.
OMR it self always made problems for me, when upgrading release-versions, that are to far away from each other.
Also iptables was upgraded to nftables, that could make problems with old settings too.

closed 2025-02-08 by github-actions[bot]