Ysurac/openmptcprouter #3640: OpenMPTCProuter Server 0.1031 VPS installation script + Debian 12 IPv6 disabled leads to "Can't get public IP address from ShadowSocks Rust"

BigFanOpenmptcprouter, 2024-11-06:
## Expected Behavior
When I go to https://192.168.100.1/cgi-bin/luci/admin/system/openmptcprouter/status
I should see:
    Version 0.1031 6.1.0-23-cloud-amd64
    Load: 0.XX 0.XX 0.XX
    Uptime: XXd XXh XXm XXs
    Proxy traffic: XX.XX GiB
    VPN traffic: XX.XX MiB
    Total traffic: XX.XX GiB

## Current Behavior
When I go to https://192.168.100.1/cgi-bin/luci/admin/system/openmptcprouter/status
I see "**Can't get public IP address from ShadowSocks Rust**"

In Status->System log I see:
> Nov  6 16:04:02 ompr user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Nov  6 16:04:02 ompr user.notice omr-bypass: OMR-ByPass is running
**Nov  6 16:04:15 ompr daemon.info omr-tracker-ss: Shadowsocks sss0 is down (can't contact via http 212.27.48.10, 198.27.92.1, 151.101.129.164)**
Nov  6 16:04:16 ompr user.notice firewall.omr-server: Firewall reload, set server part firewall reloading

**Important: The connections works just fine. Only the test fails.** Tested on 2 different Linode VPS servers in two very different IP ranges (different /8 subnets) as I assumed the VPS IPv4 is somehow blacklisted.

## Possible Solution
Workarounds:
A) Do not (even temporarily) disable the IPv6 on the VPS side. The snapshot might solve this: https://github.com/Ysurac/openmptcprouter/issues/3624 but I didn't test it yet.
B) Use a VPS script running older version (OpenMPTCProuter VPS version: 0.1030 works fine when IPv6 is disabled).

## Steps to Reproduce the Problem

  1. Do clean Debian 12 install & get the script: wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL="6.6" sh
  2. Reboot
  3. Test the connection: https://192.168.100.1/cgi-bin/luci/admin/system/openmptcprouter/status
  4. All fine for now
  5. Disable IPv6:
nano /etc/sysctl.conf
net.ipv6.conf.all.disable_ipv6=1
net.ipv6.conf.default.disable_ipv6=1
net.ipv6.conf.lo.disable_ipv6=1
net.ipv6.conf.eth0.disable_ipv6=1
and then do:
  6. sysctl -p
  7. reboot VPS
  8. Test the connection gain: https://192.168.100.1/cgi-bin/luci/admin/system/openmptcprouter/status
  9. The "Can't get public IP address from ShadowSocks Rust" shows up
  10. Note: **If you revert and re-enable the IPv6 it won't fix the problem and the test still fails (although the actual connection works fine)**. You need to do a clean re-install of the VPS.

## Context (Environment)
Debian 12 x64 with IPv6 disabled. Re-enabling it later won't solve the issue. The only solution is clean reinstall of the VPS.
**I did not upgrade** my router to 0.61-6.1 just yet...

## Specifications

  - OpenMPTCProuter version: OpenMPTCProuter version 0.60-6.1
  - OpenMPTCProuter VPS version: 0.1031
  - OpenMPTCProuter VPS provider: Linode
  - OpenMPTCProuter platform: x86_64
  - Country: No restrictions.

Ysurac, 2024-11-07:
Why enabling and then disabling IPv6 ?
Test VPS snapshot script.
If still problem, I would need ```journalctl -u shadowsocks-go```

BigFanOpenmptcprouter, 2024-11-12:
Probably related to issue [#3624](https://github.com/Ysurac/openmptcprouter/issues/3624)

Update: I can confirm that the snapshot from today "OpenMPTCProuter Server 0.1032-test" **solved the issue**.
On router side I used "openmptcprouter-v0.61-6.6-r0"

The IPv6 can be disabled on the VPS side and the "Can't get public IP address from ShadowSocks Rust" **doesn't show any more**.

closed 2025-02-15 by github-actions[bot]