Ysurac/openmptcprouter #3642: 0.61 5.4 on RPI4 - Shadowsocks vs XRAY vs V2RAY

muko82, 2024-11-07:
## Current Behavior
I tried all the three proxy with variants (VLESS, Socks, etc). In all cases the connection works pretty well. Apparently.

V2RAY/XRAY (with the variants VLESS, socks, etc):
I discovered that while using FTP to upload some file on server the connection is closing the transfers without errors but files, while managed with a python script (e.g. images) are resulting corrupted in the last few bytes (from the browser or some image editor I have no issue)

Shadowsocks: no issues

In addition:when I use XRAY/V2RAY, it seems that the system manages the available bandwidth much better than with Shadowsocks.
Can this be related to the missing support of AES on RPI4? Am I doing something wrong in the configuration (see below)?

In conclusion: with Shadowsocks I get no error with FTP and python scripts but  bandwidth management seems to be worse than with XRAY/V2RAY.

## Specifications

  - OpenMPTCProuter version: 0.61 - Kernel 5.4
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1301
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: RPI4

Proxy: XRAY/V2RAY (VLESS, SOCKS) or Shadowsocks
VPN: Glorytun TCP
Connections: 2 4G dongle with SQM and SQM Autorate active (One of the two connections is under WG VPN because the carrier filters MPTCP)
Scheduler: ECF with BBR
Syn retries: 3

darkman1983, 2024-11-07:
My recommendation is:
Don't use 5.4 Kernel, as it is legacy!!
So you won't get support for that. New Stable 0.61 has kernel 6.6 as default?!
And before you get problems:
I would also recommend flashing 6.6 kernel with a complete new install, but thats only my personal opinion.

If you problem persists with 6.6 kernel, report back here, so Ysurac can give you support.

muko82, 2024-11-07:
6.6 on rpi4 Is a complete mess. I had issues with usb power and read-only file system.

6.1 for 0.61 seems that was abandoned and I had issues while trying to compile myself.

Currently 0.61 with kernel 5.4 is the one with less problems (stable?)

Ysurac, 2024-11-07:
No more support on kernel 5.4
For RPI4 problems on 6.6, answer to the current https://github.com/Ysurac/openmptcprouter/issues/3629 issue.

closed 2024-11-07 by Ysurac