Ysurac/openmptcprouter #3644: Strict DNS order changes unexpectedly

datapharmer, 2024-11-08:
## Expected Behavior

When strict order is selected under dns settings I'd expect it to follow the order of servers listed

## Current Behavior

The order of the dns servers keeps getting changed around so the ipv6 dns resolver is first and pihole running on vps is last

## Possible Solution

No idea... keeps getting reordered!

## Steps to Reproduce the Problem

  1. Set pi hole on vps as first dns as list
  2. add additional ipv3 and ipv6 dns servers as entries 2 and 3 (as fallback in case vps is down)
  3. set ordering to strict
  4. save and apply
  5. wait a couple hours to a few days and find that the order has changed so pihole is now last in the list (maybe due to vps going down?

## Context (Environment)

## Specifications

  - OpenMPTCProuter version:  .61-6.6
  - OpenMPTCProuter VPS version:  .1031-test 6.11-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: buyvm
  - OpenMPTCProuter platform:  rpi4
  - Country:  usa

Ysurac, 2024-11-08:
When VPN between router and VPS is down, PiHole DNS IP is removed to keep a working DNS and added back when VPN is back. Order is not respected in this case.

kevinh-csalabs, 2024-11-08:
How can it be given priority when it gets back then? Domains are not being blocked properly because of this.

Ysurac, 2024-11-12:
You can set DNS you want in Services->Recursive DNS that is used when PiHole is down and use root DNS by default but can be configured to use anything you want.

datapharmer, 2024-11-12:
Perfect thanks!