Ysurac/openmptcprouter #3645: How to force all outgoing traffic to one WAN? Can't get it working with omr-bypass using IP, MAC etc.

ioogithub, 2024-11-08:
## Expected Behavior
I would like to temporarily force all traffic to go out though one of the WAN connections. I need to maintain the OMR<-VPS link for incoming traffic so disabling the server under `Settings Wizard` will not work for this solution.

Network is like this:

`Clients <-> Mesh router <->OMR <->VPS`

All traffic from clients on the network go though the mesh router then to OMR. So OMR sees all traffic coming from Mesh router.

In OMR-bypass I tried:
1. Adding the IP address of the Mesh router (on the OMR subnet) to 'Source lan IP address or network' and selecting WAN1. (did not work)
2. Adding the IP address of the Mesh router (on the OMR subnet) to 'IPs and Networks' (also did not work)
3. Adding the MAC address of the Mesh router to 'MAC-Address' (also did not work).

Is omr-bypass the correct way to accomplish this task? Which OMR-bypass section is correct to get this working? If not how else to do the job.

Note: omr-bypass is working for websites added under omr-bypass `Domains`.

## Specifications
  - OpenMPTCProuter version: [Powered by openmptcprouter v0.61rc2-6.6 r0+27346-c7ba5574f5](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter platform: x86_64

ioogithub, 2024-11-08:
Update, I think it might be working, the issue is that after I click Save and Apply on the omr-bypass page, it takes several minutes to reload the omr- bypass rules and I get a lot of errors:

This is what I see from the log, is this normal or a problem:

```
Nov  8 23:41:00 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov  8 23:41:24 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Nov  8 23:41:24 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Nov  8 23:41:24 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Nov  8 23:41:24 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:24 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:25 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Nov  8 23:41:25 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Nov  8 23:41:25 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Nov  8 23:41:25 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:25 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:25 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:25 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:25 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:26 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:26 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:26 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:26 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:26 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:26 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:28 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:28 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:28 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:28 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:28 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:28 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Nov  8 23:41:28 OpenMPTCProuter daemon.notice netifd: wan2 (5314): udhcpc: sending renew to server 100.64.0.1
[... cut ...]
Nov  8 23:41:52 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Nov  8 23:41:52 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Nov  8 23:41:52 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
```

Also should it take a few minutes (maybe 3 or 4) after adding a new bypass rule? OMR is running on a likely (over) powered dedicated x86 box.

Ysurac, 2024-11-11:
It can take a long time to start omr-bypass if you bypass some big services/ASN/...

jjak0b, 2024-12-07:
I have same issue after upgrade usind clean and stable x86_64 (openmptcprouter v0.61-6.6 r0+27346-c7ba5574f5) image.
i'm not using VPS so i'm using just OMR-bypass to use a wan1 for some devices and wan2 for others/master.

I got a lot of error 
`OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory`
and i'm experieing also #2156 (but i dont think last one is correlated).

on the clean installation i applied the configuration backuped from old 0.59.1 but i dont think this may be the cause if other people have this issue.