Ysurac/openmptcprouter #3647: Error building image for Orange Pi CM4

eXulW0lFy, 2024-11-10:
## Expected Behavior

## Current Behavior

I tried to build image for OPi cm4, but I got an error: 

``` bash
SHELL= flock /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/tmp/.root-copy.flock -c 'cp -fpR /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3/.pkgdir/simple-obfs/. /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.simple-obfs_installed
removed '/home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/bin/packages/aarch64_generic/openmptcprouter/simple-obfs_486bebd9-3_aarch64_generic.ipk'
mkdir -p /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/bin/targets/rockchip/armv8/packages /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3/ipkg-aarch64_generic/simple-obfs/CONTROL /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/staging_dir/target-aarch64_generic_musl/pkginfo
install -d -m0755 /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3/ipkg-aarch64_generic/simple-obfs/usr/bin
install -m0755 /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3/src/obfs-local /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3/ipkg-aarch64_generic/simple-obfs/usr/bin
find /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3/ipkg-aarch64_generic/simple-obfs -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="aarch64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="aarch64-openwrt-linux-musl-nm" STRIP="/home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/scripts/strip-kmod.sh" PATCHELF="/home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/staging_dir/host/bin/patchelf" /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/scripts/rstrip.sh /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3/ipkg-aarch64_generic/simple-obfs
rstrip.sh: /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3/ipkg-aarch64_generic/simple-obfs/usr/bin/obfs-local: executable
(cd /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3/ipkg-aarch64_generic/simple-obfs/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/bin/packages/aarch64_generic/openmptcprouter
/home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/staging_dir/host/bin/fakeroot /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/staging_dir/host/bin/bash /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/scripts/ipkg-build -m "" /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3/ipkg-aarch64_generic/simple-obfs /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/bin/packages/aarch64_generic/openmptcprouter
Packaged contents of /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3/ipkg-aarch64_generic/simple-obfs into /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/bin/packages/aarch64_generic/openmptcprouter/simple-obfs_486bebd9-3_aarch64_generic.ipk
echo "simple-obfs" >> /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/staging_dir/target-aarch64_generic_musl/pkginfo/simple-obfs.default.install
touch -r /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3/.built /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3/.autoremove 2>/dev/null >/dev/null
find /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/build_dir/target-aarch64_generic_musl/simple-obfs/simple-obfs-486bebd9-3 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -and -not -name '.pkgdir'  -print0 | xargs -r -0 rm -rf
make[3]: Leaving directory '/home/orangepi/sda1/openmptcprouter/feeds/openmptcprouter/simple-obfs'
time: package/feeds/openmptcprouter/simple-obfs/compile#309.57#20.25#349.78
make[2]: Leaving directory '/home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source'
make[1]: *** [package/Makefile:122: /home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source'
make: *** [/home/orangepi/sda1/openmptcprouter/opi-cm4/6.1/source/include/toplevel.mk:232: world] Error 2
```

I tried `j=4`, `j=4 V=s`, `j=1 V=s` and still got an error. It was about different package, but I managed to correctly capture log message only on my last try.  

Here's last 512K of the log at pastebin: https://pastebin.com/rwX59BrM

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 

Options:
``` plain
OMR_TARGET="myplatform" OMR_FEED_SRC="master" OMR_KERNEL="6.1"
```

OS: Debian 12 aarch64 (arm64).

Ysurac, 2024-11-15:
Try to build using kernel 6.6

eXulW0lFy, 2024-11-19:
When I ran ./build.sh it shows multiple errors:

```plain
ERROR: please fix feeds/packages/admin/qbee-agent/Makefile - see logs/feeds/packages/admin/qbee-agent/dump.txt for details
ERROR: please fix feeds/packages/devel/delve/Makefile - see logs/feeds/packages/devel/delve/dump.txt for details
ERROR: please fix feeds/packages/devel/gitlab-runner/Makefile - see logs/feeds/packages/devel/gitlab-runner/dump.txt for details
ERROR: please fix feeds/packages/libs/boringssl/Makefile - see logs/feeds/packages/libs/boringssl/dump.txt for details
ERROR: please fix feeds/packages/net/adguardhome/Makefile - see logs/feeds/packages/net/adguardhome/dump.txt for details
ERROR: please fix feeds/packages/net/alist/Makefile - see logs/feeds/packages/net/alist/dump.txt for details
ERROR: please fix feeds/packages/net/cloudflared/Makefile - see logs/feeds/packages/net/cloudflared/dump.txt for details
ERROR: please fix feeds/packages/net/cloudreve/Makefile - see logs/feeds/packages/net/cloudreve/dump.txt for details
ERROR: please fix feeds/packages/net/cni-route-override/Makefile - see logs/feeds/packages/net/cni-route-override/dump.txt for details
ERROR: please fix feeds/packages/net/croc/Makefile - see logs/feeds/packages/net/croc/dump.txt for details
ERROR: please fix feeds/packages/net/crowdsec/Makefile - see logs/feeds/packages/net/crowdsec/dump.txt for details
ERROR: please fix feeds/packages/net/crowdsec-firewall-bouncer/Makefile - see logs/feeds/packages/net/crowdsec-firewall-bouncer/dump.txt for details
ERROR: please fix feeds/packages/net/databag/Makefile - see logs/feeds/packages/net/databag/dump.txt for details
ERROR: please fix feeds/packages/net/dns-over-https/Makefile - see logs/feeds/packages/net/dns-over-https/dump.txt for details
ERROR: please fix feeds/packages/net/dnscrypt-proxy2/Makefile - see logs/feeds/packages/net/dnscrypt-proxy2/dump.txt for details
ERROR: please fix feeds/packages/net/dnslookup/Makefile - see logs/feeds/packages/net/dnslookup/dump.txt for details
ERROR: please fix feeds/packages/net/dnsproxy/Makefile - see logs/feeds/packages/net/dnsproxy/dump.txt for details
ERROR: please fix feeds/packages/net/dnstap/Makefile - see logs/feeds/packages/net/dnstap/dump.txt for details
ERROR: please fix feeds/packages/net/evilginx2/Makefile - see logs/feeds/packages/net/evilginx2/dump.txt for details
ERROR: please fix feeds/packages/net/frp/Makefile - see logs/feeds/packages/net/frp/dump.txt for details
ERROR: please fix feeds/packages/net/gateway-go/Makefile - see logs/feeds/packages/net/gateway-go/dump.txt for details
ERROR: please fix feeds/packages/net/geoipupdate/Makefile - see logs/feeds/packages/net/geoipupdate/dump.txt for details
ERROR: please fix feeds/packages/net/geth/Makefile - see logs/feeds/packages/net/geth/dump.txt for details
ERROR: please fix feeds/packages/net/gg/Makefile - see logs/feeds/packages/net/gg/dump.txt for details
ERROR: please fix feeds/packages/net/git-lfs/Makefile - see logs/feeds/packages/net/git-lfs/dump.txt for details
ERROR: please fix feeds/packages/net/irtt/Makefile - see logs/feeds/packages/net/irtt/dump.txt for details
ERROR: please fix feeds/packages/net/kcptun/Makefile - see logs/feeds/packages/net/kcptun/dump.txt for details
ERROR: please fix feeds/packages/net/librespeed-go/Makefile - see logs/feeds/packages/net/librespeed-go/dump.txt for details
ERROR: please fix feeds/packages/net/nebula/Makefile - see logs/feeds/packages/net/nebula/dump.txt for details
ERROR: please fix feeds/packages/net/netbird/Makefile - see logs/feeds/packages/net/netbird/dump.txt for details
ERROR: please fix feeds/packages/net/nextdns/Makefile - see logs/feeds/packages/net/nextdns/dump.txt for details
ERROR: please fix feeds/packages/net/obfs4proxy/Makefile - see logs/feeds/packages/net/obfs4proxy/dump.txt for details
ERROR: please fix feeds/packages/net/overture/Makefile - see logs/feeds/packages/net/overture/dump.txt for details
ERROR: please fix feeds/packages/net/rclone/Makefile - see logs/feeds/packages/net/rclone/dump.txt for details
ERROR: please fix feeds/packages/net/restic-rest-server/Makefile - see logs/feeds/packages/net/restic-rest-server/dump.txt for details
ERROR: please fix feeds/packages/net/sing-box/Makefile - see logs/feeds/packages/net/sing-box/dump.txt for details
ERROR: please fix feeds/packages/net/snowflake/Makefile - see logs/feeds/packages/net/snowflake/dump.txt for details
ERROR: please fix feeds/packages/net/speedtest-go/Makefile - see logs/feeds/packages/net/speedtest-go/dump.txt for details
ERROR: please fix feeds/packages/net/tailscale/Makefile - see logs/feeds/packages/net/tailscale/dump.txt for details
ERROR: please fix feeds/packages/net/trojan-go/Makefile - see logs/feeds/packages/net/trojan-go/dump.txt for details
ERROR: please fix feeds/packages/net/v2ray-core/Makefile - see logs/feeds/packages/net/v2ray-core/dump.txt for details
ERROR: please fix feeds/packages/net/v2raya/Makefile - see logs/feeds/packages/net/v2raya/dump.txt for details
ERROR: please fix feeds/packages/net/wgsd/Makefile - see logs/feeds/packages/net/wgsd/dump.txt for details
ERROR: please fix feeds/packages/net/wifi-presence/Makefile - see logs/feeds/packages/net/wifi-presence/dump.txt for details
ERROR: please fix feeds/packages/net/xray-core/Makefile - see logs/feeds/packages/net/xray-core/dump.txt for details
ERROR: please fix feeds/packages/net/yggdrasil/Makefile - see logs/feeds/packages/net/yggdrasil/dump.txt for details
[... cut ...]
make[2]: Leaving directory '/mnt/openmptcprouter/myplatform/6.6/source'
make[1]: *** [tools/Makefile:227: /mnt/openmptcprouter/myplatform/6.6/source/staging_dir/host/stamp/.tools_compile_nyyyyyynnnyyyyynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/mnt/openmptcprouter/myplatform/6.6/source'
make: *** [/mnt/openmptcprouter/myplatform/6.6/source/include/toplevel.mk:248: world] Error 2
```

More logs (with pastebin limits): 
https://pastebin.com/Y5YYUwJw
https://pastebin.com/iTHWEJLA

Ysurac, 2024-11-19:
You can safely ignore all "ERROR".
For the latest error difficult to say anything, did you try with a "make -j1 V=s" in source directory ?

eXulW0lFy, 2024-11-22:
Still got an error with -j1 V=s

```bash
touch /mnt/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/tcptraceroute-1.5beta7/.prepared_bb3d45e9ce01ff15bb4a22ca13ec0328_18f1e190c5d53547fed41a3eaa76e9e9_check
. /mnt/openmptcprouter/myplatform/6.6/source/include/shell.sh; zstdcat /mnt/openmptcprouter/myplatform/6.6/source/dl/tcptraceroute-1.5beta7.tar.zst | tar -C /mnt/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/tcptraceroute-1.5beta7/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /mnt/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/tcptraceroute-1.5beta7

Applying /mnt/openmptcprouter/feeds/openmptcprouter/tcptraceroute/patches/001-configure_cross_compile.patch using plaintext: 
patching file configure
touch /mnt/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/tcptraceroute-1.5beta7/.prepared_bb3d45e9ce01ff15bb4a22ca13ec0328_18f1e190c5d53547fed41a3eaa76e9e9
rm -f /mnt/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/tcptraceroute-1.5beta7/.configured_*
rm -f /mnt/openmptcprouter/myplatform/6.6/source/staging_dir/target-aarch64_generic_musl/stamp/.tcptraceroute_installed
(cd /mnt/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/tcptraceroute-1.5beta7/./; if [ -x ./configure ]; then find /mnt/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/tcptraceroute-1.5beta7/ -name config.guess | xargs -r chmod u+w; find /mnt/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/tcptraceroute-1.5beta7/ -name config.guess | xargs -r -n1 cp --remove-destination /mnt/openmptcprouter/myplatform/6.6/source/scripts/config.guess; find /mnt/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/tcptraceroute-1.5beta7/ -name config.sub | xargs -r chmod u+w; find /mnt/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/tcptraceroute-1.5beta7/ -name config.sub | xargs -r -n1 cp --remove-destination /mnt/openmptcprouter/myplatform/6.6/source/scripts/config.sub; AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/mnt/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/tcptraceroute-1.5beta7=tcptraceroute-1.5beta7 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD="aarch64-openwrt-linux-musl-ld.bfd" NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/mnt/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/tcptraceroute-1.5beta7=tcptraceroute-1.5beta7 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/mnt/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/tcptraceroute-1.5beta7=tcptraceroute-1.5beta7 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/mnt/openmptcprouter/myplatform/6.6/source/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/usr/include -I/mnt/openmptcprouter/myplatform/6.6/source/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include -I/mnt/openmptcprouter/myplatform/6.6/source/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include/fortify " LDFLAGS="-L/mnt/openmptcprouter/myplatform/6.6/source/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/usr/lib -L/mnt/openmptcprouter/myplatform/6.6/source/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro "   CPPFLAGS="$CPPFLAGS -I/mnt/openmptcprouter/myplatform/6.6/source/staging_dir/target-aarch64_generic_musl/usr/lib/libnet-1.1.x/include" LDFLAGS="$LDFLAGS -L/mnt/openmptcprouter/myplatform/6.6/source/staging_dir/target-aarch64_generic_musl/usr/lib/libnet-1.1.x/lib" PATH="/mnt/openmptcprouter/myplatform/6.6/source/staging_dir/target-aarch64_generic_musl/usr/lib/libnet-1.1.x/bin:$PATH"  ./configure --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=aarch64-unknown-linux-gnu --disable-dependency-tracking --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls   ; fi; )
configure: loading site script /mnt/openmptcprouter/myplatform/6.6/source/include/site/aarch64
checking build system type... aarch64-unknown-linux-gnu
checking host system type... aarch64-openwrt-linux-gnu
checking target system type... aarch64-openwrt-linux-gnu
checking for a BSD-compatible install... /mnt/openmptcprouter/myplatform/6.6/source/staging_dir/host/bin/install -c
checking whether build environment is sane... yes
checking whether make sets $(MAKE)... yes
checking for working aclocal-1.4... missing
checking for working autoconf... found
[... cut ...]
make[2]: Leaving directory '/mnt/openmptcprouter/myplatform/6.6/source'
make[1]: *** [package/Makefile:171: /mnt/openmptcprouter/myplatform/6.6/source/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/openmptcprouter/myplatform/6.6/source'
make: *** [/mnt/openmptcprouter/myplatform/6.6/source/include/toplevel.mk:248: world] Error 2
```

Since pastebin has filesize limits, I cat send google drive link: https://drive.google.com/file/d/1ybNcFR_8Ck2EfjuZKur0XJVE39VlSKDe/view?usp=sharing

eXulW0lFy, 2024-12-05:
Found this (https://forum.openwrt.org/t/local-build-failed-go-bootstrap-cannot-be-installed-on-linux-arm64/213582/5) and fixed golang error, but got a new one:

``` bash
make[3]: Leaving directory '/mnt/openmptcprouter/feeds/openmptcprouter/golang/golang'
time: package/feeds/openmptcprouter/golang/host-compile#1.12#1.00#3.48
make[3]: Entering directory '/mnt/openmptcprouter/feeds/openmptcprouter/v2ray-core'
mkdir -p /mnt/openmptcprouter/myplatform/6.6/source/dl
SHELL= flock /mnt/openmptcprouter/myplatform/6.6/source/tmp/.v2ray-core-5.7.0.tar.zst.flock -c '  	 /mnt/openmptcprouter/myplatform/6.6/source/scripts/dl_github_archive.py --dl-dir="/mnt/openmptcprouter/myplatform/6.6/source/dl" --url="https://github.com/v2fly/v2ray-core.git" --version="8de2f27043b00612986d96f37975cd4aa98b49c3" --subdir="v2ray-core-5.7.0" --source="v2ray-core-5.7.0.tar.zst" --hash="x" --submodules  || ( 	echo "Checking out files from the git repository..."; mkdir -p /mnt/openmptcprouter/myplatform/6.6/source/tmp/dl && cd /mnt/openmptcprouter/myplatform/6.6/source/tmp/dl && rm -rf v2ray-core-5.7.0 && [ \! -d v2ray-core-5.7.0 ] && git clone  https://github.com/v2fly/v2ray-core.git v2ray-core-5.7.0 && (cd v2ray-core-5.7.0 && git checkout 8de2f27043b00612986d96f37975cd4aa98b49c3) && export TAR_TIMESTAMP=`cd v2ray-core-5.7.0 && git log -1 --format='\''@%ct'\''` && echo "Generating formal git archive (apply .gitattributes rules)" && (cd v2ray-core-5.7.0 && git config core.abbrev 8 && git archive --format=tar HEAD --output=../v2ray-core-5.7.0.tar.git) && tar --ignore-failed-read -C v2ray-core-5.7.0 -f v2ray-core-5.7.0.tar.git -r .git .gitmodules 2>/dev/null && rm -rf v2ray-core-5.7.0 && mkdir v2ray-core-5.7.0 && tar -C v2ray-core-5.7.0 -xf v2ray-core-5.7.0.tar.git && (cd v2ray-core-5.7.0 && git submodule update --init --recursive --  && rm -rf .git .gitmodules) && echo "Packing checkout..." && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c v2ray-core-5.7.0 | 	zstd -T0 --ultra -20 -c > /mnt/openmptcprouter/myplatform/6.6/source/tmp/dl/v2ray-core-5.7.0.tar.zst && mv /mnt/openmptcprouter/myplatform/6.6/source/tmp/dl/v2ray-core-5.7.0.tar.zst /mnt/openmptcprouter/myplatform/6.6/source/dl/ && rm -rf v2ray-core-5.7.0; );     '
v2ray-core-5.7.0.tar.zst: Download from https://github.com/v2fly/v2ray-core.git failed
v2ray-core-5.7.0.tar.zst: Requires sha256sum for verification
Checking out files from the git repository...
Cloning into 'v2ray-core-5.7.0'...
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 1389 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
make[3]: *** [Makefile:146: /mnt/openmptcprouter/myplatform/6.6/source/dl/v2ray-core-5.7.0.tar.zst] Error 128
make[3]: Leaving directory '/mnt/openmptcprouter/feeds/openmptcprouter/v2ray-core'
time: package/feeds/openmptcprouter/v2ray-core/compile#0.83#1.26#119.94
    ERROR: package/feeds/openmptcprouter/v2ray-core failed to build.
make[2]: *** [package/Makefile:177: package/feeds/openmptcprouter/v2ray-core/compile] Error 1
make[2]: Leaving directory '/mnt/openmptcprouter/myplatform/6.6/source'
make[1]: *** [package/Makefile:171: /mnt/openmptcprouter/myplatform/6.6/source/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/openmptcprouter/myplatform/6.6/source'
make: *** [/mnt/openmptcprouter/myplatform/6.6/source/include/toplevel.mk:248: world] Error 2
```

eXulW0lFy, 2024-12-06:
Ok, I'm trying to build on my pc now. Got new error:
``` bash
Generating index for package ./xray-core_1.8.24-r1_aarch64_generic.ipk
Generating index for package ./xtables-nft_1.8.8-r2_aarch64_generic.ipk
Generating index for package ./zlib_1.3.1-r1_aarch64_generic.ipk
Signing package index...
make[2]: Leaving directory '/home/exulw0lf/os_builder/openmptcprouter/myplatform/6.6/source'
export MAKEFLAGS= ;make -w -r json_overview_image_info
make[2]: Entering directory '/home/exulw0lf/os_builder/openmptcprouter/myplatform/6.6/source'
WORK_DIR=/home/exulw0lf/os_builder/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_generic_musl/json_info_files /home/exulw0lf/os_builder/openmptcprouter/myplatform/6.6/source/scripts/json_overview_image_info.py /home/exulw0lf/os_builder/openmptcprouter/myplatform/6.6/source/bin/targets/rockchip/armv8/profiles.json 
JSON info file script could not find any JSON files for target
make[2]: Leaving directory '/home/exulw0lf/os_builder/openmptcprouter/myplatform/6.6/source'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/home/exulw0lf/os_builder/openmptcprouter/myplatform/6.6/source'
make[2]: Leaving directory '/home/exulw0lf/os_builder/openmptcprouter/myplatform/6.6/source'
make[1]: Leaving directory '/home/exulw0lf/os_builder/openmptcprouter/myplatform/6.6/source'
```

There's no profiles.json file. And I'm not sure, what targets I should select in menuconfig.

[image]

[image]

[image]

closed 2025-03-11 by github-actions[bot]