Ysurac/openmptcprouter #3652: 0.61/0

muko82, 2024-11-14:
## Expected Behavior
1- While Using two 4g connections, if the second one is connected later the system should start balancing traffic after seconds or minutes
2- Having two 4g connections, one of the is under Vpn (mptcp Is filtered by one provider). No bufferbloat should occur assuming other settings are properly settled.
3- using x-ray/v2ray and uploading files with FTP

## Current Behavior
1- This is not happening even after hours. In order to have it working properly I need to reboot
2- If just one connection is under VPN while uploading or streaming bufferbloat is heavily affecting the connection.
3- USING XRAY/V2RAY is causing timeout while uploading that causes file corruptions

## Possible Solution
1- No idea
2- Having both connections under VPN eliminate 90% of the problem having a bonded connection working properly but this is just a workaround
3- Using shadowsocks eliminate the problem but is a workaround that also introduces packetloss while streaming with OBS on YouTube (RTMP/hls)

## Specifications

  - OpenMPTCProuter version: 0.62 kernel 6.6 snapshot of November 12th
  - OpenMPTCProuter VPS version: 0.1302-test with script of November 13rd
  - OpenMPTCProuter platform: RPI4

muko82, 2024-11-14:
@Ysurac I don't know what did you do in the latest snapshot of November 13rd but it seems that issues 2 and 3 were solved.
Checking for the first one but it's working now.

Will it be possible to have some changelog also for snaphots?

muko82, 2024-11-15:
@Ysurac Issue number 1 apparently is the only one which persists

Ysurac, 2024-11-15:
What is the result, via SSH on the router, of ```ip a``` and ```multipath -f``` when all connections are not aggregated ?

closed 2025-02-18 by github-actions[bot]