Ysurac/openmptcprouter #3653: Snapshot information

DivineRuins, 2024-11-14:
how do i know when a snapshot available on snapshots.openmptcprouter.com is compiled, or what commit it has I saw couple fixes I need in feeds, but I don't know if the snapshot has them yet or no, thanks for the help

Ysurac, 2024-11-14:
Snapshots are build with Github Actions, so you can find the building process here: https://github.com/Ysurac/openmptcprouter-feeds/actions and the resulting snapshot in https://snapshots.openmptcprouter.com/
A new snapshot is build on each push on https://github.com/Ysurac/openmptcprouter-feeds

DivineRuins, 2024-11-14:
Thank you, I checked the actions before posting a question silly me clicked on stale action and not an actual push and didn't find any artifacts or build action.

closed 2025-02-17 by github-actions[bot]