Ysurac/openmptcprouter #3654: How can I use a WAN port as LAN?

3dzug, 2024-11-14:
I want to connect a NAS to my openmtcprouter. As I do not want to use eth0 (my computer is connected there) I want to use an unused wan port such eth5.
My setup is: eth0 -> PC eth1 -> mtcp connection eth2 -> mtcp connection eth3 to 5 unused.
Thanks

Ysurac, 2024-11-14:
In Network->Interfaces disable mptcp on the connection and put it in LAN firewall.

Ysurac, 2024-11-14:
DHCP client or static IP (can also be DHCP serer in this case).
If you want to use it exactly the same way as current lan, you can create a Bridge with both interface in Network->Interfaces, "Device" tab and put it in LAN firewall. It will be the only one with an IP.
If you want more help, post screenshot of what you have currently in Network->Interfaces.

3dzug, 2024-11-14:
Yeah so what I want is that eth0 and eth5 act like a bridge (or switch) with 192.168.100.1 on the server side.Similar to what would happen if I plug eth0 into a switch and distribute it there.

3dzug, 2024-11-25:
Hi @Ysurac I've set up the bridge but there it is still not working.
Bridge shall be now eth4 to eth0.
![Bridge1](https://github.com/user-attachments/assets/3b018866-780e-40be-
[image]
b759-10615737bb1b)

3dzug, 2024-11-25:
I understand that but I do not know how I can do that.
[image]

Ysurac, 2024-11-25:
In Device, choose the bridge. But should be more on LAN interface.

3dzug, 2024-11-25:
Can you help me with a step by step guide? Within the interface setting I cant see on lan how I can assign the bridge.

closed 2025-02-28 by github-actions[bot]