Ysurac/openmptcprouter #3656: High latency on VPS 1032 compared to 1028

patimages, 2024-11-14:
Hello, 

I am using openmptcprouter v0.61-5.4 r0+16862-170d9e44 on my router and have access to 2 VPS : 
- one with : 5.4.207-mptcp 0.1031 (located at work)
- on with : 5.4.207-mptcp 0.1028 (old on Debian 10 - located on OVH).

Since I move the first one to 0.1031, I note a high latency, especially when connecting US based sites (5 seconds to load yahoo.us). If I use the old one on 0.1028, all is blazing fast. 
This is an issue for me as I would need to use the one on 0.1031 as it is connecting me to my work. 

Further info : I use shadowsocks, glorytun TCP and BLEST->BBR. Playing with these does not seem to help. 

I would be grateful for any help and sorry in advance for Ycarus (always falls on you in the end) ! 

Thanks !

moi6576, 2024-11-14:
Hi, 
I have the exact same issue (high latency on the latest 5.4 legacy versions : it is actually worst on 6.6). Glad someone report it too as I thought the issue was on my side (although I just updated, nothing more).

Ysurac, 2024-11-15:
5.4 based version are not supported anymore.
VPS location can change latency, you can also use your browser developer tools to check what request is slower.

vempire-ghost, 2024-11-15:
> 5.4 based version are not supported anymore. VPS location can change latency, you can also use your browser developer tools to check what request is slower.

If kernel 5.4 is no longer supported, why are there still new snapshots with this kernel?  
For someone like me who needs to use this version, which version should I use?

Ysurac, 2024-11-15:
@vempire-ghost it's not the subject here... 
Snapshot are generated because I can for now. MPTCP out of tree for kernel 5.4 is deprecated. I don't make any tests or packages update of kernel 5.4, so I can't provide any support on this kernel.

patimages, 2024-11-17:
Understood. Thanks for the info

patimages, 2024-11-17:
Understood, thanks for the info.

closed 2025-02-22 by github-actions[bot]