Ysurac/openmptcprouter #3659: Advanced port forwarding to attached NAS

3dzug, 2024-11-15:
Hi there,
as I have almost no skills in port forwarding I am asking for some help. I want to configure external access to my NAS storage.
The NAS is on adpater eth5 (IP Adapter: 192.168.200.1) and NAS IP Is 192.168.200.20

I want to access the following ports from extern server 45.90.xxx.xxx and route them to eth5 and 192.168.200.20. eth0 and eth5 use are in the lan firewall setup. No further changes were done. The required ports are:
80 (link sharing), 443 (link sharing), 5000 (HTTP), 5001 (HTTPS)

Setup attached. Thanks alot.
[image]

Ysurac, 2024-11-16:
The documentation should help you: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding

3dzug, 2024-11-17:
If I do the setup like in your documentation I do not get a connection.
Setup is like this then.

[image]
[image]

If I forward the port directly without the using the nat it seems to work somehow.
[image]
I could get a connection using this port however I got an error message telling me this:

"400 Bad Request
The plain HTTP request was sent to HTTPS port"

But its better then no connection. What can I do and how can I tell that this ? Or is the setup I did already correct.
Another thing, if I also open port 80 and foward it to 192.168.200.20 I can access also the opennmtcp server which I do not want and can be found on 192.168.200.1
Is there a way to disable it?

Ysurac, 2024-11-17:
For the error message, it mean that you try to reach via http a https destination, so use https://<vpsip>:port instad of http.

3dzug, 2024-11-17:
Thanks, what about my other questions? Is the setup correct? Have I done something wrong? Can I prevent openmtcprouter from being accessible with port 80 forwarded?

closed 2025-02-21 by github-actions[bot]