Ysurac/openmptcprouter #3662: bpi-r3 mini openmptcprouter

M0rvius, 2024-11-17:
hello, i found snapshots that there is support for banana pi bpi-r3, but i'm not sure that bpi-r3 mini will support this firmware, i'm very bad at compiling firmware and i want to ask experienced people to compile firmware for bpi-r3 mini if ​​there is such an opportunity

Ysurac, 2024-11-17:
Banana BPI-R3 mini need a specific image, I've not compiled for it yet but I'm not sure BPI-R3 already work...

SoulInfernoDE, 2024-12-04:
I would like to see also support for the bpi-r4 as it has wifi 7 and can swap all other routers out of my setup

q2go, 2024-12-04:
Just FWIW, my understanding is that WiFi 7 will not be supported by OpenWrt at all for some time to come. But we should be able to use WiFi 6E already now, with some teething problems here and there.

Ysurac, 2024-12-04:
There is also an image for BPI-R4 in snapshot https://snapshots.openmptcprouter.com/
But, as BPI-R3, I don't think it's working.

closed 2025-03-09 by github-actions[bot]