Ysurac/openmptcprouter #3663: Debian 10 Address family for hostname not supported

thindata, 2024-11-17:
On a fresh install on debian 10 using

`wget -O - https://www.openmptcprouter.com/server/debian10-x86_64.sh | sh`

After a reboot when chekcing syslog i see

```
Nov 17 20:07:52 OpenMPTCProuter-server omr-admin.py[31061]: ERROR:    [Errno -9] Address family for hostname not supported
Nov 17 20:07:52 OpenMPTCProuter-server systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
Nov 17 20:07:52 OpenMPTCProuter-server systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Nov 17 20:07:52 OpenMPTCProuter-server systemd[1]: omr-admin.service: Service RestartSec=100ms expired, scheduling restart.
Nov 17 20:07:52 OpenMPTCProuter-server systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 5.
Nov 17 20:07:52 OpenMPTCProuter-server systemd[1]: Stopped OMR-Admin.
Nov 17 20:07:52 OpenMPTCProuter-server systemd[1]: omr-admin.service: Start request repeated too quickly.
Nov 17 20:07:52 OpenMPTCProuter-server systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Nov 17 20:07:52 OpenMPTCProuter-server systemd[1]: Failed to start OMR-Admin.
```

Ysurac, 2024-11-18:
Try snapshot VPS script: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

thindata, 2024-11-18:
That fixed it manby thanks