Ysurac/openmptcprouter #3664: The compatibility issue of ZBT z8102ax.

a834003831, 2024-11-18:
First of all, I would like to thank the open-source developer for their work. I would like to inquire whether the ZBT z8102ax device has already included the driver for the 5G module. If it has, could you please let me know which models are supported? I am considering purchasing this device soon. Thank you!

Ysurac, 2024-11-18:
Quectel modems are working correctly. 
But remember that Z8102AX CPU is slow, so can't reach max aggregated speed of 2 5G modems.

a834003831, 2024-11-18:
> Quectel modems are working correctly. But remember that Z8102AX CPU is slow, so can't reach max aggregated speed of 2 5G modems.

First of all, thank you for your prompt response.
﻿
﻿
What is the maximum number of Ethernet (WAN) connections supported? I hope to connect multiple 5G CPEs to improve stability.You can think of it as: Is it possible to change the LAN port to a WAN port?

Ysurac, 2024-11-18:
By default, LAN ports are used as WAN.
It can aggregate up to 8 IPs.

davidohne, 2025-01-27:
> But remember that Z8102AX CPU is slow

What would be max aggregated speed on this device/cpu in your opinion?

Ysurac, 2025-01-27:
About 400Mbps

packlet, 2025-02-07:
Any chance of an OMR build for the Z8109ax ?   Or maybe the Z8102ax build will 'just work' on it?  
Similar to Z8102ax, but has m.2, SIM slots and antennas for 4 modems...   (estimated 400Mbps cap still would be useful, imo)   https://www.zbtwifi.com/productinfo/1323992.html

Ysurac, 2025-02-08:
No idea if Z8102AX build will just work on it, maybe if it's same hardware in fact with just 4 modems...
But with same CPU it's not a so good product...

I don't know if I will get a device or not with ZBT OpenWRT changes (the DTS file)...

closed 2025-05-14 by github-actions[bot]