Ysurac/openmptcprouter #3667: Multi-IP with Shadowsocks 2022 or V2Ray-VLESS

FPi-sh, 2024-11-18:
## Expected Behavior
With shadowsocks-libev I was able to use several public IPs on VPS mapped with different lan interfaces on omr. Is it possible to do this with SS2022 or V2Ray?

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 0.61
  - OpenMPTCProuter VPS version: 0.1031
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-11-18:
For now it's still only supported by Shadowsocks-Libev

closed 2025-02-22 by github-actions[bot]

ftjnrur, 2025-03-22:
Hi, 
when i configure multi ip on shadowsocks on the vps i get on the device "can't contact Server Admin script"
could you please show me how you do multi ip configuration on shadowsocks-libev?