Ysurac/openmptcprouter #3670: Le vpn ne répond pas

djatomix, 2024-11-19:
- OpenMPTCProuter version: < openmptcprouter-v0.61-6.6-r0+27346-c7ba5574f5-x86-64-generic-ext4-combined.img.gz >
  - OpenMPTCProuter VPS version: < OpenMPTCProuter VPS 0.1031 >
  - OpenMPTCProuter VPS provider: Sapinet
  - OpenMPTCProuter platform: https://www.amazon.fr/gp/product/B07DHBHQYR/ref=ppx_yo_dt_b_asin_title_o08_s02?ie=UTF8&th=1

Good morning
I encounter the problem of "VPN not responding" on a brand new installation.

It seems to see all my interfaces but jumps from one to the other without remaining connected to more than one at a time (see: capture) however the command "uci show openmptcprouter" sends me back all my wans with their IP public. 

Do you have an idea please?

Thanks in advance.

[image]

Ysurac, 2024-11-19:
All interfaces are using same IPs, it's a bad idea. And here it's seems that there is a network issue, Bouygues is detected as provider on Starlink ?

djatomix, 2024-11-19:
yes they all have the same IP but as they are each on a different ethernet in my router, normally it works! no? 
(this does not pose a problem for me on my other configuration which is 100% identical) 
As for Bouygues detected on Eth 5, it is normal, exceptionally for the test I put a fiber box because I was lazy to take out the Starlink! ;)

Ysurac, 2024-11-20:
Can you try to disable MPTCP on all connection and check if at least it's working with one ? Then add connection back one by one.

djatomix, 2024-11-20:
Alors pour commencer, merci pour ton aide,
J'ai donc suivi tes recommandations, en désactivant l'ensemble des connexions elles fonctionnent bien une par une.
Par contre, dès que j'active deux, en même temps, il n'en prend qu'une seule.

J'ai donc essayé de changer les adresses IP comme suggérer, maintenant il récupère bien les infos de chaque passerelle avec les FAI et les IP public, et tout fonctionne sur deux interfaces, et je n'ai plus l'erreur du VPN, par contre pour mes trois autres interfaces, j'ai le message: "multipath semble bloqué sur la connexion" cf capture

[image]

Ysurac, 2024-11-20:
Le message est qu'un warning, si l'agrégation se passe bien alors pas de soucis.

djatomix, 2024-11-20:
Ah ok parfait merci beaucoup.
Je ferais le test de l'agregation demain car la je suis en zone blanche 4g/5g ... donc a part ma fibre ya rien... 
Je te tiens au courant demain.
Merci beaucoup !!

closed 2025-02-23 by github-actions[bot]