Ysurac/openmptcprouter #3672: Can't get public lP address from ShadowSocks Rust

a834003831, 2024-11-20:
Great developer, hello! I have encountered this issue multiple times: "Unable to obtain a public IP." Switching the default proxy does not resolve the problem either. I suspect it might be related to the VPS. Could you please advise on how to reset the VPS? Or how can I assist you in retrieving logs for analysis?
My device is an R4S with a 5.4 kernel, version 060.
It seems that I didn't encounter this issue before, but I recently switched to a new VPS. Thank you!
I will stay online on GitHub to assist with your analysis and help identify the issue.
[image]

a834003831, 2024-11-20:
When I switch the "Default Proxy" to XRay VLESS, it prompts: Unable to obtain a public IP address through the aggregation server ShadowSocks.

a834003831, 2024-11-20:
> Please use template. 5.4 based version are no more supported. Use latest 0.61 release, New VPS script version can not work with old releases.

Sorry, because my English is not very good, I am worried that the information I fill in the template might be incorrect, leading to errors in problem analysis. I will update the router to version 061, or is it possible to downgrade the VPS?

a834003831, 2024-11-20:
> Please use template. 5.4 based version are no more supported. Use latest 0.61 release, New VPS script version can not work with old releases.

I used BalenaEtcher-Portable to flash 061-6.6, but it failed. I tried both ext4 and squashfs formats. I also tested 062-6.6 ext4, but the result was the same. I suspected it might be an issue with the TF card, but when I used the same card to flash the 060-5.4 squashfs version, it worked perfectly. Is there any solution for this problem?

Ysurac, 2024-11-20:
What error do you get ?

a834003831, 2024-11-20:
> What error do you get ?

Just now, I tested the 062-6.6 squashfs version, but it still failed. The flashing process was normal, but after inserting the TF card into the R4S, it failed to boot. When connecting the computer to the LAN port, it showed "Unidentified Network," and the LAN port indicator light on the R4S did not turn on.
[image]

a834003831, 2024-11-20:
> What error do you get ?

I have already tried 061-5.4 ext4 and squashfs, but neither worked. So far, it seems that only 060-5.4 works for me. Is there a way to downgrade the VPS?

a834003831, 2024-11-20:
I have deployed version 6.6 to the new VPS, but my router is unable to install the firmware with the 6.6 kernel version. I will continue to wait for a new solution. Once again, thank you to the developers for their great efforts.

Ysurac, 2024-11-20:
I think I fixed issue, new snapshot is compiling now.

a834003831, 2024-11-21:
> I think I fixed issue, new snapshot is compiling now.

I have tested the latest 062-6.6, and it can now be successfully flashed. However, when setting up the WAN, after saving the configuration, 192.168.100.1 becomes inaccessible and cannot be pinged. This issue occurs in both versions, including 062-5.4. Thank you to the developers for the prompt fixes!

Ysurac, 2024-11-21:
How are you setting the WAN ?

a834003831, 2024-11-21:
> How are you setting the WAN ?

First, open: Aggregate Router, add the interface **ETH0**, rename it to **wan1**, and set it to **DHCP**. The operation is the same as in version 060-5.4.

a834003831, 2024-11-21:
For the "Type" field, I will select "Normal." I'm not sure if this translation is clear enough.

Ysurac, 2024-11-21:
You checked that LAN is not already on eth0 ?
I would need, via SSH on the router, the result of ```uci show network```

a834003831, 2024-11-21:
> You checked that LAN is not already on eth0 ? I would need, via SSH on the router, the result of `uci show network`

I connected the upstream router (5Gcpe) to the WAN port before configuring it. After the configuration, even the SSH connection becomes inaccessible. I will reflash the firmware now and provide the details to you.

a834003831, 2024-11-21:
The above information was obtained after reflashing without making any changes. I don’t dare to add a WAN interface again. I have tried multiple times, and after setting it up, SSH, ping, and web access to "192.168.100.1" all fail.

Ysurac, 2024-11-21:
You have lan interface on eth1 and wan1 & wan2 on eth0 using MACVlan.
If you want to have only one WAN on eth0, you first need to remove wan1 and wan2 in Network->Interface and then you should be able to add wan interface and put it in WAN firewall zone.

a834003831, 2024-11-21:
> You have lan interface on eth1 and wan1 & wan2 on eth0 using MACVlan. If you want to have only one WAN on eth0, you first need to remove wan1 and wan2 in Network->Interface and then you should be able to add wan interface and put it in WAN firewall zone.

I have followed the steps to delete WAN1 and WAN2 in "Network -> Interface."  
Should I next add ETH0 as WAN1 in the "Aggregation Routing" interface, or should I add it in "Network -> Interface"?

Ysurac, 2024-11-21:
Add it in Network->Interface

a834003831, 2024-11-21:
> Add it in Network->Interface

root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.device='lo'
network.loopback.metric='9998'
network.loopback.ip4table='9998'
network.loopback.ip6table='69998'
network.globals=globals
network.globals.ula_prefix='fdbf:24f7:842::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='4'
network.globals.mptcp_subflows='8'
network.globals.mptcp_add_addr_accepted='8'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_pm_type='0'
network.globals.mptcp_disable_initial_config='0'
network.globals.mptcp_force_multipath='1'
network.globals.mptcpd_enable='0'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.device='eth1'
network.lan.ifname='eth1'
network.lan.metric='9999'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.addlatency='0'
network.lan.txqueuelen='2000'
network.lan.multipath='off'
network.lan.ip4table='9999'
network.lan.ip6table='69999'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.omrvpn=interface
network.omrvpn.device='tun0'
network.omrvpn.proto='none'
network.omrvpn.ip4table='1500'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.type='tunnel'
network.omrvpn.txqueuelen='100'
network.omrvpn.metric='1500'
network.omrvpn.ip6table='61500'
network.tun0=device
network.tun0.name='tun0'
network.tun0.mtu='1500'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='1201'
network.omr6in4.multipath='off'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.gateway='fe80::a00:1'
network.omr6in4.ip6addr='fe80::a00:2/128'
network.omr6in4.auto='0'
network.omr6in4.metric='1201'
network.omr6in4.ip6table='61201'
network.eth1_fw_rule6=rule6
network.eth1_fw_rule6.priority='1'
network.eth1_fw_rule6.mark='0x65399999'
network.eth1_fw_rule6.lookup='9999'
network.wan1_fw_rule6=rule6
network.wan1_fw_rule6.priority='1'
network.wan1_fw_rule6.mark='0x65396'
network.wan1_fw_rule6.lookup='6'
network.wan2_fw_rule6=rule6
network.wan2_fw_rule6.priority='1'
network.wan2_fw_rule6.mark='0x65397'
network.wan2_fw_rule6.lookup='7'
network.tun0_fw_rule6=rule6
network.tun0_fw_rule6.priority='1'
network.tun0_fw_rule6.mark='0x65391500'
network.tun0_fw_rule6.lookup='1500'
network.all_fw_rule6=rule6
network.all_fw_rule6.priority='1'
network.all_fw_rule6.mark='0x6539'
network.all_fw_rule6.lookup='6991337'
network.wan=interface
network.wan.proto='dhcp'
network.wan.device='eth0'
network.wan.force_link='1'
network.wan.multipath='off'
network.wan.addlatency='0'
network.wan.metric='9998'
network.wan.ip4table='9998'
network.wan.ip6table='69998'
network.@device[1]=device
network.@device[1].name='eth0'
network.@device[1].ipv6='0'
network.eth0_fw_rule6=rule6
network.eth0_fw_rule6.priority='1'
network.eth0_fw_rule6.mark='0x65399998'
network.eth0_fw_rule6.lookup='9998'
network.eth1_fw_rule=rule
network.eth1_fw_rule.priority='1'
network.eth1_fw_rule.mark='0x45399999'
network.eth1_fw_rule.lookup='9999'
network.tun0_fw_rule=rule
network.tun0_fw_rule.priority='1'
network.tun0_fw_rule.mark='0x45391500'
network.tun0_fw_rule.lookup='1500'
network.eth0_fw_rule=rule
network.eth0_fw_rule.priority='1'
network.eth0_fw_rule.mark='0x45399998'
network.eth0_fw_rule.lookup='9998'
network.all_fw_rule=rule
network.all_fw_rule.priority='1'
network.all_fw_rule.mark='0x4539'
network.all_fw_rule.lookup='991337'

"I’m really sorry to bother you again. I have performed the following actions: In the 'Network - Interfaces' screen, I deleted wan1 and wan2. Additionally, in the 'Devices' screen, I deleted them and added eth0. I have not yet added anything in the 'Aggregation Routing' screen. Could you please help me check if everything is correct?"

a834003831, 2024-11-21:
In the "Aggregated Routing" interface, after adding the WAN and saving, it is still unable to access 192.168.100.1. It could be an issue with the "Routing Table" or "Firewall"?

a834003831, 2024-11-21:
Could you provide the SSH command line to change the WAN port to a real network interface? I don't need macvlan. Thank you!

a834003831, 2024-11-21:
"I've been studying it all night but couldn't solve the issue. I'm wondering if it's possible to directly overwrite the files from version 060-5.4, namely: /etc/init.d/network and the firewall files, to version 062-6.6? If you have a better solution, I'm waiting online."

a834003831, 2024-11-21:
> Add it in Network->Interface

"I've discovered another bug. When I plug in a USB network adapter (these two adapters work well on version 060-5.4), eth2 and eth3 are not recognized. In version 060-5.4, these interfaces would appear immediately after plugging them in. I hope this can be fixed. Thank you!"

"Tested both 062-5.4 and 062-6.6, and both have the issues mentioned above."

Ysurac, 2024-11-22:
You can't copy directly network and firewall files from previous version.
For the USB network adapter it can be a missing drivers. You may have to install it.

Ysurac, 2024-11-24:
What you mean with network device ? When you use MACvlan, you only need to connect to a switch. It's often a good way when you have modems with DHCP that can be disabled.

q2go, 2024-11-24:
> When you use MACvlan, you only need to connect to a switch.

MacVLAN implies that all the virtual WAN interfaces get their own (randomly chosen?!) MAC addresses, and the physical interface that's underneath needs to operate in promiscuous mode to pick all the traffic up.

Could this cause problems
1. with Ethernet switches that try too be to clever and don't properly recognize that multiple MAC addresses are being used on the same OMR-facing switch port?
2. with performance, as promiscuous mode is said to cause additional CPU load? 

>  It's often a good way when you have modems with DHCP that can be disabled.

Turning DHCP off is only needed if the same physical interface is being used for both WAN and LAN, am I right?

I got a machine with two builtin NICs, and OMR automatically chose eth0 for the LAN interface 192.168.100.1 and eth1 for wan1 and wan2, both of which are 5G WLAN routers, connected to the Ethernet switch by patch cables and with the IP addresses and netmasks statically configured into OMR.

I'd like to leave DHCP turned on as it is nice to be able to connect to each individual 5G router over WiFi, for diagnostic purposes and to be able to circumvent OMR if needed. This seems to work.

closed 2025-02-28 by github-actions[bot]