Ysurac/openmptcprouter #3675: Issue with OMR and Wireguard - Cant connect to site after activating OMR

joerbrando, 2024-11-21:
## Expected Behavior
When connecting directly via one of my intenret lines, im able to turn on Wireguard and connect to the site i need to reach.
When turning on Wireguard on my PC with OMR enabled and all 3x lines bonded, the Wireguard connection should activate and the traffic should work just like it did when i didnt have OMR enabled.

## Current Behavior
When turning on Wireguard on my PC with OMR enabled and all 3x lines bonded, the Wireguard connection activates but doesnt work properly anymore, it wont allow me to connect to the destination, but internet traffic works as expected.
Somehow the Wireguard traffic gets blocked or thrown away for some reason.

## Specifications
  - OpenMPTCProuter version: 0.61-6.6
  - OpenMPTCProuter VPS version: 0.1031 6.6.36-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: Vultr
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-11-21:
You can try to lower MTU in wireguard.
Also check that all is green in System->OpenMPTCProuter, Status page

joerbrando, 2024-12-02:
Is there anything else i should try? 
Is there any reason why using WG in the laptop doesnt work when using OMR, but it works when we are just connecting directly via the firewall without OMR?
Also, is there a way to access the WG settings via the GUI, or WG config is only done via CLI?