Ysurac/openmptcprouter #3676: VPS Upgrade script

VisionLabGroup, 2024-11-21:
Hello, 

i've a problem with upgrade of VPS.
I launch the Debian installation script for upgrade the vps but doesn't work.
The VPS Keep old version.

How I can fix it?

thank you

## Specifications

  - OpenMPTCProuter version: 11
  - OpenMPTCProuter VPS version: 0.61
  - OpenMPTCProuter VPS provider: SERVERPLAN 
  - OpenMPTCProuter platform: x86_64)

VisionLabGroup, 2024-11-21:
wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL="6.6" sh
--2024-11-21 17:54:33--  https://www.openmptcprouter.com/server/debian-x86_64.sh
Risoluzione di www.openmptcprouter.com (www.openmptcprouter.com)... 84.17.59.117, 2400:52e0:1e04::888:1
Connessione a www.openmptcprouter.com (www.openmptcprouter.com)|84.17.59.117|:443... connesso.
Richiesta HTTP inviata, in attesa di risposta... 200 OK
Lunghezza: 99094 (97K) [application/octet-stream]
Salvataggio in: "STDOUT"

-                                0%[                                                    ]       0  --.-KB/s               Check user...
Check Linux version...
Check architecture...
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Update repo.openmptcprouter.com key
--2024-11-21 17:54:34--  https://repo.openmptcprouter.com/openmptcprouter.gpg.key
Resolving repo.openmptcprouter.com (repo.openmptcprouter.com)... 84.17.59.115, 2400:52e0:1e04::888:1
Connecting to repo.openmptcprouter.com (repo.openmptcprouter.com)|84.17.59.115|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1241 (1.2K) [application/octet-stream]
Saving to: '/etc/apt/trusted.gpg.d/openmptcprouter.gpg'

/etc/apt/trusted.gpg.d/openmpt 100%[===================================================>]   1.21K  --.-KB/s    in 0s      

2024-11-21 17:54:34 (818 MB/s) - '/etc/apt/trusted.gpg.d/openmptcprouter.gpg' saved [1241/1241]

Remove lock and update packages list...
Hit:1 http://it.archive.ubuntu.com/ubuntu focal InRelease
Get:2 http://it.archive.ubuntu.com/ubuntu focal-updates InRelease [128 kB]
Get:3 http://it.archive.ubuntu.com/ubuntu focal-backports InRelease [128 kB]
Get:4 http://it.archive.ubuntu.com/ubuntu focal-updates/main i386 Packages [1054 kB]                          
Get:5 http://it.archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [3680 kB]                                     
Hit:6 https://repo.openmptcprouter.com buster InRelease                                                                   
Get:7 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [102 kB]                                                
Get:8 http://it.archive.ubuntu.com/ubuntu focal-updates/main Translation-en [563 kB]                        
Get:9 http://security.ubuntu.com/ubuntu focal-security InRelease [128 kB]                                        
Get:10 http://it.archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [3379 kB]                   
Get:11 http://it.archive.ubuntu.com/ubuntu focal-updates/restricted Translation-en [473 kB]                           
Get:12 http://it.archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1238 kB]                                
Get:13 http://it.archive.ubuntu.com/ubuntu focal-updates/universe i386 Packages [810 kB]                                
Get:14 http://it.archive.ubuntu.com/ubuntu focal-updates/universe Translation-en [297 kB]                              
Hit:15 http://archive.ubuntu.com/ubuntu bionic InRelease                                                                  
Get:16 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [3303 kB]
Get:17 http://security.ubuntu.com/ubuntu focal-security/main i386 Packages [834 kB]
Get:18 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [1016 kB]
Get:19 http://security.ubuntu.com/ubuntu focal-security/universe i386 Packages [682 kB]
Get:20 http://security.ubuntu.com/ubuntu focal-security/universe Translation-en [214 kB]
Fetched 18.0 MB in 3s (6743 kB/s)                                 
Reading package lists... Done
Install apt-transport-https, gnupg and openssh-server...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gnupg is already the newest version (2.2.19-3ubuntu2.2).
openssh-server is already the newest version (1:8.2p1-4ubuntu0.11).
apt-transport-https is already the newest version (2.0.10).
The following packages were automatically installed and are no longer required:
  linux-headers-5.4.0-45 linux-headers-5.4.0-45-generic linux-image-5.4.0-45-generic linux-modules-5.4.0-45-generic
  linux-modules-extra-5.4.0-45-generic
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 15 not upgraded.
Add OpenMPTCProuter repo...
Explanation: Prefer OpenMPTCProuter provided packages over the Debian native ones
Package: *
Pin: origin repo.openmptcprouter.com
Pin-Priority: 1001
--2024-11-21 17:54:45--  https://repo.openmptcprouter.com/openmptcprouter.gpg.key
Resolving repo.openmptcprouter.com (repo.openmptcprouter.com)... 84.17.59.115, 2400:52e0:1e04::888:1
Connecting to repo.openmptcprouter.com (repo.openmptcprouter.com)|84.17.59.115|:443... connected.
HTTP request sent, awaiting response... 200 OK
[... 2349 lines cut ...]
[Thu Nov 21 18:00:27 CET 2024] bash has been found. Changing the shebang to use bash as preferred.
[Thu Nov 21 18:00:28 CET 2024] OK
[Thu Nov 21 18:00:28 CET 2024] Install success!
====================================================================================
OpenMPTCProuter Server is now updated to version 0.1031 !
Keys are not changed, shorewall rules files preserved
You need OpenMPTCProuter >= 0.30
====================================================================================
Restarting systemd daemon...
done
Restarting systemd network...
done
Restarting mlvpn...
done
Restarting v2ray...
done
Restarting xray...
done
Restarting dsvpn...
done
Restarting glorytun tcp...
done
Restarting omr6in4...
done
Restarting OpenVPN

Ysurac, 2024-11-21:
It's updated. You can do a ```cat /etc/motd``` to check version.
Version displayed in status page can be not updated immediately.

VisionLabGroup, 2024-11-24:
Thank you.

but after upgrade the connection doesn t work.

[image]

Ysurac, 2024-11-24:
Check that you can reach https://151.36.77.177:65500/
If not, on the VPS, do a ```journalctl -u omr-admin``` (can be a problem with IPv6, fixed in current snapshot script, I will release new VPS script asap)

VisionLabGroup, 2024-11-24:
This is the log of journalctl -u omr-admin
[image]

VisionLabGroup, 2024-11-25:
Hello @Ysurac,

i've try also the actual snapshot test but i've the same problem.

Can you give me some approximate times please?

Ysurac, 2024-11-25:
It should be fixed in latest snapshot VPS script.

VisionLabGroup, 2024-11-25:
This ? 

[image]

Ysurac, 2024-11-25:
yes

VisionLabGroup, 2024-11-25:
No, i've tried and doesn't work.

Ysurac, 2024-11-25:
You rebooted after install ?
What is the result of ``` journalctl -u omr-admin``` now ?

VisionLabGroup, 2024-11-25:
yes, rebooted it after.

this is log:

root@v023xi-01:~# journalctl -u omr-admin
-- Logs begin at Mon 2024-11-25 08:42:40 CET, end at Mon 2024-11-25 12:07:16 CE>
nov 25 08:42:40 v023xi-01 systemd[1]: omr-admin.service: Main process exited, c>
nov 25 08:42:40 v023xi-01 systemd[1]: omr-admin.service: Failed with result 'ex>
nov 25 08:42:40 v023xi-01 systemd[1]: omr-admin.service: Scheduled restart job,>
nov 25 08:42:40 v023xi-01 systemd[1]: Stopped OMR-Admin.
nov 25 08:42:40 v023xi-01 systemd[1]: Started OMR-Admin.
nov 25 08:42:41 v023xi-01 omr-admin.py[249647]: ERROR:    [Errno -9] Address fa>
nov 25 08:42:41 v023xi-01 systemd[1]: omr-admin.service: Main process exited, c>
nov 25 08:42:41 v023xi-01 systemd[1]: omr-admin.service: Failed with result 'ex>
nov 25 08:42:41 v023xi-01 systemd[1]: omr-admin.service: Scheduled restart job,>
nov 25 08:42:41 v023xi-01 systemd[1]: Stopped OMR-Admin.
nov 25 08:42:41 v023xi-01 systemd[1]: Started OMR-Admin.
nov 25 08:42:41 v023xi-01 omr-admin.py[249649]: ERROR:    [Errno -9] Address fa>
nov 25 08:42:41 v023xi-01 systemd[1]: omr-admin.service: Main process exited, c>
nov 25 08:42:41 v023xi-01 systemd[1]: omr-admin.service: Failed with result 'ex>
nov 25 08:42:41 v023xi-01 systemd[1]: omr-admin.service: Scheduled restart job,>
nov 25 08:42:41 v023xi-01 systemd[1]: Stopped OMR-Admin.
nov 25 08:42:41 v023xi-01 systemd[1]: Started OMR-Admin.
nov 25 08:42:42 v023xi-01 omr-admin.py[249651]: ERROR:    [Errno -9] Address fa>
nov 25 08:42:42 v023xi-01 systemd[1]: omr-admin.service: Main process exited, c>
nov 25 08:42:42 v023xi-01 systemd[1]: omr-admin.service: Failed with result 'ex>
nov 25 08:42:42 v023xi-01 systemd[1]: omr-admin.service: Scheduled restart job,>
nov 25 08:42:42 v023xi-01 systemd[1]: Stopped OMR-Admin.
lines 1-23...skipping...
-- Logs begin at Mon 2024-11-25 08:42:40 CET, end at Mon 2024-11-25 12:07:16 CET. --
nov 25 08:42:40 v023xi-01 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
nov 25 08:42:40 v023xi-01 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
nov 25 08:42:40 v023xi-01 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 1.
nov 25 08:42:40 v023xi-01 systemd[1]: Stopped OMR-Admin.
nov 25 08:42:40 v023xi-01 systemd[1]: Started OMR-Admin.
nov 25 08:42:41 v023xi-01 omr-admin.py[249647]: ERROR:    [Errno -9] Address family for hostname not supported
nov 25 08:42:41 v023xi-01 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
nov 25 08:42:41 v023xi-01 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
nov 25 08:42:41 v023xi-01 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 2.
nov 25 08:42:41 v023xi-01 systemd[1]: Stopped OMR-Admin.
nov 25 08:42:41 v023xi-01 systemd[1]: Started OMR-Admin.
nov 25 08:42:41 v023xi-01 omr-admin.py[249649]: ERROR:    [Errno -9] Address family for hostname not supported
nov 25 08:42:41 v023xi-01 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
nov 25 08:42:41 v023xi-01 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
nov 25 08:42:41 v023xi-01 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 3.
nov 25 08:42:41 v023xi-01 systemd[1]: Stopped OMR-Admin.
nov 25 08:42:41 v023xi-01 systemd[1]: Started OMR-Admin.
nov 25 08:42:42 v023xi-01 omr-admin.py[249651]: ERROR:    [Errno -9] Address family for hostname not supported
nov 25 08:42:42 v023xi-01 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
nov 25 08:42:42 v023xi-01 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
nov 25 08:42:42 v023xi-01 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 4.
nov 25 08:42:42 v023xi-01 systemd[1]: Stopped OMR-Admin.
nov 25 08:42:42 v023xi-01 systemd[1]: Started OMR-Admin.
nov 25 08:42:42 v023xi-01 omr-admin.py[249653]: ERROR:    [Errno -9] Address family for hostname not supported
nov 25 08:42:42 v023xi-01 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
nov 25 08:42:42 v023xi-01 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
nov 25 08:42:42 v023xi-01 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 5.
nov 25 08:42:42 v023xi-01 systemd[1]: Stopped OMR-Admin.
nov 25 08:42:42 v023xi-01 systemd[1]: omr-admin.service: Start request repeated too quickly.
nov 25 08:42:42 v023xi-01 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
nov 25 08:42:42 v023xi-01 systemd[1]: Failed to start OMR-Admin.
nov 25 08:43:00 v023xi-01 systemd[1]: Started OMR-Admin.
nov 25 08:43:00 v023xi-01 omr-admin.py[250029]: ERROR:    [Errno -9] Address family for hostname not supported
nov 25 08:43:00 v023xi-01 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
nov 25 08:43:00 v023xi-01 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
nov 25 08:43:01 v023xi-01 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 1.
nov 25 08:43:01 v023xi-01 systemd[1]: Stopped OMR-Admin.
nov 25 08:43:01 v023xi-01 systemd[1]: Started OMR-Admin.
nov 25 08:43:01 v023xi-01 omr-admin.py[250063]: ERROR:    [Errno -9] Address family for hostname not supported
lines 1-40...skipping...
[... 42 lines cut ...]
nov 25 08:43:01 v023xi-01 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 2.
[... cut ...]
nov 25 08:43:02 v023xi-01 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
nov 25 08:43:03 v023xi-01 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 5.
nov 25 08:43:03 v023xi-01 systemd[1]: Stopped OMR-Admin.
nov 25 08:43:03 v023xi-01 systemd[1]: omr-admin.service: Start request repeated too quickly.
nov 25 08:43:03 v023xi-01 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
nov 25 08:43:03 v023xi-01 systemd[1]: Failed to start OMR-Admin.
nov 25 08:43:20 v023xi-01 systemd[1]: Started OMR-Admin.
lines 1-66

VisionLabGroup, 2024-11-25:
[image]

Ysurac, 2024-11-25:
Strange, this issue should be fixed in snapshot but is still here in current release...
You have IPv6 enabled or not on the VPS ?

Ysurac, 2024-11-25:
Can you reinstall it from snapshot link ? There is no reason it was at least launching before, so should still be able to launch. Make sure you don't use the VPS script release version.

Ysurac, 2024-11-25:
Ubuntu is no more supported. The script is not tested at all on Ubuntu, only on Debian. But not sure it's related here.
There is something I don't understand: previous error was related to IPv6 firewall but IPv6 is disabled. IPv6 was disabled after initial install ?
You should still be able to force host in ```/etc/openmptcprouter-vps-admin/omr-admin-config.json``` by adding or modifying "host": "0.0.0.0"

VisionLabGroup, 2024-11-25:
IPv6 has never been enabled.

i've try also with new vps with debian 10, the result is the same:

[image]

closed 2024-12-10 by VisionLabGroup