Ysurac/openmptcprouter #3678: "Thank you to the developers for their hard work in fixing the issue, but a new problem has arisen."

a834003831, 2024-11-22:
- OpenMPTCProuter version: 6.6-062
  - OpenMPTCProuter VPS version: 6.6-062
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: r4s
  - Country: china

"First of all, I am still grateful to our developers for their frequent fixes in the past few days. Just now, I saw the latest 062-6.6 version, and I immediately flashed it. I’m happy to report that with this version, I successfully configured the WAN port and got internet access. However, the USB network card I used before (which should be the 8153 chipset) is no longer supported in this version. I tried several drivers, but none could properly recognize or function. I hope the developers will add USB network card drivers for devices like the R4S, as it worked very well in version 060-5.4."

a834003831, 2024-11-22:
"I have encountered a new issue with 062-6.6. On X86, after connecting to the server, I am unable to open webpages, although ping works fine. The same issue occurs on R4S. I have confirmed this issue with multiple friends in China. How can I provide the necessary data for you to analyze?"

Ysurac, 2024-11-22:
Please use template and use a short title related to the issue.
For the USB network card, install "kmod-usb-net-rtl8152" via System->Software menu. As it's a module on a snapshot image, you may need to update to latest snapshot to be able to install it.

For the webpages issue:
* Are you able to ping a domain name ? If not, it's a DNS issue. Selecting System->OpenMPTCProuter, Wizard tab, "Advanced settings" checkbox, China in country may help
* Everything is green in status page ? What is the proxy you use (default is now Shadowsocks-Rust) ?

a834003831, 2024-11-23:
> Please use template and use a short title related to the issue. For the USB network card, install "kmod-usb-net-rtl8152" via System->Software menu. As it's a module on a snapshot image, you may need [...]

1. I searched for **kmod-usb-net-rtl8152** under **System -> Software**, but no results were found. My firmware version is:  
   `openmptcprouter-v0.62-snapshot-6.6-r0+28016-48028cd102-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img`,  
   dated November 22, 2024.

2. I have already set the country to **"China"** under **System -> OpenMPTCProuter**, but I still cannot open webpages.  
   - The status page shows all green.  
   - The proxy in use is **Shadowsocks-Rust2022**.  
   - After multiple tests, I found that disconnecting the VPS allows me to browse the internet using **WAN1**. However, as soon as the VPS is connected, webpages become inaccessible.

Ysurac, 2024-11-23:
I can see the package, you updated the list before searching ?
Did you test to ping a domain name ?
From the VPS, try a ```curl http://<domain name you choose>/```

a834003831, 2024-11-23:
> I can see the package, you updated the list before searching ? Did you test to ping a domain name ? From the VPS, try a `curl http://<domain name you choose>/` and try the same from the router via SSH. Another idea: Check via SSH on the router the result of `uci show shadowsocks-rust.ss_rules` and then try in System->OpenMPTCProuter, "Advanced settings" tab, uncheck the box "Restrict proxy to LAN zone" if checked and wait 5 minutes.

Hello, I need to update the package list, but since I am in China, I require a Hong Kong VPS to do so effectively. The issue was identified when I connected to the VPS.  

Additionally, I have tried the solution of unchecking "Restrict proxy to LAN zone" in **System->OpenMPTCProuter** under the **"Advanced settings"** tab and waiting for 5 minutes. Unfortunately, the issue persists.  

Currently, whenever I enable the VPS, I experience a "false network state." The computer recognizes the connection as active, but in reality, web pages and remote software fail to work.

a834003831, 2024-11-23:
This is the data I obtained by entering the command before connecting to the VPS
root@OpenMPTCProuter:~# uci show shadowsocks-rust.ss_rules
shadowsocks-rust.ss_rules=ss_rules
shadowsocks-rust.ss_rules.disabled='0'
shadowsocks-rust.ss_rules.redir_tcp='all'
shadowsocks-rust.ss_rules.src_default='forward'
shadowsocks-rust.ss_rules.dst_default='forward'
shadowsocks-rust.ss_rules.local_default='forward'
shadowsocks-rust.ss_rules.server='sss0'
shadowsocks-rust.ss_rules.ifnames='lo' 'eth1' 'tun0'

a834003831, 2024-11-23:
I have tested multiple VPS providers, and the issue of a pseudo-network persists as long as a VPS is connected. This means that the computer recognizes the network as accessible, but in reality, websites cannot be opened, and all remote tools fail. I will continue to monitor the release of new firmware and will flash and test it as soon as possible. Once again, my deepest gratitude to the brilliant developer.

Ysurac, 2024-11-23:
It seems there is some issue on current snapshot. I will fix it after the week end.

a834003831, 2024-11-23:
> 当前快照似乎存在一些问题。我会在周末之后修理它。

"Okay, thank you for your reply. I will continue to follow the new version. Wishing you a pleasant Sunday!"

a834003831, 2024-11-24:
> It seems there is some issue on current snapshot. I will fix it after the week end.

Hello, I saw that you updated the firmware today. I immediately tested it and recorded it. Now I am giving you feedback
When I use the stable version VPS, it shows this interface:
[image]
And I checked the public IP address through curl ipinfo. io/ip, and the computer has obtained the VPS IP address. The remote software is working normally (the SSH router has also used the VPS IP address), but the webpage cannot be opened. However, my SSH router pinged the webpage and it worked normally

Operated:1System->OpenMPTCProuter, Wizard tab, "Advanced settings" checkbox, China in country may help
2System->OpenMPTCProuter, "Advanced settings" tab, uncheck the box "Restrict proxy to LAN zone" if checked and wait 5 minutes.
root@OpenMPTCProuter:~# uci show shadowsocks-rust.ss_rules
shadowsocks-rust.ss_rules=ss_rules
shadowsocks-rust.ss_rules.disabled='0'
shadowsocks-rust.ss_rules.redir_tcp='all'
shadowsocks-rust.ss_rules.src_default='forward'
shadowsocks-rust.ss_rules.dst_default='forward'
shadowsocks-rust.ss_rules.local_default='forward'
shadowsocks-rust.ss_rules.server='sss0'

Problem 2: Updating software package error
Collected errors: * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-snapshot-6.6/r4s/targets/rockchip/armv8/packages/Packages.gz, wget returned 8. * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-snapshot-6.6/r4s/packages/aarch64_generic/base/Packages.gz, wget returned 8. * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-snapshot-6.6/r4s/packages/aarch64_generic/luci/Packages.gz, wget returned 8. * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-snapshot-6.6/r4s/packages/aarch64_generic/openmptcprouter/Packages.gz, wget returned 8. * opkg_download: Failed to download https://download.openmptcprouter.com/release/v0.62-

Summary: I have tested multiple VPS providers (both stable and the latest snapshots). Currently, it still faces several issues in China. However, I don't know how to provide the data you need when problems arise. If you're willing, I hope to add your contact tool to communicate in real time and test what you need.

a834003831, 2024-11-24:
> It seems there is some issue on current snapshot. I will fix it after the week end.

I switched the default proxy to "XRay Shadowsocks 2022," and it seems to be working fine, with webpages loading properly. However, the packages still cannot be updated, preventing me from downloading the drivers and testing the aggregation effect. Could you provide a download link for the driver so I can manually upload and install it?

a834003831, 2024-11-24:
The latest update: I replaced the original software source from `v0.62-snapshot-6.6` to `v61-6.6`. While I found the driver package, unfortunately, it is not compatible with version 6.6.59.

a834003831, 2024-11-24:
There has been a new breakthrough. I have switched the software source from v0.62-snapshot-6.6 to v0.61-snapshot-6.6, and successfully installed the driver. I also changed the default proxy, and the web page is now accessible. However, when plugging in the USB network card, it fails to start properly. I need to unplug it, boot up, and then plug it back in. Additionally, I found that after plugging in the network card, it cannot run for long and will crash automatically. SSH becomes unavailable, and the backend (192.168.100.1) cannot be accessed.

Ysurac, 2024-11-24:
I fixed the link for packages.
For Shadowsocks-Rust, it seems to be working correctly now in my case. You can check log from VPS, with ```journalctl -u shadowsocks-go```

Ysurac, 2024-11-24:
Did you try to use a powered USB hub ?

a834003831, 2024-11-24:
> Did you try to use a powered USB hub ?

I am currently using a powered USB hub.Just now, the webpage failed to open again. I am switching between multiple default proxies for testing.

a834003831, 2024-11-24:
> I fixed the link for packages. For Shadowsocks-Rust, it seems to be working correctly now in my case. You can check log from VPS, with `journalctl -u shadowsocks-go`

root@ecm-7fad:~# journalctl -u shadowsocks-go
Nov 21 10:22:13 ecm-7fad systemd[1]: /lib/systemd/system/shadowsocks-go.service:3: Invalid URL, ignoring: A
Nov 21 10:22:13 ecm-7fad systemd[1]: /lib/systemd/system/shadowsocks-go.service:3: Invalid URL, ignoring: versatile
Nov 21 10:22:13 ecm-7fad systemd[1]: /lib/systemd/system/shadowsocks-go.service:3: Invalid URL, ignoring: and
Nov 21 10:22:13 ecm-7fad systemd[1]: /lib/systemd/system/shadowsocks-go.service:3: Invalid URL, ignoring: efficient
Nov 21 10:22:13 ecm-7fad systemd[1]: /lib/systemd/system/shadowsocks-go.service:3: Invalid URL, ignoring: proxy
Nov 21 10:22:13 ecm-7fad systemd[1]: /lib/systemd/system/shadowsocks-go.service:3: Invalid URL, ignoring: platform
Nov 21 10:22:13 ecm-7fad systemd[1]: /lib/systemd/system/shadowsocks-go.service:3: Invalid URL, ignoring: for
Nov 21 10:22:13 ecm-7fad systemd[1]: /lib/systemd/system/shadowsocks-go.service:3: Invalid URL, ignoring: secure
Nov 21 10:22:13 ecm-7fad systemd[1]: /lib/systemd/system/shadowsocks-go.service:3: Invalid URL, ignoring: communications.
Nov 21 10:22:14 ecm-7fad systemd[1]: Started shadowsocks-go.service - shadowsocks-go Service.
Nov 21 10:22:14 ecm-7fad shadowsocks-go[112242]: 2024-11-21T10:22:14.109+0800 FATAL Failed to load config {"confPath": "/e>
Nov 21 10:22:14 ecm-7fad systemd[1]: shadowsocks-go.service: Main process exited, code=exited, status=1/FAILURE
Nov 21 10:22:14 ecm-7fad systemd[1]: shadowsocks-go.service: Failed with result 'exit-code'.
Nov 21 10:22:14 ecm-7fad systemd[1]: shadowsocks-go.service: Scheduled restart job, restart counter is at 1.
Nov 21 10:22:14 ecm-7fad systemd[1]: Stopped shadowsocks-go.service - shadowsocks-go Service.
Nov 21 10:22:14 ecm-7fad systemd[1]: Started shadowsocks-go.service - shadowsocks-go Service.
Nov 21 10:22:14 ecm-7fad shadowsocks-go[112256]: 2024-11-21T10:22:14.383+0800 FATAL Failed to load config {"confPath": "/e>
Nov 21 10:22:14 ecm-7fad systemd[1]: shadowsocks-go.service: Main process exited, code=exited, status=1/FAILURE
Nov 21 10:22:14 ecm-7fad systemd[1]: shadowsocks-go.service: Failed with result 'exit-code'.
Nov 21 10:22:14 ecm-7fad systemd[1]: shadowsocks-go.service: Scheduled restart job, restart counter is at 2.
Nov 21 10:22:14 ecm-7fad systemd[1]: Stopped shadowsocks-go.service - shadowsocks-go Service.
Nov 21 10:22:14 ecm-7fad systemd[1]: Started shadowsocks-go.service - shadowsocks-go Service.
Nov 21 10:22:14 ecm-7fad shadowsocks-go[112261]: 2024-11-21T10:22:14.623+0800 FATAL Failed to load config {"confPath": "/e>
Nov 21 10:22:14 ecm-7fad systemd[1]: shadowsocks-go.service: Main process exited, code=exited, status=1/FAILURE
Nov 21 10:22:14 ecm-7fad systemd[1]: shadowsocks-go.service: Failed with result 'exit-code'.
Nov 21 10:22:14 ecm-7fad systemd[1]: shadowsocks-go.service: Scheduled restart job, restart counter is at 3.
Nov 21 10:22:14 ecm-7fad systemd[1]: Stopped shadowsocks-go.service - shadowsocks-go Service.
Nov 21 10:22:14 ecm-7fad systemd[1]: Started shadowsocks-go.service - shadowsocks-go Service.
lines 1-28

a834003831, 2024-11-24:
> What do you have in `/etc/shadowsocks-go/server.json` on the VPS ?

root@ecm-7fad:~# cat /etc/shadowsocks-go/server.json
{
    "servers": [
        {
            "name": "ss-2022",
            "protocol": "2022-blake3-aes-256-gcm",
            "tcpListeners": [
            {
                "network": "tcp",
                "address": ":65280",
                "fastOpen": true,
                "reusePort": false,
                "multipath": true
            }
            ],
            "enableTCP": true,
            "listenerTFO": true,
            "enableUDP": true,
            "mtu": 1500,
            "psk": "bVJcpecz6Nin60941V31Dxwx9vFw5QsJmxIGt29ZasI=",
            "uPSKStorePath": "/etc/shadowsocks-go/upsks.json"
        }
    ],
    "stats": {
        "enabled": true
    },
    "api": {
        "enabled": true,
        "debugPprof": false,
        "enableTrustedProxyCheck": false,
        "trustedProxies": [],
        "proxyHeader": "X-Forwarded-For",
        "listen": "127.0.0.1:65279",
        "certFile": "",
        "keyFile": "",
        "clientCertFile": "",
        "secretPath": "",
        "fiberConfigPath": ""
    }

Ysurac, 2024-11-24:
The config file seems ok. What do you have in ```/usr/lib/systemd/system/shadowsocks-go.service``` ?

a834003831, 2024-11-24:
> The config file seems ok. What do you have in `/usr/lib/systemd/system/shadowsocks-go.service` ?

}root@ecm-7fad:~# cat /usr/lib/systemd/system/shadowsocks-go.service
[Unit]
Description=shadowsocks-go Service
Documentation=A versatile and efficient proxy platform for secure communications. 
After=network.target nss-lookup.target
Wants=network-online.target

[Service]
User=root
CapabilityBoundingSet=CAP_NET_ADMIN CAP_NET_BIND_SERVICE CAP_NET_RAW
AmbientCapabilities=CAP_NET_ADMIN CAP_NET_BIND_SERVICE CAP_NET_RAW
NoNewPrivileges=true
ExecStart=/usr/bin/shadowsocks-go -confPath /etc/shadowsocks-go/server.json -logLevel error
Restart=always
RestartPreventExitStatus=23
StartLimitInterval=0

[Install]
WantedBy=multi-user.target

Ysurac, 2024-11-24:
In the config file, on your copy paste it's missing last "}" or it's really not here (and should be) ?
Else I will try a fresh install on a VPS tomorrow (in a few hours here)

a834003831, 2024-11-24:
> In the config file `/etc/shadowsocks-go/server.json`, on your copy paste it's missing last "}" or it's really not here (and should be) ? Else I will try a fresh install on a VPS tomorrow (in a few hours here)

[image]
[image]

a834003831, 2024-11-24:
I translated your reply, but I can't fully understand it. I'll take a screenshot and send it to you so we can clearly analyze the issue.

Ysurac, 2024-11-24:
It's really missing last } in /etc/shadowsocks-go/server.json, no idea how this can be possible...
Add the character "}" at the end of the file ```/etc/shadowsocks-go/server.json``` on a new line, and it should work.

a834003831, 2024-11-24:
> It's really missing last } in /etc/shadowsocks-go/server.json, no idea how this can be possible... Add the character "}" at the end of the file `/etc/shadowsocks-go/server.json` on a new line, and it should work.

"I used the `vi` command and saw that the file originally had this symbol at the end. Now, I have added another one, but it appears white. Is this correct?"
[image]

a834003831, 2024-11-24:
"It shouldn't be necessary to add this symbol again, right? I think it's just an issue with the SSH software that's preventing it from displaying correctly."

Ysurac, 2024-11-24:
Yes, only one is needed, strange it was not in screenshot and paste...
I will try tomorrow on a fresh install and check what can be the Shadowsocks-go issue, as it's not that.

a834003831, 2024-11-24:
> Yes, only one is needed, strange it was not in screenshot and paste... I will try tomorrow on a fresh install and check what can be the Shadowsocks-go issue, as it's not that.

"Okay, thank you for your hard work, great developer. I will step away from my computer now as it's already past midnight here in China. If you need to test anything, feel free to reply under my post anytime, and I will respond to each of your messages as soon as possible."

a834003831, 2024-11-25:
> Yes, only one is needed, strange it was not in screenshot and paste... I will try tomorrow on a fresh install and check what can be the Shadowsocks-go issue, as it's not that.

Hello, I noticed that when my friends and I pull the VPS image, it’s version 6.6.36, but the router is running version 6.6.59. Could this be the reason why our tests are not going smoothly?
[image]

Ysurac, 2024-11-25:
There is no problem with kernel, for 6.6 router version, it only need to be > 6.6.0 on the VPS.
For now I 'm not able to reproduce the Shadowsocks-go issue on VPS, no idea why it's not starting... I'm still trying.

a834003831, 2024-11-25:
> There is no problem with kernel, for 6.6 router version, it only need to be > 6.6.0 on the VPS. For now I 'm not able to reproduce the Shadowsocks-go issue on VPS, no idea why it's not starting... I'm still trying.

Alright, we’re not sure what actions we can take to assist you, but we still want to thank you. If there’s anything you need us to test, feel free to reach out to me anytime. My friends and I are more than happy to provide you with data.

a834003831, 2024-11-27:
> I have no issues on Shadowsocks-go with a fresh install on a Digital Ocean VPS on Debian 12. Did you try a full new install ?

Hello, I noticed that you released the latest router firmware this morning. I've already re-flashed it and made a fresh installation of the VPS. Currently, the only problem I've found is that when the VPS doesn't take effect, the page of "Aggregated Routes - Status" fails to refresh. At that time, the network had already automatically switched to WAN1.

a834003831, 2024-11-30:
> [[image]](/Ysurac) [Ysurac](/Ysurac) mentioned this issue [ Nov 27, 2024 ](#ref-issue-2697792704)
> [ OMR direct output ip not use VPS ip #3686](/Ysurac/openmptcprouter/issues/3686)

Hello, when I use my network daily, 80% of my traffic is UDP. How should I configure my router? For example, how should I set the default proxy, and what solution should I use for MPTCP?

closed 2025-03-05 by github-actions[bot]