Ysurac/openmptcprouter #3684: Aggregate seems doesn't work

Amrizamaa, 2024-11-26:
## Expected Behavior

It should work with aggregation and have like 50mbp/s debit minimum

## Current Behavior

on my side it's seems the aggregate doesn't work properly, like for now the vps can't get the public ip address and it seems only have my adsl works as maximum debit
i have a message like the vps  can't get public ip address

[image]
[image]

When i'm only on wi-fi of my 5G i have 50mbp/s but when i'm wired to openmptcp router i have no more of 5mbp/s

[image]

## Specifications

  - OpenMPTCProuter version: 6.6.50
  - OpenMPTCProuter VPS version: 0.1031 / 6.6.36
  - OpenMPTCProuter VPS provider: Iglao / Skorpia (2vps)
  - OpenMPTCProuter platform: x64 theorically

Ysurac, 2024-11-26:
As indicated on screenshot, fix the time on the router. As there is 1 hour of difference, I think you set the time without setting the timezone in System->System

Ysurac, 2024-11-26:
Can you take a screenshot of status page when time is fixed ?
What is the result of ```journalctl -u shadowsocks-go``` on the VPS ?

Amrizamaa, 2024-11-26:
[image]

https://pastebin.com/gpX6eWEZ

only copied last minute sorry for my time response

Ysurac, 2024-11-26:
What do you have in Status->System log ?

Amrizamaa, 2024-12-28:
This issue is solved without any modification idk what happened

closed 2024-12-28 by Amrizamaa