Ysurac/openmptcprouter #3686: OMR direct output

lhc1993, 2024-11-27:
## Expected Behavior

I set the IP address of the VPS. However, in multiple OMRs supported by the SIM, the directoutput IP address changes to the IP address of the corresponding card each time the card is inserted.
## Current Behavior

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

lhc1993, 2024-11-27:
OpenMPTCProuter version: self-compilation 6.6
OpenMPTCProuter VPS version: 5.4.207-mptcp

lhc1993, 2024-11-27:
[image]

Ysurac, 2024-11-27:
Yes, it's using direct output, so not via VPS.
You updated VPS ? Can you do a ```uname -a``` on the VPS ?

lhc1993, 2024-11-27:
vps确认过了，有时是direct output，有时是vps ip

lhc1993, 2024-11-27:
VPS confirmed, sometimes it's direct output, sometimes it's VPS IP

lhc1993, 2024-11-27:
Linux iZ2ze9o5glccm4dcv0c5mpZ 5.4.207-mptcp #1 SMP Sun Jul 24 14:39:44 UTC 2022 x86_64 GNU/Linux

lhc1993, 2024-11-27:
[image]

lhc1993, 2024-11-27:
After the update, inserting the card still displays direct output

lhc1993, 2024-11-27:
[image]

lhc1993, 2024-11-27:
[image]

Ysurac, 2024-11-27:
same error as #3678 
Shadowsocks-go doesn't start. (the invalid URL is not the cause)
What is the result of ```ps aux | grep shadow``` ?

lhc1993, 2024-11-27:
[image]

lhc1993, 2024-11-27:
[image]

lhc1993, 2024-11-27:
[image]

lhc1993, 2024-11-27:
As long as I unplug the SIM card or WAN port cable and plug it back in, this address will become the IP address of the newly added device

Ysurac, 2024-11-27:
So this doesn't change anything after save ? This command retrieve all keys again (it was if shadowsocks-rust key was missing)
On the router, what is the result of ```uci show shadowsocks-rust``` and what you have in ```Status->System log``` ?

lhc1993, 2024-11-27:
[image]

lhc1993, 2024-11-27:
When the IP address changes, this print will appear

lhc1993, 2024-11-27:
Then the IP address will become the corresponding detected IP address

Ysurac, 2024-11-27:
This seems normal. It's more logs about shadowsocks that I would like

lhc1993, 2024-11-27:
There are no logs left behind, it stays on that line

lhc1993, 2024-11-27:
After two minutes, it changed again, sometimes it can take a while to change

Ysurac, 2024-11-27:
You can scroll up to see previous logs. Use use "logread" via SSH on the router.
I would also need ```uci show shadowsocks-rust``` via SSH on the router.
I don't need more screenshot while it's same error.

lhc1993, 2024-11-27:
root@OpenMPTCProuter:~# uci show shadowsocks-rust
shadowsocks-rust.hi1=ss_redir
shadowsocks-rust.hi1.server='sss0'
shadowsocks-rust.hi1.local_address='::'
shadowsocks-rust.hi1.local_port='1100'
shadowsocks-rust.hi1.mode='tcp_and_udp'
shadowsocks-rust.hi1.timeout='1000'
shadowsocks-rust.hi1.fast_open='0'
shadowsocks-rust.hi1.verbose='0'
shadowsocks-rust.hi1.syslog='1'
shadowsocks-rust.hi1.reuse_port='0'
shadowsocks-rust.hi1.mptcp='1'
shadowsocks-rust.hi1.ipv6_first='1'
shadowsocks-rust.hi1.no_delay='0'
shadowsocks-rust.hi1.disabled='0'
shadowsocks-rust.hi2=ss_redir
shadowsocks-rust.hi2.server='sss1'
shadowsocks-rust.hi2.local_address='::'
shadowsocks-rust.hi2.local_port='1101'
shadowsocks-rust.hi2.mode='tcp_and_udp'
shadowsocks-rust.hi2.timeout='1000'
shadowsocks-rust.hi2.fast_open='0'
shadowsocks-rust.hi2.verbose='0'
shadowsocks-rust.hi2.syslog='1'
shadowsocks-rust.hi2.reuse_port='0'
shadowsocks-rust.hi2.mptcp='1'
shadowsocks-rust.hi2.ipv6_first='1'
shadowsocks-rust.hi2.no_delay='0'
shadowsocks-rust.ss_rules=ss_rules
shadowsocks-rust.ss_rules.disabled='0'
shadowsocks-rust.ss_rules.redir_tcp='all'
shadowsocks-rust.ss_rules.src_default='forward'
shadowsocks-rust.ss_rules.dst_default='forward'
shadowsocks-rust.ss_rules.local_default='forward'
shadowsocks-rust.ss_rules.server='sss0'
shadowsocks-rust.ss_rules.ifnames='eth0'
shadowsocks-rust.sss0=server
shadowsocks-rust.sss0.disabled='0'
shadowsocks-rust.sss0.server='121.199.5.55'
shadowsocks-rust.sss0.server_port='65280'
shadowsocks-rust.sss0.method='2022-blake3-aes-256-gcm'
shadowsocks-rust.sss0.obfs='0'
shadowsocks-rust.sss0.password='tTf3KMHA3Sk1BU6Omo1LdAWWTKnpG9rSFRThO2RZO48=:F6fmM790Y7ebj4CxEMW4OcNfG2xqlUj0CYzW6yB6eL0='
shadowsocks-rust.sss1=server
shadowsocks-rust.sss1.disabled='1'
shadowsocks-rust.sss1.server='121.199.5.55'
shadowsocks-rust.sss1.server_port='65280'
shadowsocks-rust.sss1.method='2022-blake3-aes-256-gcm'
shadowsocks-rust.sss1.obfs='0'
shadowsocks-rust.sss1.password='tTf3KMHA3Sk1BU6Omo1LdAWWTKnpG9rSFRThO2RZO48=:F6fmM790Y7ebj4CxEMW4OcNfG2xqlUj0CYzW6yB6eL0='
shadowsocks-rust.dns=ss_tunnel
shadowsocks-rust.dns.disabled='1'
shadowsocks-rust.dns.mode='tcp_and_udp'
shadowsocks-rust.dns.server='sss0'
shadowsocks-rust.dns.local_port='5353'
shadowsocks-rust.dns.tunnel_address='8.8.8.8:53'
shadowsocks-rust.tracker_sss0=ss_local
shadowsocks-rust.tracker_sss0.server='sss0'
shadowsocks-rust.tracker_sss0.local_address='127.0.0.1'
shadowsocks-rust.tracker_sss0.local_port='1111'
shadowsocks-rust.tracker_sss0.mode='tcp_and_udp'
shadowsocks-rust.tracker_sss0.timeout='600'
shadowsocks-rust.tracker_sss0.fast_open='0'
shadowsocks-rust.tracker_sss0.syslog='0'
shadowsocks-rust.tracker_sss0.reuse_port='1'
shadowsocks-rust.tracker_sss0.mptcp='1'
shadowsocks-rust.tracker_sss0.verbose='0'

lhc1993, 2024-11-27:
Are there any issues with these configurations

Ysurac, 2024-11-27:
Can you try from VPS, to run ```curl -I 212.27.48.10``` (it's one IP used to check is proxy is working or not) ?

lhc1993, 2024-11-27:
It's a problem with the firewall. Thank you very much. I was able to connect after turning off the firewall. Thank you very much

Ysurac, 2024-11-27:
ok thanks for the feedback.
What was blocked by the firewall ? All ports or some ? (maybe I can add a check to detect the problem)

lhc1993, 2024-11-28:
Some ports, not all ports, need to be debugged. Currently, the network interface that is not connected is omr64bin.

lhc1993, 2024-11-28:
“Nov 28 17:05:35 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later”
At present, it appears to be this error, but I don't know how to troubleshoot it

lhc1993, 2024-11-29:
Can you help under the guidance? What service is used to obtain the public IP address from the server? Is it possible to connect to the server as long as the port of the server is connected?

Ysurac, 2024-11-29:
you can check via https://<vpsip>:65500/ if you get an answer, if yes VPS API can be reach.
Check also that correct Admin Key is set for the VPS.

closed 2025-03-04 by github-actions[bot]