Ysurac/openmptcprouter #3687: No external NAS access with Shadowsocks

3dzug, 2024-11-27:
I am using a NAS attached to my network. When I am using Shadowsocks protocol I do not get access from external to the NAS. If I change to XRAY Fless I do get access (all ports are set). However some applications like Anyviewer do not get a connection. If I change back to Shadowsocks it is working for the apps but fails the external access.

3dzug, 2024-11-30:
Do I need to be more detailed here?

I tried several times again but as soon as I change to Shadowsocks back it seems that the port forwarding fails too.

alchool, 2024-12-13:
which version are you using? 
I usually connect to NAS with an external VPN, openvpn. I upgraded to 0.61 and I couldn't download or open any files.  Either with shadowsocks-libev up or disable.
So I resumed 0.60 backup image and Downloads works again. 
With SS running on vm logs I get a lot of 
 __ss-server[239776]: getpeername: Transport endpoint is not connected
 ss-server[239776]: remote recv: Connection reset by peer__

closed 2025-03-18 by github-actions[bot]