Ysurac/openmptcprouter #3690: Open MPTCP Router Installation, Works With One PC But Not As Routers WAN, Kernel Panic?

z4cth3p4c, 2024-11-28:
Expected Behavior

When the OpenMPTCProuter is connected to the network, it should successfully bond three connections together and provide the cumulative speed to the rest of the network through the router, whether connected directly to a laptop or via another router. The system should remain stable and operational without kernel panics.

Current Behavior

The OpenMPTCProuter successfully bonds three connections and delivers the cumulative speed when directly connected to a laptop. However, when the LAN cable from the OpenMPTCProuter is plugged into another router as the primary WAN, the OpenMPTCP hardware's LAN card lights turn off, and the hardware attempts to power cycle. An error is displayed indicating a kernel panic when the OpenMPTCP hardware is connected to the rest of the network.

Specifications

OpenMPTCProuter version: v0.61-6.6-r0+27346-c7ba5574f5-x86-64-generic-ext4-combined-efi.img.gz

OpenMPTCProuter VPS version: Kernel 6.6

OpenMPTCProuter VPS provider: Kamatera

OpenMPTCProuter platform: x86_6

Ysurac, 2024-11-28:
Even without traffic ?
What is the hardware used exactly or at least what is the network chipset ?

z4cth3p4c, 2024-11-28:
We have tested an I350-T4 quad-port NIC and two dual-port Intel NICs, but all exhibited the same symptoms. The dual-port NICs were based on the Intel 82575/82576 chipset. We have not yet tested the LAN functionality independently by disconnecting the load balancer's internet connection, but this would be a valuable test to perform. Thank you for the expedited response.

Ysurac, 2024-11-28:
Can you capture the kernel panic ?
Can you test with latest snapshot image (to check if kernel update fix this...) ? https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

z4cth3p4c, 2024-11-29:
@Ysurac,

Thank you for your prompt response. We have attached the kernel panic error as captured from the system (see the image).

We conducted the test as suggested:

LAN functionality: Works fine when the load balancer's internet connection is disconnected.
WAN functionality: The kernel panic occurs as soon as we add the internet connection to the WAN.
We are currently testing the latest snapshot version to determine if the kernel update resolves the issue. We’ll report back with the results once testing is complete.

[image]

z4cth3p4c, 2024-11-29:
We have successfully installed the latest snapshot image and this resolved the issue.

Ysurac, 2024-11-29:
Cool, thanks for the feedback. So this will be resolved in next release.