Ysurac/openmptcprouter #3692: FTP Files corrupted with Xray Vless

3dzug, 2024-11-29:
When using XRay Vless with Gloryrun UPD (also with off) uploaded compressed files like rar are corrupted.

Repro:
1. Select XRay Vless , Gloryrun UDP
2. Create a rar archive
3. Upload with FTP
4. Download again
5. Extract

Expected Behavior:
files should be extracted

Actual Behavior:
Archive Corrupted, (Unexptected End of Archive)

If I change to Shadowsocks I dont have this problem.

openmptcprouter v0.61-6.6 r0

Ysurac, 2024-11-29:
Please use template

3dzug, 2024-11-29:
@Ysurac  Template attached. Please let me know if you want ftp access for testing.

## Expected Behavior

An Uploaded ftp file should have the exact byte size than the original file and should be not corrupted.

## Current Behavior

Using XRay Vless or other protocols except Shadowsocks the filesize is not exact and the downloaded file is corrupt.

Nov 29 21:11:22 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.160 f4:34:f0:70:1c:40 
Nov 29 21:11:22 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.160 f4:34:f0:70:1c:40 BuroF43F0701C40
Nov 29 21:13:53 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.212 0a:b4:fe:6b:02:e3 
Nov 29 21:13:53 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.212 0a:b4:fe:6b:02:e3 repeaterbuero
Nov 29 21:25:51 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.124 d6:ca:ad:56:3a:fa 
Nov 29 21:25:51 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.124 d6:ca:ad:56:3a:fa iPhone
Nov 29 21:26:32 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.231 4e:5f:49:5c:16:a3 
Nov 29 21:26:32 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.231 4e:5f:49:5c:16:a3 
Nov 29 21:27:03 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.231 4e:5f:49:5c:16:a3 
Nov 29 21:27:03 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.231 4e:5f:49:5c:16:a3 
Nov 29 21:27:50 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.124 d6:ca:ad:56:3a:fa 
Nov 29 21:27:50 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.124 d6:ca:ad:56:3a:fa iPhone
Nov 29 21:27:50 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.124 d6:ca:ad:56:3a:fa 
Nov 29 21:27:50 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.124 d6:ca:ad:56:3a:fa iPhone
Nov 29 21:28:31 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.231 4e:5f:49:5c:16:a3 
Nov 29 21:28:31 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.231 4e:5f:49:5c:16:a3 
Nov 29 21:28:33 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.231 4e:5f:49:5c:16:a3 
Nov 29 21:28:33 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.231 4e:5f:49:5c:16:a3 
Nov 29 21:30:45 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.206 e8:aa:cb:94:9e:f7 
Nov 29 21:30:45 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.206 e8:aa:cb:94:9e:f7 Samsung
Nov 29 21:41:22 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.117 9a:0f:51:2d:d5:f9 
Nov 29 21:41:22 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.117 9a:0f:51:2d:d5:f9 
Nov 29 21:44:56 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.124 d6:ca:ad:56:3a:fa 
Nov 29 21:44:56 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.124 d6:ca:ad:56:3a:fa iPhone
Nov 29 21:46:09 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.228 74:56:3c:b5:0c:77 
Nov 29 21:46:09 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.228 74:56:3c:b5:0c:77 Matthias-NG
Nov 29 21:49:46 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(LanBridge) 192.168.100.216 02:4e:98:e3:fe:5a 
Nov 29 21:49:46 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(LanBridge) 192.168.100.216 02:4e:98:e3:fe:5a 
Nov 29 21:52:21 OpenMPTCProuter daemon.err uhttpd[7329]: [info] luci: accepted login on / for root from 192.168.100.228

## Possible Solution

Using Shadowsocks does not produce this issue

## Steps to Reproduce the Problem

  1. Set Proxy to XRay Vless and VPN to Glorytun UDP
  2. Create a 250 mb zip and rar file with winrar
  3. Upload it via FTP to a server
  4. Download the file
  5. Check different filesizes
  6. Try to extract ( I can't)
  

## Context (Environment)

I try to upload a compressed archive

If you cant create this problem please message me and I'll give you access to my server and you can test.

## Specifications

  - OpenMPTCProuter version: 0.61-6.6
  - OpenMPTCProuter VPS version: 0.1031 6.6.36y64v3-xanmod1
  - OpenMPTCProuter VPS provider: HT-Hosting
  - OpenMPTCProuter platform: x86_64
  - Country: Germany

Ysurac, 2024-11-30:
Can you test latest snapshot: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots (on both VPS and router) ?
If it's still not work, can you give me the result of ```journalctl -u xray``` on the timestamp you try FTP upload ?

3dzug, 2024-11-30:
Is there a description how I have to install the router snapshot? I have no skills on that topic.

Ysurac, 2024-11-30:
Exactly like others images, but using the links in the previous link. Via System->Firmware update

3dzug, 2024-12-22:
Hi, unfortunately my skills dont fit to perform this task. I tried twice to setup all but I failed as one time I did not get access to the local router and second time the vpn tunnel was not created. 
I restored one of my configuration so maybe the keys were wrong as I set the vps up again brand new.
I also do not know if the current image caused this or me.
So I'd wait until the new version is out. 
To test it I need to know if I can update the existing vps with the new script and get all keys remained or I need to know where all  the keys were located as it seems that they were not updated with the new key.
Sorry.

Ysurac, 2024-12-22:
In System->OpenMPTCProuter, Wizard tab, "advanced settings" checkbox, you can force key retrieve.
When you update VPS key are still the same (but new key can exist if you update from very old version)

3dzug, 2024-12-22:
So I can just do an vps upgrade in the existing machine and do not need to reset the vps first and install again?Matthias

Ysurac, 2024-12-22:
yes

3dzug, 2024-12-23:
The issue is not present on the latest snapshot. Thanks for fixing.

closed 2024-12-23 by 3dzug