Ysurac/openmptcprouter #3693: OMR-Bypass doesn't work for subdomains (or only works for top-level domains)

Schinkentanz, 2024-12-01:
## Expected Behavior
When using OMR-Bypass with `netflix` enabled in "Protocols and services" or `nflxvideo.net` configured as a domain in "Domains", the bypass should work not only for the top-level domain, but also for any subdomain (e.g. `random.sub.domain.name.nflxvideo.net`).

> [!NOTE]
> Netflix is just an example, it's the same for any other subdomain.
> It also doesn't work, if only one bypass option is used.

## Current Behavior
When running `traceroute nflxvideo.net` on a device where the MPTCP router is configured as the DNS server, the request is correctly routed through the configured master interface:
```
❯ traceroute nflxvideo.net
traceroute: Warning: nflxvideo.net has multiple addresses; using 107.20.175.192
traceroute to nflxvideo.net (107.20.175.192), 64 hops max, 40 byte packets
 1  192.168.0.1 (192.168.0.1)  6.355 ms  3.000 ms  2.871 ms
 2  openmptcprouter.lan (192.168.42.1)  3.322 ms  3.011 ms  3.210 ms
 3  192.168.178.1 (192.168.178.1)  6.758 ms  5.734 ms  5.293 ms
.... <redacted>
```

When running `traceroute ipv4-c088-ord001-dev-ix.1.oca.nflxvideo.net` on the same device, the request will not be routed through the configured master interface, but will use the VPS:

```
❯ traceroute ipv4-c088-ord001-dev-ix.1.oca.nflxvideo.net
traceroute to ipv4-c088-ord001-dev-ix.1.oca.nflxvideo.net (198.38.109.219), 64 hops max, 40 byte packets
 1  192.168.0.1 (192.168.0.1)  6.151 ms  3.118 ms  2.984 ms
 2  openmptcprouter.lan (192.168.42.1)  3.307 ms  3.451 ms  4.952 ms
 3  10.255.252.1 (10.255.252.1)  18.369 ms  17.544 ms  17.893 ms
.... <redacted>
```

## Steps to Reproduce the Problem
> [!NOTE]
> Tested at time of issue creation
  1. Install the latest snapshot on a fresh VPS (as described in [Wiki](https://github.com/ysurac/openmptcprouter/wiki/Snapshots))
  2. Install the latest [snapshot squashfs image for RPI5](https://snapshots.openmptcprouter.com/6.6/rpi5/targets/bcm27xx/bcm2712/) on an empty SD card
  3. Setup router in a minimal way
    1. Add server IP & key
    2. Add 2 WAN interfaces
    3. Setup OMR-Bypass as described above
    4. Use `traceroute` on any connected client

## Specifications
  - OpenMPTCProuter version: `openmptcprouter-v0.62-snapshot-6.6-r0+28016-48028cd102-bcm27xx-bcm2712-rpi-5-squashfs-factory`
  - OpenMPTCProuter VPS version: `0.1032-test 6.6.36-x64v2-xanmod1`
  - OpenMPTCProuter platform: RPI5

Ysurac, 2024-12-02:
Are you sure you are using only OpenMPTCProuter IP address as DNS ?

Ysurac, 2024-12-03:
What is the result of ```/etc/init.d/omr-bypass restart``` and ```/etc/init.d/firewall restart``` via SSH on the router ?
I would also need the result of ```nft list ruleset```

Schinkentanz, 2024-12-03:
Sure, thanks for taking a closer look!

```
root@OpenMPTCProuter:~# /etc/init.d/omr-bypass restart
root@OpenMPTCProuter:~#
```

```
root@OpenMPTCProuter:~# /etc/init.d/firewall restart
Section omr_dst_bypass_eth0_dstip_4 (omr_dst_bypass_eth0_rule) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstip_4_accept (omr_dst_bypass_eth0_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_eth0_srcip_4 (omr_dst_bypass_eth0_srcip) is disabled, ignoring section
Section omr_dst_bypass_eth0_mac_4 (omr_dst_bypass_eth0_mac) is disabled, ignoring section
Section omr_dst_bypass_eth0_srcport_tcp_4 (omr_dst_bypass_eth0_srcport_tcp) is disabled, ignoring section
Section omr_dst_bypass_eth0_srcport_udp_4 (omr_dst_bypass_eth0_srcport_udp) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstport_tcp_4 (omr_dst_bypass_eth0_dstport_tcp) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstport_udp_4 (omr_dst_bypass_eth0_dstport_udp) is disabled, ignoring section
Section omr_dst_bypass_wan1_dstip_4 (omr_dst_bypass_wan1_rule) is disabled, ignoring section
Section omr_dst_bypass_wan1_dstip_4_accept (omr_dst_bypass_wan1_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_wan1_srcip_4 (omr_dst_bypass_wan1_srcip) is disabled, ignoring section
Section omr_dst_bypass_wan1_mac_4 (omr_dst_bypass_wan1_mac) is disabled, ignoring section
Section omr_dst_bypass_wan1_srcport_tcp_4 (omr_dst_bypass_wan1_srcport_tcp) is disabled, ignoring section
Section omr_dst_bypass_wan1_srcport_udp_4 (omr_dst_bypass_wan1_srcport_udp) is disabled, ignoring section
Section omr_dst_bypass_wan1_dstport_tcp_4 (omr_dst_bypass_wan1_dstport_tcp) is disabled, ignoring section
Section omr_dst_bypass_wan1_dstport_udp_4 (omr_dst_bypass_wan1_dstport_udp) is disabled, ignoring section
Section omr_dst_bypass_wan2_dstip_4 (omr_dst_bypass_wan2_rule) is disabled, ignoring section
Section omr_dst_bypass_wan2_dstip_4_accept (omr_dst_bypass_wan2_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_wan2_srcip_4 (omr_dst_bypass_wan2_srcip) is disabled, ignoring section
Section omr_dst_bypass_wan2_mac_4 (omr_dst_bypass_wan2_mac) is disabled, ignoring section
Section omr_dst_bypass_wan2_srcport_tcp_4 (omr_dst_bypass_wan2_srcport_tcp) is disabled, ignoring section
Section omr_dst_bypass_wan2_srcport_udp_4 (omr_dst_bypass_wan2_srcport_udp) is disabled, ignoring section
Section omr_dst_bypass_wan2_dstport_tcp_4 (omr_dst_bypass_wan2_dstport_tcp) is disabled, ignoring section
Section omr_dst_bypass_wan2_dstport_udp_4 (omr_dst_bypass_wan2_dstport_udp) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstip_4 (omr_dst_bypass_tun0_rule) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstip_4_accept (omr_dst_bypass_tun0_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_tun0_srcip_4 (omr_dst_bypass_tun0_srcip) is disabled, ignoring section
Section omr_dst_bypass_tun0_mac_4 (omr_dst_bypass_tun0_mac) is disabled, ignoring section
Section omr_dst_bypass_tun0_srcport_tcp_4 (omr_dst_bypass_tun0_srcport_tcp) is disabled, ignoring section
Section omr_dst_bypass_tun0_srcport_udp_4 (omr_dst_bypass_tun0_srcport_udp) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstport_tcp_4 (omr_dst_bypass_tun0_dstport_tcp) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstport_udp_4 (omr_dst_bypass_tun0_dstport_udp) is disabled, ignoring section
Section omr_dst_bypass_all_srcip_4 (omr_dst_bypass_all_srcip) is disabled, ignoring section
Section omr_dst_bypass_all_mac_4 (omr_dst_bypass_all_mac) is disabled, ignoring section
Section omr_dst_bypass_all_srcport_tcp_4 (omr_dst_bypass_all_srcport_tcp) is disabled, ignoring section
Section omr_dst_bypass_all_srcport_udp_4 (omr_dst_bypass_all_srcport_udp) is disabled, ignoring section
Section omr_dst_bypass_all_dstport_tcp_4 (omr_dst_bypass_all_dstport_tcp) is disabled, ignoring section
Section omr_dst_bypass_all_dstport_udp_4 (omr_dst_bypass_all_dstport_udp) is disabled, ignoring section
Section user specifies unreachable path '/etc/firewall.user', ignoring section
Section omr_bypass option 'reload' is not supported by fw4
Automatically including '/usr/share/nftables.d/table-post/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/dstnat/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/forward/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/srcnat/20-miniupnpd.nft'
root@OpenMPTCProuter:~#
```

```
root@OpenMPTCProuter:~# nft list ruleset
table ip mangle {
	chain PREROUTING {
		type filter hook prerouting priority mangle; policy accept;
	}

	chain INPUT {
		type filter hook input priority mangle; policy accept;
		counter packets 10115 bytes 10363839 jump omr-bypass-dpi
	}

	chain FORWARD {
		type filter hook forward priority mangle; policy accept;
[... 582 lines cut ...]
		meta l4proto tcp ip daddr @omr_dscp_ef_4 counter packets 0 bytes 0 ip dscp set ef comment "!fw4: omr_dscp_ef_4"
		meta l4proto udp ip daddr @omr_dscp_ef_4 counter packets 0 bytes 0 ip dscp set ef comment "!fw4: omr_dscp_ef_4"
		meta l4proto icmp ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 counter packets 43 bytes 26256 ip dscp set cs7 comment "!fw4: omr_dscp_rule1"
		ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 udp sport { 53, 123, 5353 } udp dport 0-65535 counter packets 12 bytes 912 ip dscp set cs4 comment "!fw4: omr_dscp_rule2"
[... cut ...]
		iifname "eth0" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone lan IPv4/IPv6 ingress MTU fixing"
		iifname { "wan1", "wan2" } tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 ingress MTU fixing"
		iifname "tun0" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone vpn IPv4/IPv6 ingress MTU fixing"
	}

	chain upnp_forward {
	}

	chain upnp_prerouting {
	}

	chain upnp_postrouting {
	}
}
root@OpenMPTCProuter:~#
```

Ysurac, 2024-12-05:
Can you try latest snapshots ? I fixed some issues.

Schinkentanz, 2024-12-07:
Thanks for the update @Ysurac. I've tried the latest snapshot and it fixes the initial problem, but when adding the `amazon_aws` or `whatsapp` services, the routing for `netflix` no longer works (the configured domains are still routed correctly (e.g. `ifconfig.me`)). I've also installed Pi-Hole, but the behaviour is the same with or without it. If I remove these services, the routing works correctly again.

```
❯ ✅ traceroute nflxvideo.net
 1  192.168.0.1 (192.168.0.1)  14.499 ms  3.236 ms  3.097 ms
 2  openmptcprouter.lan (192.168.42.1)  3.622 ms  3.193 ms  3.210 ms
 3  192.168.178.1 (192.168.178.1)  6.173 ms  4.882 ms  7.659 ms
.... <redacted>
```

```
❯ ❌ traceroute ipv4-c088-ord001-dev-ix.1.oca.nflxvideo.net
 1  192.168.0.1 (192.168.0.1)  5.764 ms  3.356 ms  2.895 ms
 2  openmptcprouter.lan (192.168.42.1)  4.613 ms  3.508 ms  3.238 ms
 3  10.255.252.1 (10.255.252.1)  19.042 ms  18.521 ms  18.231 ms
```

```
❯ ✅ traceroute ifconfig.me
 1  192.168.0.1 (192.168.0.1)  17.401 ms  5.172 ms  2.744 ms
 2  openmptcprouter.lan (192.168.42.1)  3.666 ms  3.492 ms  3.403 ms
 3  192.168.178.1 (192.168.178.1)  6.686 ms  5.141 ms  5.029 ms
```

```
❯ ✅ traceroute random.subdomain.ifconfig.me
traceroute to random.subdomain.ifconfig.me (34.160.111.145), 64 hops max, 40 byte packets
 1  192.168.0.1 (192.168.0.1)  5.068 ms  3.177 ms  3.054 ms
 2  openmptcprouter.lan (192.168.42.1)  3.916 ms  3.745 ms  3.351 ms
 3  192.168.178.1 (192.168.178.1)  5.865 ms  10.092 ms  11.031 ms
```

```
❯ ✅ traceroute email-smtp.eu-west-1.amazonaws.com
 1  192.168.0.1 (192.168.0.1)  95.632 ms  3.041 ms  3.114 ms
 2  openmptcprouter.lan (192.168.42.1)  3.851 ms  3.352 ms  3.226 ms
 3  192.168.178.1 (192.168.178.1)  5.643 ms  6.034 ms  5.265 ms
```

> [!NOTE]
>  The output for `nft list ruleset` also includes rules for `whatsapp`, even though I've completely removed the service.
> The configured domains are:
> - `ifconfig.me`
> 
> The configured services are:
> - `netflix`
> - `amazon_aws`

<details>

<summary>Click here to see output of "nft list ruleset"</summary>

```
table ip mangle {
	chain PREROUTING {
		type filter hook prerouting priority mangle; policy accept;
	}

	chain INPUT {
		type filter hook input priority mangle; policy accept;
		counter packets 15418 bytes 2577720 jump omr-bypass-dpi
	}

	chain FORWARD {
		type filter hook forward priority mangle; policy accept;
		counter packets 1273 bytes 222381 jump omr-bypass-dpi
	}

	chain OUTPUT {
		type route hook output priority mangle; policy accept;
[... 1091 lines cut ...]
		meta l4proto udp ip daddr @omr_dscp_ef_4 counter packets 0 bytes 0 ip dscp set ef comment "!fw4: omr_dscp_ef_4"
		meta l4proto icmp ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 counter packets 213 bytes 19328 ip dscp set cs7 comment "!fw4: omr_dscp_rule1"
		ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 udp sport { 53, 123, 5353 } udp dport 0-65535 counter packets 26 bytes 2136 ip dscp set cs4 comment "!fw4: omr_dscp_rule2"
		ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 tcp sport { 53, 5353 } tcp dport 0-65535 counter packets 0 bytes 0 ip dscp set cs4 comment "!fw4: omr_dscp_rule3"
		ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 tcp sport 0-65535 tcp dport 65500 counter packets 0 bytes 0 ip dscp set cs4 comment "!fw4: omr_dscp_rule4"
		ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 tcp sport 0-65535 tcp dport { 65001, 65011, 65301, 65401 } counter packets 0 bytes 0 ip dscp set cs7 comment "!fw4: omr_dscp_rule5"
		ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 udp sport 0-65535 udp dport { 65001, 65301 } counter packets 0 bytes 0 ip dscp set cs7 comment "!fw4: omr_dscp_rule6"
		ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 tcp sport 0-65535 tcp dport { 65101, 65228 } counter packets 0 bytes 0 ip dscp set cs6 comment "!fw4: omr_dscp_rule7"
		iifname "eth0" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone lan IPv4/IPv6 ingress MTU fixing"
		iifname { "wan1", "wan2" } tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 ingress MTU fixing"
		iifname "tun0" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone vpn IPv4/IPv6 ingress MTU fixing"
	}

	chain upnp_forward {
	}

	chain upnp_prerouting {
	}

	chain upnp_postrouting {
	}
}
```

</details>

Ysurac, 2024-12-07:
There is some issues on whatapps and aws ranges...

Ysurac, 2024-12-10:
Should be better in latest snapshot

Schinkentanz, 2024-12-22:
Thanks for the update @Ysurac. Today I got around to testing the OMR bypass configuration with a freshly installed VPS and the latest RPI5 snapshot, but it seems that the bypass mechanism isn't working at all at the moment. I also don't see (via `htop`) the usual work happening on the router after applying new rules

- OpenMPTCProuter version:
`openmptcprouter-v0.62-snapshot-6.6-r0+28194-cc69be0c13-bcm27xx-bcm2712-rpi-5-squashfs-factory`
- OpenMPTCProuter VPS version:
`0.1032-test 6.6.36-x64v2-xanmod1`

```
root@OpenMPTCProuter:~# /etc/init.d/omr-bypass restart
root@OpenMPTCProuter:~#
```

```
root@OpenMPTCProuter:~# /etc/init.d/omr-bypass status
active with no instances
root@OpenMPTCProuter:~#
```

```
root@OpenMPTCProuter:~# /etc/init.d/firewall restart
Section user specifies unreachable path '/etc/firewall.user', ignoring section
Section omr_bypass option 'reload' is not supported by fw4
Section omr_bypass is disabled, ignoring section
Automatically including '/usr/share/nftables.d/table-post/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/dstnat/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/forward/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/srcnat/20-miniupnpd.nft'
root@OpenMPTCProuter:~#
```

<details>

<summary>Click here to see output of "nft list ruleset"</summary>

```
root@OpenMPTCProuter:~# nft list ruleset
table ip mangle {
	chain PREROUTING {
		type filter hook prerouting priority mangle; policy accept;
	}

	chain INPUT {
		type filter hook input priority mangle; policy accept;
	}

	chain FORWARD {
		type filter hook forward priority mangle; policy accept;
	}

	chain OUTPUT {
		type route hook output priority mangle; policy accept;
	}

	chain POSTROUTING {
		type filter hook postrouting priority mangle; policy accept;
	}
}
table inet fw4 {
	ct helper amanda {
		type "amanda" protocol udp
		l3proto inet
	}

	ct helper ftp {
		type "ftp" protocol tcp
		l3proto inet
	}

	ct helper RAS {
		type "RAS" protocol udp
[... 472 lines cut ...]
		meta l4proto icmp ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 counter packets 25 bytes 1812 ip dscp set cs7 comment "!fw4: omr_dscp_rule1"
		ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 udp sport { 53, 123, 5353 } udp dport 0-65535 counter packets 1 bytes 76 ip dscp set cs4 comment "!fw4: omr_dscp_rule2"
		ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 tcp sport { 53, 5353 } tcp dport 0-65535 counter packets 0 bytes 0 ip dscp set cs4 comment "!fw4: omr_dscp_rule3"
		ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 tcp sport 0-65535 tcp dport 65500 counter packets 0 bytes 0 ip dscp set cs4 comment "!fw4: omr_dscp_rule4"
		ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 tcp sport 0-65535 tcp dport { 65001, 65011, 65301, 65401 } counter packets 0 bytes 0 ip dscp set cs7 comment "!fw4: omr_dscp_rule5"
		ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 udp sport 0-65535 udp dport { 65001, 65301 } counter packets 0 bytes 0 ip dscp set cs7 comment "!fw4: omr_dscp_rule6"
		ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 tcp sport 0-65535 tcp dport { 65101, 65228 } counter packets 0 bytes 0 ip dscp set cs6 comment "!fw4: omr_dscp_rule7"
		iifname "eth0" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone lan IPv4/IPv6 ingress MTU fixing"
		iifname { "wan1", "wan2" } tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 ingress MTU fixing"
		iifname { "tun0", "tun1", "tun2", "tun3" } tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone vpn IPv4/IPv6 ingress MTU fixing"
	}

	chain upnp_forward {
	}

	chain upnp_prerouting {
	}

	chain upnp_postrouting {
	}
}
root@OpenMPTCProuter:~#
```

</details>

SpencerXZX, 2024-12-23:
I'm seeing this as well on the 6.6 snapshot, I think the "Section omr_bypass is disabled, ignoring section" is relevant. None of OMR Bypass is working.

Ysurac, 2024-12-23:
I would need the result of ```uci show omr-bypass``` and of ```uci show firewall``` via SSH on the router

SpencerXZX, 2024-12-23:
Router version: 0.62-snapshot-6.6
VPS Version: 0.1032-test 6.6.36-x64v3-xanmod1

uci show omr-bypass:
```
omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com' 'canalplus.com' 'canalplus-                                                                                                                                   cdn.net' 'canalplus.pro' 'canal-plus.net'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorne                                                                                                                                   r.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.amazonvideo=proto
omr-bypass.amazonvideo.url='cloudfront.net' 'llnw.net'
omr-bypass.free=proto
omr-bypass.free.url='free.fr' 'freebox.fr' 'oqee.tv' 'oqee.net' 'proxad.net'
omr-bypass.orange=proto
omr-bypass.orange.url='orange.fr' 'sosh.fr' 'liveperson.net' 'liveperson.com' 'l                                                                                                                                   psn.net' 'lpsnmedia.net' 'francetelecom.fr'
omr-bypass.sky=proto
omr-bypass.sky.url='sky.com' 'skycdp.com' 'skyanywhere.com' 'epgsky.com' 'skycdn                                                                                                                                   .it'
omr-bypass.captive_portal=proto
omr-bypass.captive_portal.url='captive.apple.com' 'connectivitycheck.gstatic.com                                                                                                                                   ' 'clients3.google.com' 'www.msftconnecttest.com' 'www.msftncsi.com' 'nmcheck.gn                                                                                                                                   ome.org' 'networkcheck.kde.org'
omr-bypass.eth0=interface
omr-bypass.eth0.id='9999'
omr-bypass.eth1=interface
omr-bypass.eth1.id='3'
omr-bypass.eth2=interface
omr-bypass.eth2.id='4'
omr-bypass.eth3=interface
omr-bypass.eth3.id='5'
omr-bypass.tun0=interface
omr-bypass.tun0.id='1500'
omr-bypass.global=global
omr-bypass.global.vpn_ipv4_md5='68b329da9893e34099c7d8ad5cb9c940'
omr-bypass.global.vpn_ipv6_md5='68b329da9893e34099c7d8ad5cb9c940'
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='bestbuy.com'
omr-bypass.@domains[0].interface='eth2'
omr-bypass.@domains[0].family='ipv4ipv6'
```

uci show firewall:
```
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='REJECT'
firewall.@defaults[0].output='REJECT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].fullcone='0'
firewall.@defaults[0].flow_offloading='0'
firewall.@defaults[0].flow_offloading_hw='0'
firewall.zone_lan=zone
firewall.zone_lan.name='lan'
firewall.zone_lan.network='lan'
firewall.zone_lan.input='ACCEPT'
firewall.zone_lan.output='ACCEPT'
firewall.zone_lan.forward='ACCEPT'
firewall.zone_lan.auto_helper='1'
firewall.zone_lan.mtu_fix='1'
firewall.zone_wan=zone
firewall.zone_wan.name='wan'
firewall.zone_wan.input='REJECT'
firewall.zone_wan.output='ACCEPT'
firewall.zone_wan.forward='REJECT'
firewall.zone_wan.fullcone4='0'
firewall.zone_wan.fullcone6='0'
[... 348 lines cut ...]
firewall.omr_dscp_rule7.target='DSCP'
firewall.omr_dscp_rule7.set_dscp='CS6'
firewall.omr_dscp_rule7.src='*'
firewall.omr_dscp_rule7.dest='*'
firewall.omr_dscp_rule7.src_ip='0.0.0.0/0'
firewall.omr_dscp_rule7.dest_ip='0.0.0.0/0'
firewall.omr_dscp_rule7.proto='tcp'
firewall.omr_dscp_rule7.enabled='1'
firewall.omr_dscp_rule7.src_port='0-65535'
firewall.omr_dscp_rule7.dest_port='65101' '65228'
firewall.omr_dscp_rule8=rule
firewall.omr_dscp_rule8.name='omr_dscp_rule8'
firewall.omr_dscp_rule8.target='DSCP'
firewall.omr_dscp_rule8.set_dscp='EF'
firewall.omr_dscp_rule8.src='*'
firewall.omr_dscp_rule8.dest='*'
firewall.omr_dscp_rule8.src_ip='0.0.0.0/0'
firewall.omr_dscp_rule8.dest_ip='0.0.0.0/0'
firewall.omr_dscp_rule8.proto='tcp'
firewall.omr_dscp_rule8.enabled='1'
firewall.omr_dscp_rule8.src_port='0-65535'
firewall.omr_dscp_rule8.dest_port='21360'
```

I've verified that OMR is the DNS server assigned to my PC. Domains added to OMR-Bypass still go through the proxy.

Ysurac, 2024-12-23:
As it's a domain bypass, I would also need the ```uci show dhcp``` result.
For the ```firewall.omr_bypass.enabled``` it's set to 1 when omr-bypass service is started and set to 0 when the service is stopped. It's set directly by init script.

SpencerXZX, 2024-12-23:
uci show dhcp:
```
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].cachesize='1000'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].filter_aaaa='0'
dhcp.@dnsmasq[0].filter_a='0'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].dnsforwardmax='1500'
dhcp.@dnsmasq[0].interface='loopback' 'lan'
dhcp.@dnsmasq[0].notinterface='wan3' 'wan1' 'wan2'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.dhcpv4='server'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.omr_dscp_cs0=ipset
dhcp.omr_dscp_cs0.name='omr_dscp_cs0_4'
dhcp.omr_dscp_cs1=ipset
dhcp.omr_dscp_cs1.name='omr_dscp_cs1_4'
dhcp.omr_dscp_cs1.domain='download.qq.com' 'steamcontent.com' 'gs2.ww.prod.dl.pl                                                                                                                                   aystation.net' 'dropbox.com' 'dropboxstatic.com' 'dropbox-dns.com' 'log.getdropb                                                                                                                                   ox.com' 'drive.google.com' 'drive-thirdparty.googleusercontent.com' 'docs.google                                                                                                                                   .com' 'docs.googleusercontent.com' 'gvt1.com' 'mmg-fna.whatsapp.net' 'upload.you                                                                                                                                   tube.com' 'upload.video.google.com' 'windowsupdate.com' 'update.microsoft.com'
dhcp.omr_dscp_cs2=ipset
dhcp.omr_dscp_cs2.name='omr_dscp_cs2_4'
dhcp.omr_dscp_cs2.domain='googletagmanager.com' 'googleusercontent.com' 'google.                                                                                                                                   com' 'fbcdn.net' 'akamaihd.net' 'whatsapp.net' 'whatsapp.com' 'zoom.us' 'googlea                                                                                                                                   pis.com' '1e100.net' 'hwcdn.net'
dhcp.omr_dscp_cs3=ipset
dhcp.omr_dscp_cs3.name='omr_dscp_cs3_4'
dhcp.omr_dscp_cs4=ipset
dhcp.omr_dscp_cs4.name='omr_dscp_cs4_4'
dhcp.omr_dscp_cs4.domain='googlevideo.com' 'nflxvideo.net' 's3.ll.dash.row.aiv-c                                                                                                                                   dn.net' 'd25xi40x97liuc.cloudfront.net' 'aiv-delivery.net' 'fbcdn.net' 'ttvnw.ne                                                                                                                                   t' 'vevo.com' 'audio-fa.scdn.com' 'deezer.com' 'sndcdn.com' 'last.fm' 'v.redd.it                                                                                                                                   '
dhcp.omr_dscp_cs5=ipset
dhcp.omr_dscp_cs5.name='omr_dscp_cs5_4'
dhcp.omr_dscp_cs5.domain='tv.milkywan.fr'
dhcp.omr_dscp_cs6=ipset
dhcp.omr_dscp_cs6.name='omr_dscp_cs6_4'
dhcp.omr_dscp_cs7=ipset
dhcp.omr_dscp_cs7.name='omr_dscp_cs7_4'
dhcp.omr_dscp_ef=ipset
dhcp.omr_dscp_ef.name='omr_dscp_ef_4'
```

hle5128, 2024-12-24:
im having a same issue bypass not working at all, running latest snapshot openmptcprouter v0.62-snapshot-6.6 r0+28194 @Ysurac 
none of bypass is working, dns is set to router. 
`uci show omr-bypass`

`root@OpenMPTCProuter:~# uci show omr-bypass
omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com' 'canalplus.com' 'canalplus-cdn.net' 'canalplus.pro' 'canal-plus.net'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.amazonvideo=proto
omr-bypass.amazonvideo.url='cloudfront.net' 'llnw.net'
omr-bypass.free=proto
omr-bypass.free.url='free.fr' 'freebox.fr' 'oqee.tv' 'oqee.net' 'proxad.net'
omr-bypass.orange=proto
omr-bypass.orange.url='orange.fr' 'sosh.fr' 'liveperson.net' 'liveperson.com' 'lpsn.net' 'lpsnmedia.net' 'francetelecom.fr' 'sky.com' 'skycdp.com' 'skyanywhere.com' 'epgsky.com' 'skycdn.it'
omr-bypass.sky=proto
omr-bypass.eth1=interface
omr-bypass.eth1.id='9999'
omr-bypass.wan1=interface
omr-bypass.wan1.id='5'
omr-bypass.wan2=interface
omr-bypass.wan2.id='7'
omr-bypass.tun0=interface
omr-bypass.tun0.id='1500'
omr-bypass.global=global
omr-bypass.global.vpn_ipv4_md5='68b329da9893e34099c7d8ad5cb9c940'
omr-bypass.global.vpn_ipv6_md5='68b329da9893e34099c7d8ad5cb9c940'
omr-bypass.eth3=interface
omr-bypass.eth3.id='8'
omr-bypass.eth0=interface
omr-bypass.eth0.id='5'
omr-bypass.eth2=interface
omr-bypass.eth2.id='7'
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='ifconfig.me'
omr-bypass.@domains[0].family='ipv4ipv6'
omr-bypass.@domains[1]=domains
omr-bypass.@domains[1].name='reddit.com'
omr-bypass.@domains[1].family='ipv4ipv6'
omr-bypass.@domains[2]=domains
omr-bypass.@domains[2].name='suncoastcreditunion.com'
omr-bypass.@domains[2].family='ipv4ipv6'
omr-bypass.@domains[3]=domains
omr-bypass.@domains[3].name='*.immedia-semi.com'
omr-bypass.@domains[3].family='ipv4ipv6'
omr-bypass.@domains[4]=domains
omr-bypass.@domains[4].name='ring.com'
omr-bypass.@domains[4].family='ipv4ipv6'
omr-bypass.@domains[5]=domains
omr-bypass.@domains[5].name='smartthings.com'
omr-bypass.@domains[5].family='ipv4ipv6'
omr-bypass.@asns[0]=asns
omr-bypass.@asns[0].asn='16509'
omr-bypass.@asns[0].note='blink'
omr-bypass.@dest_port[0]=dest_port
omr-bypass.@dest_port[0].dport='10001'
omr-bypass.@dest_port[0].proto='tcp'
omr-bypass.@dest_port[0].note='wyze'
omr-bypass.@dest_port[1]=dest_port
omr-bypass.@dest_port[1].dport='22345'
omr-bypass.@dest_port[1].proto='tcp'
omr-bypass.@dest_port[1].note='wyze'
omr-bypass.@dpis[0]=dpis
[... 558 lines cut ...]
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.omr_dscp_cs0=ipset
dhcp.omr_dscp_cs0.name='omr_dscp_cs0_4'
dhcp.omr_dscp_cs1=ipset
dhcp.omr_dscp_cs1.name='omr_dscp_cs1_4'
dhcp.omr_dscp_cs1.domain='download.qq.com' 'steamcontent.com' 'gs2.ww.prod.dl.playstation.net' 'dropbox.com' 'dropboxstatic.com' 'dropbox-dns.com' 'log.getdropbox.com' 'drive.google.com' 'drive-thirdparty.googleusercontent.com' 'docs.google.com' 'docs.googleusercontent.com' 'gvt1.com' 'mmg-fna.whatsapp.net' 'upload.youtube.com' 'upload.video.google.com' 'windowsupdate.com' 'update.microsoft.com'
dhcp.omr_dscp_cs2=ipset
dhcp.omr_dscp_cs2.name='omr_dscp_cs2_4'
dhcp.omr_dscp_cs2.domain='googletagmanager.com' 'googleusercontent.com' 'google.com' 'fbcdn.net' 'akamaihd.net' 'whatsapp.net' 'whatsapp.com' 'zoom.us' 'googleapis.com' '1e100.net' 'hwcdn.net'
dhcp.omr_dscp_cs3=ipset
dhcp.omr_dscp_cs3.name='omr_dscp_cs3_4'
dhcp.omr_dscp_cs4=ipset
dhcp.omr_dscp_cs4.name='omr_dscp_cs4_4'
dhcp.omr_dscp_cs4.domain='googlevideo.com' 'nflxvideo.net' 's3.ll.dash.row.aiv-cdn.net' 'd25xi40x97liuc.cloudfront.net' 'aiv-delivery.net' 'fbcdn.net' 'ttvnw.net' 'vevo.com' 'audio-fa.scdn.com' 'deezer.com' 'sndcdn.com' 'last.fm' 'v.redd.it'
dhcp.omr_dscp_cs5=ipset
dhcp.omr_dscp_cs5.name='omr_dscp_cs5_4'
dhcp.omr_dscp_cs5.domain='tv.milkywan.fr'
dhcp.omr_dscp_cs6=ipset
dhcp.omr_dscp_cs6.name='omr_dscp_cs6_4'
dhcp.omr_dscp_cs7=ipset
dhcp.omr_dscp_cs7.name='omr_dscp_cs7_4'
dhcp.omr_dscp_ef=ipset
dhcp.omr_dscp_ef.name='omr_dscp_ef_4'`

hle5128, 2024-12-24:
`jimmyle-m16@Tintin-M4P ~ % traceroute ring.com
traceroute: Warning: ring.com has multiple addresses; using 52.46.150.230
traceroute to ring.com (52.46.150.230), 64 hops max, 40 byte packets
 1  192.168.100.1 (192.168.100.1)  53.818 ms  2.027 ms  1.849 ms
 2  10.255.252.1 (10.255.252.1)  171.467 ms  82.744 ms  139.467 ms
 3  VPS IP  53.106 ms  35.573 ms  59.518 ms
 4  te0-0-1-2.nr11.b002802-4.mia01.atlas.cogentco.com (38.142.42.161)  120.556 ms  26.665 ms  50.520 ms
 5  te0-6-0-7.rcr21.b002802-2.mia01.atlas.cogentco.com (154.24.4.217)  123.570 ms
    te0-1-0-7.rcr21.b002802-2.mia01.atlas.cogentco.com (154.24.1.93)  71.330 ms
    te0-6-0-7.rcr21.b002802-2.mia01.atlas.cogentco.com (154.24.4.217)  45.836 ms
 6  be3411.ccr22.mia01.atlas.cogentco.com (154.54.26.41)  51.008 ms^C`

Ysurac, 2024-12-24:
OMR-ByPass doesn't seems to be running in both case. Maybe it crash... I will check after christmas.

Ysurac, 2024-12-26:
Should be better in latest snapshot. It's still compiling...

SpencerXZX, 2024-12-26:
Hi Ysurac,
I have to run:

/etc/init.d/omr-bypass restart

after making changes in omr-bypass in order to get it to work. It then works until I make another change, such as adding a domain. Then I have to restart again. I manually applied the omr bypass patch, didn't see an updated snapshot for x86.

Ysurac, 2024-12-26:
The changes seems to work, it was due to an "exit 0" in stop part of init script. Now after a ```/etc/init.d/omr-bypass reload```, ```uci get firewall.omr_bypass.enabled``` should return 1.
Else this can be another issue...

hle5128, 2024-12-26:
after booth, adding a fresh domain in bypass and hit save result these error and bypass did not work on ifconfig.me

```
1root@OpenMPTCProuter:~# nft list ruleset
table ip mangle {
        chain PREROUTING {
                type filter hook prerouting priority mangle; policy accept;
        }

        chain INPUT {
                type filter hook input priority mangle; policy accept;
                counter packets 119543 bytes 305199825 jump omr-bypass-dpi
        }

        chain FORWARD {
                type filter hook forward priority mangle; policy accept;
                counter packets 308 bytes 33885 jump omr-bypass-dpi
        }

        chain OUTPUT {
                type route hook output priority mangle; policy accept;
        }

        chain POSTROUTING {
                type filter hook postrouting priority mangle; policy accept;
        }

        chain omr-bypass-dpi {
        }
}
table inet fw4 {
        ct helper amanda {
                type "amanda" protocol udp
                l3proto inet
        }

        ct helper ftp {
                type "ftp" protocol tcp
                l3proto inet
        }

        ct helper RAS {
                type "RAS" protocol udp
                l3proto inet
        }

        ct helper Q.931 {
                type "Q.931" protocol tcp
                l3proto inet
        }

        ct helper irc {
                type "irc" protocol tcp
                l3proto ip
        }

        ct helper netbios-ns {
                type "netbios-ns" protocol udp
                l3proto ip
        }

        ct helper pptp {
                type "pptp" protocol tcp
                l3proto ip
        }

        ct helper sane {
                type "sane" protocol tcp
                l3proto inet
        }
[... 566 lines cut ...]
Dec 26 17:27:58 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Dec 26 17:28:04 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Dec 26 17:28:05 OpenMPTCProuter user.notice omr-bypass: Reload dnsmasq...
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Dec 26 17:28:05 OpenMPTCProuter daemon.err procd: Got unexpected signal 1
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Dec 26 17:28:05 OpenMPTCProuter daemon.warn dnsmasq[1]: warning: interface omrvpn does not currently exist
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5353
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Dec 26 17:28:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Dec 26 17:28:05 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
```

SpencerXZX, 2024-12-26:
uci get firewall.omr_bypass.enabled does return 1 for me.

However, after I hit "Save and Apply" I run a traceroute on one of the domains, and it goes through my VPS. I then execute: 

/etc/init.d/omr-bypass restart

And traceroute again, and it properly goes straight to my WAN ISP and not through VPS.

SpencerXZX, 2024-12-27:
Update: This also has to be done on first boot otherwise the rule does not work. I ran ```/etc/init.d/omr-bypass restart``` between these traceroute tests after a fresh reboot on the December 27th snapshot for x86_64.

[image]

sieade245, 2024-12-28:
> Update: This also has to be done on first boot otherwise the rule does not work. I ran `/etc/init.d/omr-bypass restart` between these traceroute tests after a fresh reboot on the December 27th snapshot for x86_64.
> 
> <img alt="image" width="492" src="https://private-user-images.githubusercontent.com/34640122/398957662-6e1d4815-1aa5-4ce4-bc1c-81f07aed4a87.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU0MjY0MDcsIm5iZiI6MTczNTQyNjEwNywicGF0aCI6Ii8zNDY0MDEyMi8zOTg5NTc2NjItNmUxZDQ4MTUtMWFhNS00Y2U0LWJjMWMtODFmMDdhZWQ0YTg3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjI4VDIyNDgyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY4MWMyOGNkM2ZmNzE2MzUyNWU4Njg4YjU1YzBkMDg3NTQ2YTIxZDY2ZmQxYzYzZmZkY2YxZmI5MTcwNGEzMTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.nbJdOTQpe7Klu7qayyBzg8HZtFi__mbi50WdSNLDy3Q">

Hi just wanted to say thanks for the fault diagnosis work on this. I've been tearing my hair out myself trying to set up the bypass. I've received several hundreds of negative wife points as netflix would work then not work then work as I was working on it. Like you I was sure I had the settings correct. I was forcing the restart a different way by changing the default proxy method each time which seemed to fix it..until I made another change to the rules. Really glad to see it wasn't just me doing it wrong!

hle5128, 2024-12-29:
i noticed that more often bypass stopped working, i have to restart command "/etc/init.d/omr-bypass restart" make it work again, i  would said they crashed more often

Ysurac, 2024-12-29:
Check latest snapshot, reload and restart is now the same. This can be slow to start on some services bypass (for example amazon)

hle5128, 2024-12-30:
> Check latest snapshot, reload and restart is now the same. This can be slow to start on some services bypass (for example amazon)

I'm using the latest snapshot: [openmptcprouter-v0.62-snapshot-6.6-r0+28194-cc69be0c13-rockchip-armv8-friendlyarm_nanopi-r5s], which is causing the omrbypass to crash frequently. 

Notice a sample log below:
at `Dec 29 23:11:28 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running` when I restart the firewall manually `/etc/init.d/omr-bypass restart`
success curl ifconfig.me reveals the ip of the WAN.
During this time, everything worked, then 
at `Dec 29 23:19:33 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...`
when the omr-bypass auto-crashed and restarted and stopped working,
then it crashed again auto-reload `Dec 29 23:31:58 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running`

but when it auto-reload, the omr-bypass said running but omr-bypass is not working (confirmed by ifconfig.me). 
We have to manually execute `/etc/init.d/omr-bypass restart` to work again, but then it will go to a loop crash again.

Here is the full log of 3 cycles crashed, even though it restarted bypass itself, but it didn’t work until we had to execute the `/etc/init.d/omr-bypass restart`

```
Dec 29 23:11:28 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Dec 29 23:11:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) cc:db:a7:0c:f6:54 
Dec 29 23:11:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.120 cc:db:a7:0c:f6:54 
Dec 29 23:11:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.120 cc:db:a7:0c:f6:54 
Dec 29 23:11:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.120 cc:db:a7:0c:f6:54 espressif
Dec 29 23:12:11 OpenMPTCProuter user.notice post-tracking-002-error: wan2 (eth2) switched off because check error and ping from 192.168.6.27 error (1.0.0.1,114.114.115.115,1.2.4.8)
Dec 29 23:12:12 OpenMPTCProuter user.notice post-tracking-002-error: Delete default route to VPS IP dev eth2
Dec 29 23:13:00 OpenMPTCProuter user.notice post-tracking-003-up: wan2 (eth2) switched up
Dec 29 23:13:01 OpenMPTCProuter user.notice post-tracking-003-up: Interface route not yet set, set route ip r add default via 192.168.6.1 dev eth2 metric 6
Dec 29 23:13:01 OpenMPTCProuter user.notice post-tracking-003-up: Status change, reload MPTCP config for eth2
Dec 29 23:13:06 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.200 60:e8:5b:7e:89:82 
Dec 29 23:13:06 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.200 60:e8:5b:7e:89:82 airthings-view
Dec 29 23:13:06 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.200 60:e8:5b:7e:89:82 
Dec 29 23:13:06 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.200 60:e8:5b:7e:89:82 airthings-view
Dec 29 23:13:38 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.200 60:e8:5b:7e:89:82 
Dec 29 23:13:38 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.200 60:e8:5b:7e:89:82 airthings-view
Dec 29 23:14:28 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.183 64:52:99:a3:0c:cb 
Dec 29 23:14:28 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.183 64:52:99:a3:0c:cb MyQ-90A
Dec 29 23:14:31 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.200 60:e8:5b:7e:89:82 
Dec 29 23:14:31 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.200 60:e8:5b:7e:89:82 airthings-view
Dec 29 23:14:31 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.200 60:e8:5b:7e:89:82 
Dec 29 23:14:31 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.200 60:e8:5b:7e:89:82 airthings-view
Dec 29 23:15:01 OpenMPTCProuter user.notice post-tracking-002-error: wan1 (eth0) switched off because check error and ping from 192.168.5.122 error (4.2.2.1,8.8.8.8,80.67.169.12)
Dec 29 23:15:01 OpenMPTCProuter user.notice post-tracking-002-error: Delete default route to VPS IP dev eth0
Dec 29 23:15:24 OpenMPTCProuter user.notice post-tracking-003-up: wan1 (eth0) switched up
Dec 29 23:15:25 OpenMPTCProuter user.notice post-tracking-003-up: Interface route not yet set, set route ip r add default via 192.168.5.1 dev eth0 metric 5
Dec 29 23:15:26 OpenMPTCProuter user.notice post-tracking-003-up: Status change, reload MPTCP config for eth0
Dec 29 23:16:08 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.200 60:e8:5b:7e:89:82 
Dec 29 23:16:08 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.200 60:e8:5b:7e:89:82 airthings-view
Dec 29 23:17:04 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) cc:db:a7:0c:f6:54 
Dec 29 23:17:04 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.120 cc:db:a7:0c:f6:54 
Dec 29 23:17:04 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.120 cc:db:a7:0c:f6:54 
Dec 29 23:17:04 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.120 cc:db:a7:0c:f6:54 espressif
Dec 29 23:17:06 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.120 cc:db:a7:0c:f6:54 
Dec 29 23:17:06 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.120 cc:db:a7:0c:f6:54 espressif
[... cut ...]
Dec 29 23:31:56 OpenMPTCProuter user.notice post-tracking-003-up: Status change, reload MPTCP config for eth2
Dec 29 23:31:57 OpenMPTCProuter user.notice omr-bypass: Reload dnsmasq...
Dec 29 23:31:58 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Dec 29 23:31:58 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Dec 29 23:31:58 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Dec 29 23:31:58 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
```

hle5128, 2024-12-31:
It appears there is an ongoing issue. The omrvpn and the system frequently crash or restart. It seems that whenever a restart occurs, the bypass is expected to run, but it doesn't. As a result, each time the system restarts, we need to manually run `/etc/init.d/omr-bypass restart` to get it working again. We are uncertain why the router keeps restarting the connection.

Ysurac, 2025-01-01:
I removed OMR-ByPass restart in latest snapshot, but I will test asap.

hle5128, 2025-01-03:
> I removed OMR-ByPass restart in latest snapshot, but I will test asap.

So far, the version [openmptcprouter-v0.62-snapshot-6.6-r0+28194-cc69be0c13-rockchip-armv8-friendlyarm_nanopi-r5s-ext4..>] has resolved the restart issue, and it has been running without any crashes omr-bypass or problems for over half a day. I will continue to report if any crashes occur for the rest of the day.

Schinkentanz, 2025-01-20:
Tested again today with the latest snapshot (`openmptcprouter-v0.62-snapshot-6.6-r0+28308-c06d4df974-bcm27xx-bcm2712-rpi-5-squashfs-factory`) and a fresh VPS install. Everything now works as intended for me:

- [x] OMR-Bypass handles subdomains & TLDs
- [x] Port forwarding & SNAT (combined with a FQDN resolver)
- [x] Pi-Hole

Thanks @Ysurac. Closing this for now. If anyone is still experiencing problems, feel free to comment or start a new issue. Thanks to all who contributed!

closed 2025-01-20 by Schinkentanz

sieade245, 2025-10-24:
> Hi [schinkentanz](https://github.com/schinkentanz), [hle5128](https://github.com/hle5128), [sieade245](https://github.com/sieade245) and [sieade245](https://github.com/sieade245)
> 
> I have been working on trying to get omr-bypass working for 2 days now on newer images without success. Were you every able to get omr-bypass working with any of the latest shapshots or any other versions? If so could you please advise on which version you got working?
> 
> I am running version 0.61rc2-6.6 and omr-bypass works as expected however no version past this are working. I have tested all of the fresh images between 0.61 up to the latest 0.64-snapshot-6.12 r0+31388-17badf6099 without success.
> 
> I also suspect that version 6.2 may work however I was unable to test it as the x86-64 version 6.2 as the images are currently 404 error or corrupt and will not boot.
> 
> OMr-bypass will temporarily work on all of these versions for a short time only after the Save & Apply button is clicked on the wizard page but after a reboot or after some time a script runs and breaks it again. Are you aware of which scripts are likely involved in setting these routes? I am willing to look into the issue more deeply but I don't really know where to start.
> 
> I have opened a new issue here: [#4122](https://github.com/Ysurac/openmptcprouter/issues/4122) but I haven't received any feedback yet.
> 
> Any info that you could provide would be appreciated.
> 
> Thank you.

Hi. In my case no I was never able to get omr bypass to be stable. As Netflix etc doesn't work without it that the end of my mtcp journey. Unfortunately I simply gave up in the end and went back to one