Ysurac/openmptcprouter #3697: UPnP not working

VisionLabGroup, 2024-12-06:
Hello, 

With version 0.61 I can no longer forward ports from the NAS with UPnP for VPN connection.

With version 0.59 everything works perfectly.

Any advice?

  - OpenMPTCProuter version: v0.61-6.6 r0+27346-c7ba5574f5
  - OpenMPTCProuter VPS version: 0.1032 test
  - OpenMPTCProuter VPS provider: Serverplan
  - OpenMPTCProuter platform: x86_64
  - Country: Italy

Ysurac, 2024-12-06:
Can you give me the result of ```uci show upnpd``` via SSH from the router ?

VisionLabGroup, 2024-12-09:
This is the result:

upnpd.config=upnpd
upnpd.config.enabled='1'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='1'
upnpd.config.external_iface='omrvpn'
upnpd.config.external_ip='MY_SERVER_IP'
upnpd.config.uuid='7a4fb65f-9e30-4020-a07b-242870b4e486'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-64999'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'

VisionLabGroup, 2024-12-12:
@Ysurac do you have any suggestions?

VisionLabGroup, 2024-12-15:
Ok, How do I check if VPN works properly?

Where can i find the old version already compiled ready to make us the usb key?

sieade245, 2024-12-28:
Hi. I'm certainly no expert in this however I noticed that in the rules you have listed, you currently are blocking all the ports with your Deny rule. This is the "out of the box" setting. On mine, I have removed this rule using the web interface and also turned off "secure mode". When I did this, my UPNP instantly started working and about 10 UPNP connections were displayed at the top of the page. This might not be the best way to fix it but it fixed mine

VisionLabGroup, 2024-12-29:
@sieade245 Thanks.
I tried as you said but still not going