Ysurac/openmptcprouter #3701: Port Forwarding no Internal IP address

MarcelHService, 2024-12-08:
## Expected Behavior
Hello,
I have a Problem with the Port Forwarding Internal IP address.
i don't have any IP's there.

## Current Behavior
i reinstalled the VPS, Router and change the Switch.

System-Log: https://pastebin.com/4m5aJ9XS

## Specifications
  - OpenMPTCProuter version: v0.61-6.6 r0+27346-c7ba5574f
  - OpenMPTCProuter VPS version: 6.6.36-x64v3-xanmod1 0.1031
  - OpenMPTCProuter VPS provider: Strato.de
  - OpenMPTCProuter platform: x86_64

[image]
[image]
[image]

Ysurac, 2024-12-10:
IPs are available when they are provided by DHCP. You can enter manually IP you want.

MarcelHService, 2024-12-10:
Hey,

I have eth0 to a Network Switch and all devices on the Switch get the Ip via DHCP. 

I Test the Port forrwarding from the Guide but tun0 can't recieve any Packets. 

And this on the Setting any or by typing down custom ip.

winkmichael, 2024-12-11:
I have the same issue. The list is always blank for me, but as Ysurac said you can manually type out the ip address. Note when typing the IP address you must hit enter, before hitting save.

AndreaLeidi, 2024-12-11:
Same here. I bet it's related to the package "luci-app-firewall", that is OMR forked. A software update to the original version id proposed and it will probably break something, so I go on manually as suggested.

MarcelHService, 2024-12-11:
perfekt now it's work.

thanks all :)

closed 2024-12-11 by MarcelHService